libimobiledevice/idevicerestore #495: How to compile for windows?

deyvs02, 2022-06-14:
How to compile for windows?

xanadu-lee, 2022-06-18:
Compile with MinGW

deyvs02, 2022-06-22:
> MinGW

You can compile it for me, or show me a tutorial on how to use MinGW I have no experience with this

deyvs02, 2022-06-22:
F:\Downloads\walton\idevicerestore-master\idevicerestore-master>cd "f:\Downloads\walton\idevicerestore-master\idevicerestore-master\src\"

f:\Downloads\walton\idevicerestore-master\idevicerestore-master\src>C:\MinGW\bin\gcc -c idevicerestore.c
idevicerestore.c:34:25: fatal error: plist/plist.h: No such file or directory
 #include <plist/plist.h>
                         ^
compilation terminated.

xanadu-lee, 2022-06-23:
The following documents may be helpful to you.
[github-windows.txt](https://github.com/libimobiledevice/idevicerestore/files/8965020/github-windows.txt)

xanadu-lee, 2022-06-23:
The following commands may help you.

~~~
pacman -S mingw-w64-i686-toolchain
pacman -S --needed base-devel msys2-devel
pacman -S libreadline-devel
pacman -S mingw-w64-i686-libusb
pacman -S mingw-w64-i686-libzip
pacman -S mingw-w64-i686-libxml2
pacman -S mingw-w64-i686-libssh2
pacman -S mingw-w64-i686-curl
pacman -S vim
pacman -S git

export PATH=/usr/bin:$PATH
export CXX_PREFIX=${HOME}/mingw
export PKG_CONFIG_PATH=${CXX_PREFIX}/lib/pkgconfig

zlib
./configure --prefix=${CXX_PREFIX}

libzip
cmake -DCMAKE_INSTALL_PREFIX=${CXX_PREFIX} .

openssl
./config shared no-asm enable-ssl2 enable-ssl3 enable-tls -fPIC -DOPENSSL_PIC enable-weak-ssl-ciphers

libssh2
./buildconf
./configure

curl
./buildconf --prefix=${CXX_PREFIX} --with-libssh2
./configure --prefix=${CXX_PREFIX} --with-libssh2

libplist
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=2.2.1

libimobiledevice-glue
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.0

libusbmuxd
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=2.0.3

libimobiledevice
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.3.1

libusb
./autogen.sh --prefix=${CXX_PREFIX} --disable-udev

libirecovery
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.1

libideviceactivation
./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.1.2

idevicerestore
./autogen.sh --prefix=${CXX_PREFIX} PACKAGE_VERSION=1.0.1

~~~

deyvs02, 2022-07-19:
> The following commands may help you.
> 
> ```
> pacman -S mingw-w64-i686-toolchain
> pacman -S --needed base-devel msys2-devel
> pacman -S libreadline-devel
> pacman -S mingw-w64-i686-libusb
> pacman -S mingw-w64-i686-libzip
> pacman -S mingw-w64-i686-libxml2
> pacman -S mingw-w64-i686-libssh2
> pacman -S mingw-w64-i686-curl
> pacman -S vim
> pacman -S git
> 
> 
> export PATH=/usr/bin:$PATH
> export CXX_PREFIX=${HOME}/mingw
> export PKG_CONFIG_PATH=${CXX_PREFIX}/lib/pkgconfig
> 
> 
> zlib
> ./configure --prefix=${CXX_PREFIX}
> 
> libzip
> cmake -DCMAKE_INSTALL_PREFIX=${CXX_PREFIX} .
> 
> openssl
> ./config --prefix=${CXX_PREFIX} shared no-asm enable-ssl2 enable-ssl3 enable-tls -fPIC -DOPENSSL_PIC enable-weak-ssl-ciphers
> 
> libssh2
> ./buildconf --prefix=${CXX_PREFIX}
> ./configure --prefix=${CXX_PREFIX}
> 
> curl
> ./buildconf --prefix=${CXX_PREFIX} --with-libssh2
> ./configure --prefix=${CXX_PREFIX} --with-libssh2
> 
> libplist
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=2.2.1
> 
> libimobiledevice-glue
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.0
> 
> libusbmuxd
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=2.0.3
> 
> libimobiledevice
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.3.1
> 
> libusb
> ./autogen.sh --prefix=${CXX_PREFIX} --disable-udev
> 
> libirecovery
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.1
> 
> libideviceactivation
> ./autogen.sh --prefix=${CXX_PREFIX} --without-cython PACKAGE_VERSION=1.1.2
> 
> idevicerestore
> ./autogen.sh --prefix=${CXX_PREFIX} PACKAGE_VERSION=1.0.1
> ```

**Thanks for the help I tried for several days to compile it but I couldn't. You can close the topic**

xanadu-lee, 2022-07-20:
If you need binary files, you can send your email.
Also, it should be up to you to close.

deyvs02, 2022-07-20:
> If you need binary files, you can send your email. Also, it should be up to you to close.

deyvs02@hotmail.com