libimobiledevice/idevicerestore #497: 32 bit cannot request memory.

xanadu-lee, 2022-06-18:
In restore_send_source_boot_object_v4, it is not possible to request more than 2GB of memory.
I wonder if it is possible to consider applying by segment?
Because the transfer is in 8192 bytes.

[image]
[image]

If the system does not allow memory allocation, extract_component fails.

xing181512, 2022-08-24:
I have the same problem, have you solved it?

xanadu-lee, 2022-08-24:
Solved by modifying the code.
When I decompressed at one time, I didn't actually decompress it, and then every time I sent it, I pressed the mark to decompress it.

nikias, 2022-08-24:
In the past we didn't have such large components so this wasn't a problem. Ideally the code needs to be changed to stream the data directly while decompressing from zip, or at least extract the image to disk first.

xanadu-lee, 2022-08-24:
@nikias 
I've solved it myself, and since I don't know the whole process well, I think you must have a more elegant solution.

closed 2022-08-24 by xanadu-lee