libimobiledevice/idevicerestore #498: make[2]: *** [idevicerestore-ipsw.o] Error 1

AldeabAhmed, 2022-06-19:
aldeab@Ahmeds-MBP idevicerestore % make             
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       idevicerestore-ipsw.o
ipsw.c:155:10: error: use of undeclared identifier 'FG_WHITE'
        cprintf(FG_WHITE "Product Version: " FG_BRIGHT_YELLOW "%s" COLOR...
                ^
ipsw.c:158:10: error: use of undeclared identifier 'FG_WHITE'
        cprintf(FG_WHITE "Supported Product Types:" COLOR_RESET);
                ^
ipsw.c:170:18: error: expected ')'
                                        cprintf(" " FG_BRIGHT_CYAN "%s" ...
                                                    ^
ipsw.c:170:13: note: to match this '('
                                        cprintf(" " FG_BRIGHT_CYAN "%s" ...
                                               ^
ipsw.c:179:10: error: use of undeclared identifier 'FG_WHITE'
        cprintf(FG_WHITE "Build Identities:" COLOR_RESET "\n");
                ^
ipsw.c:235:17: error: expected ')'
                        cprintf("  " FG_WHITE "[%d] Variant: " FG_BRIGHT...
                                     ^
ipsw.c:235:11: note: to match this '('
                        cprintf("  " FG_WHITE "[%d] Variant: " FG_BRIGHT...
                               ^
ipsw.c:288:29: error: expected ')'
                                        cprintf("    ChipID: " FG_GREEN ...
                                                               ^
ipsw.c:288:13: note: to match this '('
                                        cprintf("    ChipID: " FG_GREEN ...
                                               ^
ipsw.c:290:29: error: expected ')'
                                        cprintf("    ChipID: " FG_GREEN ...
                                                               ^
ipsw.c:290:13: note: to match this '('
                                        cprintf("    ChipID: " FG_GREEN ...
                                               ^
ipsw.c:1069:2: warning: 'SHA1_Init' is deprecated [-Wdeprecated-declarations]
        SHA1_Init(&sha1ctx);
        ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:49:1: note: 'SHA1_Init'
      has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Init(SHA_CTX *c);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1073:3: warning: 'SHA1_Update' is deprecated [-Wdeprecated-declarations]
                SHA1_Update(&sha1ctx, (const void*)buf, sz);
                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:50:1: note: 
      'SHA1_Update' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Update(SHA_CTX *c, const void *data, size_t len);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1075:2: warning: 'SHA1_Final' is deprecated [-Wdeprecated-declarations]
        SHA1_Final(tsha1, &sha1ctx);
        ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:51:1: note: 
      'SHA1_Final' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Final(unsigned char *md, SHA_CTX *c);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
3 warnings and 7 errors generated.
make[2]: *** [idevicerestore-ipsw.o] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
aldeab@Ahmeds-MBP idevicerestore %

AldeabAhmed, 2022-06-19:
When I build/make the source I get this error!

nikias, 2022-06-19:
You need latest [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue)

AldeabAhmed, 2022-06-19:
**NOW I got this error!**

aldeab@Ahmeds-MBP idevicerestore % make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CCLD     idevicerestore
ld: warning: directory not found for option '-L/usr/local/Cellar/libimobiledevice/1.3.0/lib'
Undefined symbols for architecture x86_64:
  "_plist_from_json", referenced from:
      _ipsw_get_signed_firmwares in idevicerestore-ipsw.o
  "_reverse_proxy_client_create_with_port", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_free", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_set_log_callback", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_set_status_callback", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_start_proxy", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_get_type", referenced from:
      _rp_log_cb in idevicerestore-restore.o
      _rp_status_cb in idevicerestore-restore.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [idevicerestore] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
aldeab@Ahmeds-MBP idevicerestore %

AldeabAhmed, 2022-06-19:
Configuration for idevicerestore 1.0.0-98-g3525150:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build idevicerestore 1.0.0-98-g3525150,
  and then 'make install' for installation.

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
  CC       idevicerestore-ftab.o
  CC       idevicerestore-ipsw.o
ipsw.c:1069:2: warning: 'SHA1_Init' is deprecated [-Wdeprecated-declarations]
        SHA1_Init(&sha1ctx);
        ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:49:1: note: 'SHA1_Init'
      has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Init(SHA_CTX *c);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1073:3: warning: 'SHA1_Update' is deprecated [-Wdeprecated-declarations]
                SHA1_Update(&sha1ctx, (const void*)buf, sz);
                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:50:1: note: 
      'SHA1_Update' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Update(SHA_CTX *c, const void *data, size_t len);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1075:2: warning: 'SHA1_Final' is deprecated [-Wdeprecated-declarations]
        SHA1_Final(tsha1, &sha1ctx);
        ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/sha.h:51:1: note: 
      'SHA1_Final' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Final(unsigned char *md, SHA_CTX *c);
^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:182:49: note: 
      expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/usr/local/Cellar/openssl@3/3.0.3/include/openssl/macros.h:62:52: note: 
      expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
3 warnings generated.
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
  CC       idevicerestore-recovery.o
  CC       idevicerestore-restore.o
  CC       idevicerestore-asr.o
[... 43 lines cut ...]
Password:
Making install in src
  CCLD     idevicerestore
ld: warning: directory not found for option '-L/usr/local/Cellar/libimobiledevice/1.3.0/lib'
Undefined symbols for architecture x86_64:
  "_plist_from_json", referenced from:
      _ipsw_get_signed_firmwares in idevicerestore-ipsw.o
  "_reverse_proxy_client_create_with_port", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_free", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_set_log_callback", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_set_status_callback", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_client_start_proxy", referenced from:
      _restore_device in idevicerestore-restore.o
  "_reverse_proxy_get_type", referenced from:
      _rp_log_cb in idevicerestore-restore.o
      _rp_status_cb in idevicerestore-restore.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [idevicerestore] Error 1
make: *** [install-recursive] Error 1
aldeab@Ahmeds-MBP idevicerestore %

AldeabAhmed, 2022-06-19:
What did I miss?
I've successfully installed the package, libimobiledevice-glue!

nikias, 2022-06-19:
You need latest libplist and libimobiledevice...

nikias, 2022-06-20:
Also beware that the latest idevicerestore code might be failing during restore, so you might want to move back 3 commits.

AldeabAhmed, 2022-06-20:
_**I think I finally succeeded!**_
Last login: Mon Jun 20 16:31:16 on ttys005
aldeab@Ahmeds-MBP ~ % idevicerestore -v
idevicerestore 1.0.0-98-g3525150
aldeab@Ahmeds-MBP ~ %