libimobiledevice/idevicerestore #506: not compatible with 16.0b2 ipsw's - error Cryptex1LocalPolicy

sitay1, 2022-06-23:
ERROR: restore_send_firmware_updater_data: Got unknown updater name 'Cryptex1LocalPolicy'.

`devicerestore 1.0.0-97-ge65d34b
Found device in Recovery mode
Identified device as d64ap, iPhone14,3
Extracting BuildManifest from IPSW
Product Version: 16.0
Product Build: 20A5303i Major: 20
INFO: device serial number is H9LLJ77F4G
Device supports Image4: true
Variant: Developer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Volumes/Samsung_T5/binaries/ipsw/16b2/iPhone14,3_16.0_20A5303i_Restore/078-13971-055.dmg'
Found ECID 5147742653087774
Getting ApNonce in recovery mode... c6 17 53 d8 da 60 73 f2 92 a7 48 16 7d d5 62 ad ec 93 74 d2 61 15 20 ce f6 ce 42 9d 6f f0 a8 03
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 62 3b 7b 7d 00 9b 9a e8 ad 7f 3e 77 34 73 f3 c0 39 71 d0 42
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d64.RELEASE.im4p (Firmware/dfu/iBEC.d64.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (986045 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-8419.0.79.0.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2778~iphone.im4p (Firmware/all_flash/applelogo@2778~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29366 bytes)...
Extracting h14_ane_fw_bia_d6x.im4p (Firmware/ane/h14_ane_fw_bia_d6x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (5043819 bytes)...
Extracting aopfw-iphone14aop.im4p (Firmware/AOP/aopfw-iphone14aop.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (2492379 bytes)...
Extracting AppleAVE2FW_H14.im4p (Firmware/ave/AppleAVE2FW_H14.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1354115 bytes)...
Extracting armfw_g14p.im4p (Firmware/agx/armfw_g14p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (1761211 bytes)...
Extracting adc-kratos-d6x.im4p (Firmware/isp_bni/adc-kratos-d6x.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (24697710 bytes)...
Extracting D64_MtpFirmware.im4p (Firmware/D64_MtpFirmware.im4p)...
Personalizing IMG4 component MtpFirmware...
Sending MtpFirmware (3213154 bytes)...
Extracting t8110pmp.im4p (Firmware/pmp/t8110pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (530395 bytes)...
Extracting rans.t8110.release.im4p (Firmware/rans.t8110.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (989720 bytes)...
Extracting iphone14d6xdcp.im4p (Firmware/dcp/iphone14d6xdcp.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (4597723 bytes)...
Extracting 078-13514-089.dmg.trustcache (Firmware/078-13514-089.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (16054 bytes)...
Extracting SmartIOFirmware_ASCv5.im4p (Firmware/SmartIOFirmware_ASCv5.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1163471 bytes)...
Extracting WirelessPower.iphone14.im4p (Firmware/WirelessPower/WirelessPower.iphone14.im4p)...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (202397 bytes)...
Extracting iBootData.d64.RELEASE.im4p (Firmware/all_flash/iBootData.d64.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
[... 169 lines cut ...]
Checkpoint 4294968986 complete with code 0
Checkpoint 4294968987 complete with code 0
Checkpoint 1618 complete with code 0
About to send BuildIdentity Dict...
common.c:printing 0 bytes plist:
(null)Sending BuildIdentityDict now...
Done sending BuildIdentityDict
About to send BuildIdentity Dict...
common.c:printing 0 bytes plist:
(null)Sending BuildIdentityDict now...
Done sending BuildIdentityDict
Sending Cryptex1 TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Cryptex1,Ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
About to send BuildIdentity Dict...
common.c:printing 0 bytes plist:
(null)Sending BuildIdentityDict now...
Done sending BuildIdentityDict
ERROR: restore_send_firmware_updater_data: Got unknown updater name 'Cryptex1LocalPolicy'.
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device`

Lethaltail, 2022-06-24:
Should now be working with #507.

Lethaltail, 2022-07-06:
Fixes are still looking good on Beta 3!

nikias, 2022-07-06:
Thanks for confirming!

sitay1, 2022-07-07:
The fixes you guys did seems to fix that issue. Thanks. Closing it.

closed 2022-07-07 by sitay1

WilsontheWolf, 2022-07-13:
I am having a similar issue, restoring the 16.0 Beta 3 Update. https://haste.antti.codes/iyihomofup.yaml

WilsontheWolf, 2022-07-13:
Ok so I reverted back to commit eff9fab and it worked, so there appears to be a regression somewhere