libimobiledevice/idevicerestore #508: Segmentation fault in latest build

lolman9998, 2022-06-24:
Quick notes - Arch linux with latest kernel, AUR version of idevicerestore, device is an iPad mini 1st gen with cellular (A1455). In order for this device to be detected I had to connect to a rear USB port on my desktop. I will be omitting parts of the log tagged with key, as these might be something I shouldn't share, and these don't seem to be a part of the issue. Let me know if the full log is necessary.

```
irecv_event_cb: device 00000244e119bdf2 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:AD9F869900D2D2B09D44A3C3AB7AC7D65DB0BE94
found device with ECID 00000244e119bdf2
Setting to configuration 1
Setting to interface 0:0
Identified device as p107ap, iPad2,7
*   Trying 172.67.83.193:443...
* Connected to api.ipsw.me (172.67.83.193) port 443 (#0)
* ALPN: offers h2
* ALPN: offers http/1.1
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN: server accepted h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=Cloudflare, Inc.; CN=sni.cloudflaressl.com
*  start date: May 31 00:00:00 2022 GMT
*  expire date: May 31 23:59:59 2023 GMT
*  subjectAltName: host "api.ipsw.me" matched cert's "*.ipsw.me"
*  issuer: C=US; O=Cloudflare, Inc.; CN=Cloudflare Inc ECC CA-3
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* Using HTTP2, server supports multiplexing
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* h2h3 [:method: GET]
* h2h3 [:path: /v3/device/iPad2,7]
* h2h3 [:scheme: https]
* h2h3 [:authority: api.ipsw.me]
* h2h3 [user-agent: InetURL/1.0 idevicerestore/1.0.0]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0x559e2902b8f0)
> GET /v3/device/iPad2,7 HTTP/2
Host: api.ipsw.me
user-agent: InetURL/1.0 idevicerestore/1.0.0
accept: */*

* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 256)!
< HTTP/2 200 
< date: Fri, 24 Jun 2022 05:10:13 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-origin: *
< access-control-expose-headers: X-Data-Location, X-Server-Location
< cache-control: public
< content-md5: HUwIglkR1vvGoY8g3/C+rw==
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Fri, 24 Jun 2022 05:20:24 GMT
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-server-location: 278e4565bea8
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< last-modified: Fri, 24 Jun 2022 04:20:24 GMT
< cf-cache-status: HIT
< age: 212
< expect-ct: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=PsYWc8q6RU1vE6WdduXFH3g64WY7pb2vPJCvXycU80Dsix3BTgPUiVasFMkgSNA2Jfcwim%2BlREl4rGXusgr0FNGWm6TcPwJbfWAX8R%2FJBBmBRiKwFlnr%2BBY7tNpT"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 7202f4aaf8b68720-ORD
< alt-svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400
< 
* Connection #0 to host api.ipsw.me left intact
The following firmwares are currently being signed for iPad2,7:
  [1] 9.3.6 (build 13G37)
  [2] 9.3.5 (build 13G36)
Select the firmware you want to restore: 1
Selected firmware 9.3.6 (build 13G37)
Verifying 'iPad2,7_9.3.6_13G37_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.6
Product Build: 13G37 Major: 13
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:AD9F869900D2D2B09D44A3C3AB7AC7D65DB0BE94
found device with ECID 00000244e119bdf2
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is F9FPF174F19M
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPad2,7_9.3.6_13G37_Restore/058-48288-040.dmg'
Getting device's ECID for TSS request
Found ECID 2494857592306
Getting ApNonce in recovery mode... ad 9f 86 99 00 d2 d2 b0 9d 44 a3 c3 ab 7a c7 d6 5d b0 be 94 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Applying restore request rules for entry LLB
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Applying restore request rules for entry iBEC
DEBUG: Applying restore request rules for entry iBSS
DEBUG: Applying restore request rules for entry iBoot
common.c:printing 3780 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
```

--omitted--

```
Received SHSH blobs
Sending APTicket (2732 bytes)
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.p107.RELEASE.dfu...
DEBUG: tss_response_get_data_by_key: No entry 'ApImg4Ticket' in TSS response
DEBUG: tss_response_get_blob_by_entry: No entry 'iBEC' in TSS response
NOTE: No SHSH blob found for component iBEC
Not personalizing component iBEC...
Sending iBEC (283020 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 00000244e119bdf2 (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 00000244e119bdf2 (udid: N/A) connected in Recovery mode
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:AD9F869900D2D2B09D44A3C3AB7AC7D65DB0BE94
found device with ECID 00000244e119bdf2
Setting to configuration 1
Setting to interface 0:0
Segmentation fault
```

 I have tried with and without the arguments -d -e and -l, same result either way.

nikias, 2022-06-27:
Any chance you compile the latest code and if the error persists run this in `gdb` or `lldb` and get a backtrace once it segfaults?

tcs320, 2022-11-15:
Hi, I have the same Problem my last line are
Updating baseband (19)
About to send BasebandData...
sending request without baseband nonce
Sending Baseband TSS request...
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
Segmentation fault (core dumped)

Is there something I can try?

nikias, 2022-11-15:
Can you run in `gdb` (or `lldb`) and type `bt` once it segfaults?

tcs320, 2022-11-15:
Hi, 
yes I can Test and send you the results but Im not a developer I don't unterstand what the means what you wrote.
If you can describe what you need and what I have to do exactly I can do it.

I download gdb and read some explanations but I not sure what I should do with it.

nikias, 2022-11-15:
Ah ok so the command line would be:
```bash
gdb idevicerestore
```
Wait for it to print the initial gdb stuff
Then in the prompt write
```bash
(gdb) run --erase --latest
```
Now wait for the crash, then type `bt`:
```bash
(gdb) bt
```
That will print debug info that hopefully point out where  and why it crashes.

Thanks for trying to solve this issue.

tcs320, 2022-11-15:
Hahah no thank you for build such a great tool. 
I think it is not the fault from your tool.

I think my Phone is just that bad broken.
I hope that helps you....

`Sending Baseband TSS request...
[New Thread 0x7fffdffff640 (LWP 2062)]
[Thread 0x7fffdffff640 (LWP 2062) exited]
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs

Thread 1 "idevicerestore" received signal SIGSEGV, Segmentation fault.
0x00007ffff7893977 in __GI_fseek (fp=fp@entry=0x0, offset=offset@entry=0, whence=whence@entry=2) at ./libio/fseek.c:35
35      ./libio/fseek.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7893977 in __GI_fseek (fp=fp@entry=0x0, offset=offset@entry=0, whence=whence@entry=2) at ./libio/fseek.c:35
#1  0x00005555555731ff in restore_send_baseband_data (restore=restore@entry=0x55555588e5f0, client=client@entry=0x5555555c6280, 
    build_identity=build_identity@entry=0x5555556a3c40, message=message@entry=0x555555883c50) at restore.c:1901
#2  0x000055555557b93b in restore_handle_data_request_msg (client=client@entry=0x5555555c6280, device=device@entry=0x5555558ab8a0, 
    restore=restore@entry=0x55555588e5f0, message=0x555555883c50, build_identity=build_identity@entry=0x5555556a3c40, 
    filesystem=filesystem@entry=0x5555557ec280 "iPhone13,1_16.1_20B82_Restore/078-65626-104.dmg") at restore.c:3991
#3  0x000055555557d116 in restore_device (client=client@entry=0x5555555c6280, build_identity=build_identity@entry=0x5555556a3c40, 
    filesystem=filesystem@entry=0x5555557ec280 "iPhone13,1_16.1_20B82_Restore/078-65626-104.dmg") at restore.c:4496
#4  0x00005555555611b5 in idevicerestore_start (client=client@entry=0x5555555c6280) at idevicerestore.c:1417
#5  0x000055555555c293 in main (argc=<optimized out>, argv=<optimized out>) at idevicerestore.c:1817
(gdb) `

nikias, 2022-11-16:
OK so I have some kind of idea where it happens now, and also why (invalid file handle) but I can't really figure out what is causing this. It seems like when opening the baseband firmware file (ZIP) it somehow fails.
Anyway, this looks like the Ubuntu 22.04 package version; any chance you compile the latest code from source yourself?

nikias, 2022-11-16:
Also just to make it clear it's a different location than the originally posted issue.

tcs320, 2022-11-16:
And Again I don't really know what you mean by that. I used the direction on the GitHub Page to install it. It mostly worked as you described it there. I had problems installing libimobiledevice-glue.

I can install it again differently but you need to explain how :-).
Or of I should use a different operating system

tcs320, 2022-11-16:
So... I just startet over with Debian and I got this errors during the installation.

1. E: Unable to locate package libimobiledevice-glue-dev
Fixed by downloading and installing https://github.com/libimobiledevice/libimobiledevice-glue

2. configure: error: libplist with JSON format support required to build idevicerestore
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.

Fix by installing https://github.com/libimobiledevice/libplist and python3-dev

3. idevicerestore: symbol lookup error: idevicerestore: undefined symbol: plist_from_json
No Fix this error is new I didnt get this one on Ubuntu.

nikias, 2022-11-16:
Try my script here:
https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh?dl=0

tcs320, 2022-11-16:
Hi that didn't work as well. 

1 error 
root@test:~# apt install cython
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package cython is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'cython' has no installation candidate

2. 
configure: error: Package requirements (libirecovery-1.0 >= 1.0.1) were not met:

No package 'libirecovery-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

nikias, 2022-11-17:
Feel free to reach out to me over telegram, find me in the https:/t.me/libimobiledevice channel and drop me a PM.

sinetek, 2024-02-07:
Any update on this issue? Still happens on master.

nikias, 2024-02-08:
Can you get me an updated backtrace with the latest build? I want to understand where exactly in the code it fails. I don't see a seek/fseek in that function...

sinetek, 2024-02-08:
> Can you get me an updated backtrace with the latest build? I want to understand where exactly in the code it fails. I don't see a seek/fseek in that function...

well i don't want to screw up my phone again, but it was exactly like the trace in the picture above, including a call to GI_FSEEK