libimobiledevice/idevicerestore #509: Not compatible with iOS 15.5, iPhone 11 Pro Bootloop

Paralelzz, 2022-06-25:
Hello, i have an iPhone 11 Pro, and i just found out that **the baseband is damaged**, so i used **iDeviceRestore** to exclude BB/NOR updates.
Problem is, it won't take my official IPSW, can anyone help? Here are the logs

luixiuno@MBP-de-usuario ~ % idevicerestore -x /Users/luixiuno/Downloads/iPhone11\,2\,iPhone11\,4\,iPhone11\,6\,iPhone12\,3\,iPhone12\,5_15.5_19F77_Restore.ipsw 
Found device in Recovery mode
Identified device as d421ap, iPhone12,3
Extracting BuildManifest from IPSW
Product Version: 15.5
Product Build: 19F77 Major: 19
INFO: device serial number is C39ZJ09ZN6Y4
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 078-12678-105.dmg
[=========                                         ]  16.3%ERROR: frite: /Users/luixiuno/Downloads/iPhone11,2,iPhone11,4,iPhone11,6,iPhone12,3,iPhone12,5_15.5_19F77_Restore/078-12678-105.dmg.extract
ERROR: Unable to extract filesystem from IPSW
Removing /Users/luixiuno/Downloads/iPhone11,2,iPhone11,4,iPhone11,6,iPhone12,3,iPhone12,5_15.5_19F77_Restore/078-12678-105.dmg.extract

nikias, 2022-06-25:
-x option doesn't work with recent devices. Your problem seems to be that the extraction of the ipsw doesn't work; it fails to write a block to the extracted file. Do you have enough disk space?
You can also manually extract the ipsw and pass the directory where you extracted it as "ipsw filename" to idevicerestore.
In any case that doesn't mean it's not compatible as you mentioned, since it didn't even try to restore.
Regarding the baseband issue, I think you might have to apply this patch for it to work: https://github.com/libimobiledevice/idevicerestore/pull/504

Paralelzz, 2022-06-25:
Thanks, my disk HAS been filling up a lot, i will try to fix it.

nikias, 2022-06-27:
Instead of the patch I mentioned, you can also just use `--ignore-errors` command line option.

deyvs02, 2022-07-04:
> Hello, i have an iPhone 11 Pro, and i just found out that **the baseband is damaged**, so i used **iDeviceRestore** to exclude BB/NOR updates. Problem is, it won't take my official IPSW, can anyone [...]

Please can you compile it for windows and send me the executables, I have them but with release year 2019, it doesn't work for our ios.