libimobiledevice/idevicerestore #511: TSS request fails on signed firmware

kokoscript, 2022-06-26:
I'm having a bit of trouble trying to restore an iPhone 12 using any of the currently signed IPSWs; in other words, 15.5, 15.6b3, and 16b2 all fail in the same way:

15.5:
```
% ./idevicerestore "/home/kokoro/iPhone13,2,iPhone13,3_15.5_19F77_Restore.ipsw"
idevicerestore 1.0.0-100-gaa25a29
Found device in Recovery mode
ECID: 2556605611769886
Identified device as d53gap, iPhone13,2
Extracting BuildManifest from IPSW
Product Version: 15.5
Product Build: 19F77 Major: 19
INFO: device serial number is ------------
Device supports Image4: true
Variant: Research Customer Upgrade Install (IPSW)
ERROR: Unable to find RestoreBehavior node
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 078-12547-101.dmg
[==================================================] 100.0%
Getting ApNonce in recovery mode... 2f ed 5a e8 dc eb 6d 8f cf 82 65 06 b5 9d 1d 3a 68 26 cf b8 41 ae ba 26 85 18 f6 fc af de ae 42 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... f6 33 87 96 bd fa 71 27 6e 26 dd ba 1d e2 42 a6 52 ba 7a bd 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

15.6b3:
```
% ./idevicerestore "/home/kokoro/iPhone13,2,iPhone13,3_15.6_19G5046d_Restore.ipsw"
idevicerestore 1.0.0-100-gaa25a29
Found device in Recovery mode
ECID: 2556605611769886
Identified device as d53gap, iPhone13,2
Extracting BuildManifest from IPSW
Product Version: 15.6
Product Build: 19G5046d Major: 19
INFO: device serial number is ------------
Device supports Image4: true
Variant: Research Developer Upgrade Install (IPSW)
ERROR: Unable to find RestoreBehavior node
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 078-39767-021.dmg
[==================================================] 100.0%
Getting ApNonce in recovery mode... 2f ed 5a e8 dc eb 6d 8f cf 82 65 06 b5 9d 1d 3a 68 26 cf b8 41 ae ba 26 85 18 f6 fc af de ae 42 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... f6 33 87 96 bd fa 71 27 6e 26 dd ba 1d e2 42 a6 52 ba 7a bd 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

16b2:
```
% ./idevicerestore "/home/kokoro/iPhone13,2,iPhone13,3_16.0_20A5303i_Restore.ipsw" 
idevicerestore 1.0.0-100-gaa25a29
Found device in Recovery mode
ECID: 2556605611769886
Identified device as d53gap, iPhone13,2
Extracting BuildManifest from IPSW
Product Version: 16.0
Product Build: 20A5303i Major: 20
INFO: device serial number is ------------
Device supports Image4: true
Variant: Research Developer Upgrade Install (IPSW)
ERROR: Unable to find RestoreBehavior node
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 078-13983-056.dmg
[==================================================] 100.0%
Getting ApNonce in recovery mode... 84 22 53 03 83 ee b2 6c c1 49 80 ff 5b 4e 66 ef 34 1d 0c 94 5a 2d c4 07 94 60 ee ed f1 29 d0 a0 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 7a 20 e3 e1 6f 76 9f 1b 3a 84 d8 fa 5a a1 f8 65 15 33 b0 34 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

This is on Debian 11 with all of the libimobiledevice components compiled from source. Some Macs on the same network were able to install 15.5 and 15.6b3 to the device using the same IPSWs.

Any help is appreciated.

nikias, 2022-06-27:
Can you check if it works with latest code? There was a small change with commit eff9fab5db946903ae8706d7ec6809176c7ca8b5 that might just fix this.

kokoscript, 2022-06-27:
I actually applied that commit to my local copy before it was merged in; unfortunately there's no change with or without it.

nikias, 2022-06-27:
Yes I just realized I misread your output, it fails for you in a pretty early stage. Can you run it with -d ?

kokoscript, 2022-06-27:
Sure thing:
```
% ./idevicerestore -d "/home/kokoro/iPhone13,2,iPhone13,3_16.0_20A5303i_Restore.ipsw" 
idevicerestore 1.0.0-100-gaa25a29
irecv_event_cb: device 000915382152001e (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 000915382152001e
Setting to configuration 1
Setting to interface 0:0
ECID: 2556605611769886
Identified device as d53gap, iPhone13,2
Extracting BuildManifest from IPSW
Product Version: 16.0
Product Build: 20A5303i Major: 20
opening device 05ac:1281...
found device with ECID 000915382152001e
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is ------------
Device supports Image4: true
Variant: Research Developer Upgrade Install (IPSW)
ERROR: Unable to find RestoreBehavior node
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/kokoro/iPhone13,2,iPhone13,3_16.0_20A5303i_Restore/078-13983-056.dmg'
Getting ApNonce in recovery mode... ec 54 29 03 d6 5c cd 31 68 5a 7a b0 a7 8b 74 77 c3 64 69 4a 64 38 16 92 d5 59 b6 99 7b 46 6f 9f 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 5e 0a 61 4d 44 35 09 21 52 58 02 85 4d 2f 6b 87 05 32 24 4c 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry ANE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry ANS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AVE
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,HapticAssets
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AudioCodecFirmware
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,DigestMap' as it doesn't have RestoreRequestRules
DEBUG: tss_request_add_ap_tags: Skipping 'BMU,FirmwareMap' as it doesn't have RestoreRequestRules
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
[... 1000 lines cut ...]
	7FQpA9ZczTFoWnqwp4t0d8NkaUpkOBaS1Vm2mXtGb58=
	</data>
	<key>@ApImg4Ticket</key>
	<true/>
	<key>ApSecurityMode</key>
	<true/>
	<key>ApProductionMode</key>
	<true/>
	<key>SepNonce</key>
	<data>
	XgphTUQ1CSFSWAKFTS9rhwUyJEw=
	</data>
	<key>PearlCertificationRootPub</key>
	<data>
	BEBGXhKwc7q3iFvkUoGDP6j2drpxSCxsSCODaDQIqGwd53wZJ0xIJIv0RTf2TS7+/u4K
	zhrANzb19r+TQzwqFJMphp3mI3yY4pukIFc/kWS7DLQAx/ftWBXX6vl4ig3wEg==
	</data>
</dict>
</plist>
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

nikias, 2022-06-27:
I see what happens here; it tries to use the "Research *" variants when trying to determine the build identity to use. I committed a few changes, it should ignore those now and pick the right one.

kokoscript, 2022-06-27:
That did the trick! Thank you so much!

closed 2022-06-27 by kokoscript