libimobiledevice/idevicerestore #512: Device restore fails after placing ipad into recovery mode

MorticiaGrey, 2022-06-26:
(On manjaro linux, recently updated ipad)
After executing the command `idevicerestore --latest` I get this output:

```
idevicerestore 1.0.0-100-gaa25a29
Found device in Normal mode
ECID: 6255448278695974
Identified device as j71bap, iPad7,5
The following firmwares are currently being signed for iPad7,5:
  [1] 15.5 (build 19F77)
Select the firmware you want to restore: 1
Selected firmware 15.5 (build 19F77)
Verifying 'iPad_64bit_TouchID_ASTC_15.5_19F77_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 15.5
Product Build: 19F77 Major: 19
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPad_64bit_TouchID_ASTC_15.5_19F77_Restore/078-12488-106.dmg'
Checking if device requires stashbag...
Device requires stashbag.
******************************************************************************
* Please enter your passcode on the device.  The device will store a token   *
* that will be used after restore to access the user data partition.  This   *
* prevents an 'Attempting data recovery' process occurring after reboot that *
* may take a long time to complete and will _also_ require the passcode.     *
******************************************************************************
Stashbag created.
Getting ApNonce in normal mode... 30 2f c4 02 af 00 23 c4 9a 9d 2e c6 b2 d9 41 0b 8a 13 d2 e3 ac 16 d3 bb 6b 99 9e a4 27 32 5e 8b 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 17 92 b8 be 9f 27 6e 83 2e f0 a1 83 5c 27 73 d2 a1 67 ea 5e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Committing stashbag...
Stashbag committed!
Entering recovery mode...
ERROR: Failed to enter recovery mode
ERROR: Unable to place device into recovery mode from normal mode
```
My ipad is in recovery mode, and I don't own a mac or windows computer to use itunes to fix it with. On repeated attempts `idevicerestore` gives the error `ERROR: Unable to discover device mode. Please make sure a device is attached.`

nikias, 2022-06-26:
Your problem might be that the recovery USB device file is not accessible by the user you are running idevicerestore with. Usually libirecovery installs a udev rule to make sure normal users have r/w access to the USB device files. Even though not recommended, you can try to run idevicerestore as root to see if that fixes the problem. The better way to fix this would be to make sure the udev rule is properly working, and the current user is in the group that the rule will set the permissions to.
Maybe you can check what `/lib/udev/rules.d/39-libirecovery.rules` specifies, and then check with `groups` if you user is part of the mentioned group (usually `plugdev` or similar).

MorticiaGrey, 2022-06-26:
I ran as root and it made it past the segment that is was getting stuck, it's still working but it looks like that worked

closed 2022-06-26 by MorticiaGrey