libimobiledevice/idevicerestore #513: Iphone4S GasGauge Error

ElectroBoy404NotFound, 2022-06-28:
Hello, I have an iPhone4S with a broken battery, so I replaced it with a third party one and now im unable to restore it. I even threw the old away! So any help will be nice! Also, as it does not have any exploit to put unauthorised software, i cant edit the ipsw and remove the gasgauge.

(I'm using ubuntu 20.04 and edited the src to disable the saved log)

Things i tried:
I tried `--ignore-errors` but it just told the same for 2 days straight!

log:
```
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev/disk0s1s2 baseband data= log=
entering mount_filesystems
entering mount_partition
executing /sbin/fsck_hfs
journal_replay(/dev/disk0s1s1) returned 0
** /dev/rdisk0s1s1
        Cache size should be multiple of cache block size (currently 32K)
        Using cacheBlockSize=32K cacheTotalBlock=2019 cacheSize=64608K.
   Executing fsck_hfs (version hfs-305.10.1).
** Checking Journaled HFS Plus volume.
** Detected a case-sensitive volume.
   The volume name is Genoa13G37.N94OS
** Checking extents overflow file.
** Checking catalog file.
** Checking multi-linked files.
** Checking catalog hierarchy.
** Checking extended attributes file.
** Checking volume bitmap.
** Checking volume information.
** Trimming unused blocks.
** The volume Genoa13G37.N94OS appears to be OK.
        CheckHFS returned 0, fsmodified = 0
executing /sbin/mount_hfs
/dev/disk0s1s1 mounted on /mnt1
System mounted read-write
entering mount_partition
executing /sbin/fsck_hfs
journal_replay(/dev/disk0s1s2) returned 0
** /dev/rdisk0s1s2
        Cache size should be multiple of cache block size (currently 32K)
        Using cacheBlockSize=32K cacheTotalBlock=2019 cacheSize=64608K.
   Executing fsck_hfs (version hfs-305.10.1).
** Checking Journaled HFS Plus volume.
** Detected a case-sensitive volume.
   The volume name is Data
** Checking extents overflow file.
** Checking catalog file.
** Checking multi-linked files.
** Checking catalog hierarchy.
** Checking extended attributes file.
** Checking volume bitmap.
** Checking volume information.
** Trimming unused blocks.
** The volume Data appears to be OK.
        CheckHFS returned 0, fsmodified = 0
executing /sbin/mount_hfs
mount_hfs: Could not exec re-keying daemon /usr/libexec/rolld: No such file or directory
/dev/disk0s1s2 mounted on /mnt2
Data mounted read-write
RestoredFDRRestoreSavedData success
entering install_kernel_cache
writing kernelcache to /mnt1/System/Library/Caches/com.apple.kernelcaches/kernelcache
entering install_secure_vault_daemon
entering get_hardware_platform
platform-name = s5l8940x
linking /mnt1/usr/sbin/securekeyvaultd to /mnt1/usr/sbin/securekeyvaultd.s5l8940x
no securekeyvaultd for s5l8940x
entering ramrod_ticket_write
writing ticket to /mnt1/System/Library/Caches/apticket.der
entering RestoredFDRRecover
entering update_device_firmware
entering add_baseband_partition_options
entering ramrod_update_device_firmware
entering update_iBoot
found 'AppleImage3NORAccess' object in IORegistry
entering update_iBoot_image3_NOR_access
write_image3_data: flashing illb data (length = 0x24a1a)
SYSLOG: Jan  1 00:07:34 localhost kernel[0] <Notice>: AppleImage3NORAccess::_getSuperBlock imageVersion: 3
SYSLOG: Jan  1 00:07:34 localhost kernel[0] <Notice>: AppleImage3NORAccess::_getSuperBlock imageVersion: 3
write_image3_data: flashing SCAB data (length = 0xaf0)
ticket updated
write_image3_data: flashing ibot data (length = 0x4418c)
write_image3_data: flashing recm data (length = 0x143cc)
write_image3_data: flashing logo data (length = 0x1a8c)
write_image3_data: flashing bat0 data (length = 0xaa4c)
write_image3_data: flashing chg1 data (length = 0x88cc)
write_image3_data: flashing batF data (length = 0x1484c)
write_image3_data: flashing dtre data (length = 0x1464c)
write_image3_data: flashing chg0 data (length = 0x260c)
write_image3_data: flashing bat1 data (length = 0x80c)
write_image3_data: flashing glyP data (length = 0x2e4c)
SYSLOG: Jan  1 00:07:34 localhost kernel[0] <Notice>: [nand_part_core:INF@2297] first 2 pages of boot blocks written with copies of partition table
SYSLOG: Jan  1 00:07:36 localhost kernel[0] <Notice>: [nand_part_core:INF@3177] next 98 pages of boot blocks written with LLB image of size 150080 bytes
SYSLOG: Jan  1 00:07:36 localhost kernel[0] <Notice>: [nand_part_core:INF@3180] final 156 pages of boot blocks written with random data in full page format
entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -1)
Update[Gas Gauge] failed - 2 retries remaining...
entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -1)
Update[Gas Gauge] failed - 1 retries remaining...
entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -1)
Update[Gas Gauge] failed - 0 retries remaining...
0: RamrodErrorDomain/3ea: update_gas_gauge: gasgauge_swupdate returned -1

ERROR: Unable to successfully restore device
```

nikias, 2022-06-28:
I am not sure this error can be ignored since apparently `--ignore-errors` didn't help. Even though I don't think this will give any different results, can you get a full debug log output by running with `-d`, maybe there is some error condition where the code breaks out of the loop.

ElectroBoy404NotFound, 2022-06-28:
> I am not sure this error can be ignored since apparently `--ignore-errors` didn't help. Even though I don't think this will give any different results, can you get a full debug log output by running with `-d`, maybe there is some error condition where the code breaks out of the loop.

I even tried that, edited the code to print every state, but after gasgauge update, it does absulutly nothing.

ElectroBoy404NotFound, 2022-06-28:
Like, it is only getting that error fro the iphone

ElectroBoy404NotFound, 2022-07-01:
@nikias

ProRe1xxmu, 2024-08-20:
tried eighth --ignore-errors and -d flag 
`entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -4)
Update[Gas Gauge] failed - 2 retries remaining...
entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -4)
Update[Gas Gauge] failed - 1 retries remaining...
entering update_gas_gauge
Update[Gas Gauge] error - The operation couldn\U2019t be completed. (RamrodErrorDomain error 1002 - update_gas_gauge: gasgauge_swupdate returned -4)
Update[Gas Gauge] failed - 0 retries remaining...
0: RamrodErrorDomain/3ea: update_gas_gauge: gasgauge_swupdate returned -4

WARNING: Attempting to continue after critical error, restore might fail...
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
No data to read (timeout)
`
and more and more and more....
and where is sw_gas_gauge method located? i need it to see what -4 or -1 error code means 
what is going after gas_gauge update (in successful case)? it tries to read smth, but what exactly...