libimobiledevice/idevicerestore #516: iPhone 13 series fails to enter restore mode on iOS 15.6

a-gunter, 2022-07-22:
As far as I know this only happens on the 13 series devices (mini, regular, pro, pro max). iOS 15.5 is fine, but something is new on iOS 15.6. When restoring from recovery mode, after sending iBoot components -> DeviceTree -> RestoreSEP -> KernelCache, the device fails to transition to restore mode and instead boots back into recovery.

This was done on macOS using Apple's usbmuxd.

Debug log is attached; if needed, I also have pcaps of an unsuccessful idevicerestore attempt and a successful Apple Configurator 2 attempt (stopped the capture once the device transitioned to restore mode). They're uploaded [here](https://drive.google.com/drive/folders/1xYmAA2UKqxxz-N9nnl455rD9ikVFDwCC?usp=sharing).

idevicerestore debug log:
[13ProMax_15.6_idevicerestore_debug.txt](https://github.com/libimobiledevice/idevicerestore/files/9169596/13ProMax_15.6_idevicerestore_debug.txt)

Anyone else seeing the same behavior?

ashishgah, 2022-07-25:
yes, idevicerestore is not working for 13 series devices. i am also facing same issue.

kindly reply, if anyone has resolve this issue.

truonggiang0710, 2022-07-26:
Me too. @nikias do you have any ideas?

truonggiang0710, 2022-07-27:
@a-gunter @ashishgah Do you have any update?

Naz4h, 2022-07-27:
I have the same problem. Tried on an iPhone 13 mini (iPhone14,4), both when starting in normal or recovery mode. "ERROR: Device failed to enter restore mode." appears. 
This error appears with the last version of libimobile and idevicerestore (I compiled it on 12th July).
The log is attached.
[idevicerestore_iPhone14,4.txt](https://github.com/libimobiledevice/idevicerestore/files/9199850/idevicerestore_iPhone14.4.txt)
.

xing181512, 2022-08-02:
Me too，if anyone has resolve this issue.

mroger-tech, 2022-08-03:
Me too,if anyone has resolve this issue.

pod2g, 2022-08-04:
Hi here. I got it to work by adding the following to line 670 of `src/tss.c`.

```
if (_plist_dict_get_bool(info_dict, "IsFTAB")) {
    debug("DEBUG: %s: Skipping '%s' because IsFTAB\n", __func__, key);
    continue;
}
```

I haven't submitted a patch or pull request because I am not sure of all the implications, so I let the expert @nikias to double check!

Thanks my friend!

Naz4h, 2022-08-04:
I tried it and it works @pod2g !

Thank you so much !!!

xanadu-lee, 2022-08-05:
very effective, thank you @pod2g

ashishgah, 2022-08-05:
yes, it is working. great.. thank you @pod2g .

mroger-tech, 2022-08-05:
it  is working,and I'm curious how did you find this method.

pod2g, 2022-08-05:
> I tried it and it works @pod2g !
> 
> 
> 
> Just for curiosity how do you find that ?
> 
> 
> 
> Thank you so much !!!

Hey.

idevicerestore does not seem to check the return code of the usb control request for the "firmware" iBEC command following the upload of the Restore ANS firmware component (or any command actually); but it is failing.

I captured the USB communication of iTunes with the device and compared the Restore ANS payload of iTunes vs idevicerestore.

Only difference was about the manifest of idevicerestore having additional components.

These components had the aforementioned property in the BuildManifest.plist .

I tried to exclude them from the TSS request and that worked.

Greetings.

doronz88, 2022-08-21:
Did anyone try that on different devices? (other than iPhone 13)

truonggiang0710, 2022-08-22:
@doronz88 I try all the iPhone X and above, it's OK.

nikias, 2022-08-23:
Committed with 4a516e5f5e0c509de966cc1ac45622f452f23e0b.

closed 2022-08-23 by a-gunter

McVICTORY, 2023-04-10:
> > I tried it and it works @pod2g !
> > Just for curiosity how do you find that ?
> > Thank you so much !!!
> 
> Hey.
> 
> idevicerestore does not seem to check the return code of the usb control request for the "firmware" iBEC command following the upload of the Restore ANS firmware component (or any command actually); but it was failing.
> 
> I captured the USB communication of iTunes with the device and compared the Restore ANS payload of iTunes vs idevicerestore.
> 
> Only difference was about the manifest of idevicerestore having additional components.
> 
> These components had the aforementioned property in the BuildManifest.plist .
> 
> I tried to exclude them from the TSS request and that worked.
> 
> Greetings.

> Hi here. I got it to work by adding the following to line 670 of `src/tss.c`.
> 
> ```
> if (_plist_dict_get_bool(info_dict, "IsFTAB")) {
>     debug("DEBUG: %s: Skipping '%s' because IsFTAB\n", __func__, key);
>     continue;
> }
> ```
> 
> I haven't submitted a patch or pull request because I am not sure of all the implications, so I let the expert @nikias double check!
> 
> Thanks my friend!

how can I apply this, having same issue with an iphone 13 pro max, not entering DFU and stuck at recovery screen

jiepbhaone, 2023-07-31:
> Hi here. I got it to work by adding the following to line 670 of `src/tss.c`.
>
> ```
> if (_plist_dict_get_bool(info_dict, "IsFTAB")) {
>     debug("DEBUG: %s: Skipping '%s' because IsFTAB\n", __func__, [...]

im sorry im new to this and i dont really understand where should add this. Could someone help pls

AiXanadu, 2023-08-04:
`
		// Compared to ac2, not needed
		if ((strcmp(key, "BaseSystem") == 0)) {
			continue;
		}

		/* FIXME: only used with diagnostics firmware */
		if (strcmp(key, "Diags") == 0) {
			continue;
		}

		plist_t info_dict = plist_dict_get_item(manifest_entry, "Info");
		if (!info_dict) {
			continue;
		}

		if (_plist_dict_get_bool(parameters, "ApSupportsImg4")) {
			if (!plist_dict_get_item(info_dict, "RestoreRequestRules")) {
				debug("DEBUG: %s: Skipping '%s' as it doesn't have RestoreRequestRules\n", __func__, key);
				continue;
			}
		}
                // here
if (_plist_dict_get_bool(info_dict, "IsFTAB")) {
    debug("DEBUG: %s: Skipping '%s' because IsFTAB\n", __func__, key);
    continue;
}

		if (_plist_dict_get_bool(parameters, "_OnlyFWComponents")) {
			if (!_plist_dict_get_bool(manifest_entry, "Trusted")) {
				debug("DEBUG: %s: Skipping '%s' as it is not trusted\n", __func__, key);
				continue;
			}

			if (!_plist_dict_get_bool(info_dict, "IsFirmwarePayload") && !_plist_dict_get_bool(info_dict, "IsSecondaryFirmwarePayload") && !_plist_dict_get_bool(info_dict, "IsFUDFirmware")) {
				debug("DEBUG: %s: Skipping '%s' as it is neither firmware nor secondary nor FUD firmware payload\n", __func__, key);
				continue;
			}
		}
`

@jiepbhaone

nikias, 2023-08-04:
The changes have been added to git last August already, just compile it from source @jiepbhaone @McVICTORY