libimobiledevice/idevicerestore #518: iOS 16.0 Beta 5 - Got unknown updater name 'Cryptex1LocalPolicy'.

Lethaltail, 2022-08-09:
Restoring iOS 16.0 Beta 5 is failing. I seemed to cause more harm than good with eff9fab. Cryptex1LocalPolicy still needs an actual updater implementation ( i.e. doronz88/pymobiledevice3@6558530 )
```
Received Cryptex1,Ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
DEBUG: restore_send_firmware_updater_data: Got FirmwareUpdaterData request:
common.c:printing 2715 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>DataRequestMsg</string>
        <key>DataType</key>
        <string>FirmwareUpdaterData</string>
        <key>Arguments</key>
        <dict>
                <key>MessageArgInfo</key>
                <dict>
                        <key>ApECID</key>
                        <integer>534E4950</integer>
                        <key>ApBoardID</key>
                        <integer>10</integer>
                        <key>ApChipID</key>
                        <integer>33025</integer>
                        <key>ApSecurityDomain</key>
                        <integer>1</integer>
                        <key>ApSecurityMode</key>
                        <true/>
                        <key>Ap,RecoveryOSPolicyNonceHash</key>
                        <data>
                        J1OmioAbjjYnWcspT9dKLuApk7MNuYuWDgHDWL3/xcU=
                        </data>
                        <key>ApProductionMode</key>
                        <true/>
                        <key>Ap,VolumeUUID</key>
                        <data>
                        AAAAAAAAAAAAAAAAAAAAAA==
                        </data>
                        <key>Ap,NextStageCryptex1IM4MHash</key>
                        <data>
                        ssx6nt7NODNCK65nYtWX6C5P86VOaCaWTvgL6KsIrt8ojwfzvHPY
                        ftGj8IqmEJZY
                        </data>
                        <key>Ap,NextStageIM4MHash</key>
                        <data>
                        UdJYR95U4Pwe971sLY3oxMKUA3GB++WLIH5mZTr/n5sIUhyQYf5l
                        efdt5I7aYasQ
                        </data>
                </dict>
                <key>MessageArgUpdaterLoopCount</key>
                <integer>0</integer>
                <key>DataChunkSize</key>
                <integer>262144</integer>
                <key>MessageForceRepersonalization</key>
                <true/>
                <key>MessageArgType</key>
                <string>FirmwareResponseData</string>
                <key>DeviceGeneratedTags</key>
                <dict>
                        <key>BuildIdentityTags</key>
                        <array/>
                        <key>ResponseTags</key>
                        <array>
                                <string>ApImg4Ticket</string>
                        </array>
                </dict>
                <key>DeviceGeneratedRequest</key>
                <dict>
                        <key>Ap,VolumeUUID</key>
                        <data>
                        AAAAAAAAAAAAAAAAAAAAAA==
                        </data>
                        <key>ApECID</key>
                        <integer>534E4950</integer>
                        <key>ApBoardID</key>
                        <integer>10</integer>
                        <key>ApSecurityMode</key>
                        <true/>
                        <key>Ap,NextStageCryptex1IM4MHash</key>
                        <data>
                        ssx6nt7NODNCK65nYtWX6C5P86VOaCaWTvgL6KsIrt8ojwfzvHPY
                        ftGj8IqmEJZY
                        </data>
                        <key>ApProductionMode</key>
                        <true/>
                        <key>Ap,RecoveryOSPolicyNonceHash</key>
                        <data>
                        J1OmioAbjjYnWcspT9dKLuApk7MNuYuWDgHDWL3/xcU=
                        </data>
                        <key>@ApImg4Ticket</key>
                        <true/>
                        <key>Ap,LocalBoot</key>
                        <true/>
                        <key>ApSecurityDomain</key>
                        <integer>1</integer>
                        <key>Ap,NextStageIM4MHash</key>
                        <data>
                        UdJYR95U4Pwe971sLY3oxMKUA3GB++WLIH5mZTr/n5sIUhyQYf5l
                        efdt5I7aYasQ
                        </data>
                        <key>Ap,LocalPolicy</key>
                        <dict>
                                <key>Digest</key>
                                <data>
                                0QFUOMSokXKjBI1erryy3mV3dcZq+GiRaqeWGQI9goah
                                RhDHJeSRzmf0DL1Yt3hy
                                </data>
                                <key>Trusted</key>
                                <true/>
                        </dict>
                        <key>ApChipID</key>
                        <integer>33025</integer>
                </dict>
                <key>MessageArgUpdaterName</key>
                <string>Cryptex1LocalPolicy</string>
        </dict>
</dict>
</plist>
ERROR: restore_send_firmware_updater_data: Got unknown updater name 'Cryptex1LocalPolicy'.
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
No data to read (timeout)
ReverseProxy[Ctrl]: Terminating
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
```

nikias, 2022-08-29:
Should be working now with d0921e4b468500874773561a341cd662e3fb73fa.

nikias, 2022-08-31:
@a-gunter does it work with older iOS 16 beta versions using Apple Configurator?

Singing-Sung, 2022-08-31:
> Tried this with a 13 Pro Max on iOS 16.0 beta 8 and I get a failed TSS response (status code 94 -- not eligible for the requested build)... But also can't get iOS 16 to restore using Apple Configurator either. For what it's worth I also got the same error when trying the changes at [doronz88/pymobiledevice3@6558530](https://github.com/doronz88/pymobiledevice3/commit/6558530) (which looks like the exact same implementation) so I'm not sure if it doesn't work or if it's on a per-model basis or something.
> 
> Debug log [here](https://github.com/libimobiledevice/idevicerestore/files/9456383/ios16_restore_cryptex.txt)

I faced the same problems on iPhone 12 Pro, too.

> @a-gunter does it work with older iOS 16 beta versions using Apple Configurator?

I tried 16.0 beta 7 and had the same problems. I did not try the older 16.0 beta version.

My platform is Windows 11.

skortela, 2022-08-31:
Same issue with iPhone 12 mini..

a-gunter, 2022-08-31:
Yeah I couldn't get Configurator to work but now I suspect it's an issue with my MobileDevice framework version not being updated (running 1368.12.2) for some reason. It errors out with "AMRestoreErrorDomain error 10 - Failed to handle message type StatusMsg (Protocol error)". Didn't mess around with it enough to try to fix it, though... Wasn't seeing the usual pop-up to update to the latest framework version.

I ended up trying to restore via 3utools and the restore succeeded there. I was able to see the successful TSS response get sent to the device after the CryptexLocal1Policy updater was requested, but I don't have any information about the actual TSS request. I'm not familiar with analyzing / decrypting the network packets so I didn't go much further than that. If needed I might be able to provide more info but may need a poke in the right direction...

Neil-Tu0117, 2022-09-09:
Same issue too（iphone 13）：
(null)Sending BuildIdentityDict now...
Done sending BuildIdentityDict
Sending Cryptex1 TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Cryptex1,Ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
About to send BuildIdentity Dict...
D:\libimobiledevice\libimobiledevice_vs_master_0909_1\idevicerestore\src\common.c:printing 0 bytes plist:
(null)Sending BuildIdentityDict now...
Done sending BuildIdentityDict
Sending Cryptex1LocalPolicy TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to fetch Cryptex1LocalPolicy ticket
ERROR: restore_send_firmware_updater_data: Couldn't get Cryptex1LocalPolicy firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device

rekyagis, 2022-09-09:
I have same issue on iPhone 13 Pro Max with 16 RC.
It is because the TSS request for Cryptex1LocalPolicy is different from AppleConfigurator's one.
Attached patch file will solve this issue.
Thanks.

[idevicerestore_Cryptex1LocalPolicy_TSS.diff.txt](https://github.com/libimobiledevice/idevicerestore/files/9535120/idevicerestore_Cryptex1LocalPolicy_TSS.diff.txt)

Neil-Tu0117, 2022-09-09:
> I have same issue on iPhone 13 Pro Max with 16 RC. It is because the TSS request for Cryptex1LocalPolicy is different from AppleConfigurator's one. Attached patch file will solve this issue. [...]

 I tried it and it works @rekyagis 

Thank you so much !!!

nikias, 2022-09-09:
@rekyagis nice find!

I ended up creating this patch:

```diff
diff --git a/src/restore.c b/src/restore.c
index 30fccbe..5ea86bc 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2881,8 +2881,18 @@ static plist_t restore_get_cryptex1_firmware_data(restored_client_t restore, str

        parameters = plist_new_dict();

-       /* add manifest for current build_identity to parameters (Cryptex1 will require the manifest in a seperate message) */
-       tss_parameters_add_from_manifest(parameters, build_identity, false);
+       plist_t build_identity_tags = plist_access_path(arguments, 2, "DeviceGeneratedTags", "BuildIdentityTags");
+       if (PLIST_IS_ARRAY(build_identity_tags)) {
+               uint32_t i = 0;
+               for (i = 0; i < plist_array_get_size(build_identity_tags); i++) {
+                       plist_t node = plist_array_get_item(build_identity_tags, i);
+                       const char* key = plist_get_string_ptr(node, NULL);
+                       plist_t item = plist_dict_get_item(build_identity, key);
+                       if (item) {
+                               plist_dict_set_item(parameters, key, plist_copy(item));
+                       }
+               }
+       }

        plist_dict_set_item(parameters, "ApProductionMode", plist_new_bool(1));
        plist_dict_set_item(parameters, "ApSecurityMode", plist_new_bool(1));
```

Appreciate if anyone will test it.

rekyagis, 2022-09-12:
@nikias Thanks for your continuous work.

Unfortunately, your patch does not work.
I found that Cryptex1 TSS request requires "ApBoardID" and "ApChipID" and Cryptex1LocalPolicy's must not have "UniqueBuildID".
We may need to add "ApBoardID" and "ApChipID" to Cryptex1 TSS request from build_identity since FirmwareUpdaterData message by device does not contains them. And we have to remove "UniqueBuildID" for Cryptex1LocalPolicy TSS request. (Cryptex1 TSS request does not require it, eather.)

Attached patch does so, and it works for me.
I wonder that we may remove "UniqueBuildID" from tss_request_add_common_tags function, but I'm not sure.

Thanks.

[idevicerestore_Cryptex1LocalPolicy_TSS.diff.txt](https://github.com/libimobiledevice/idevicerestore/files/9544291/idevicerestore_Cryptex1LocalPolicy_TSS.diff.txt)

heissmarkus, 2022-09-12:
@rekyagis  @nikias 
Tested both your original patch and the changes from @rekyagis with some devices.

iPhone X worked without a problem

iPhone SE (3rd generation) failed rather late (Patch of @rekyagis  included)
[logging_20A362_iPhone14,6.txt](https://github.com/libimobiledevice/idevicerestore/files/9547843/logging_20A362_iPhone14.6.txt)

Log OK with iTunes with the same iPhone SE (3rd generation)
[iPhoneUpdater 131.log](https://github.com/libimobiledevice/idevicerestore/files/9547849/iPhoneUpdater.131.log)

rekyagis, 2022-09-13:
@heissmarkus Thank you for your survey. But it seems that your idevicerestore is not patched correctly.
By checking your "logging_20A362_iPhone14,6.txt", neither @nikias 's patch nor mine is not applied. TSS request of Cryptex1 or Cryptex1LocalPolicy does not contain tags of "BMU,*", "Savage,*" ... if patched, but your log says that it does.

My restore log (with iPhone 13 Pro Max, 16.0(20A362)) is here.
[restorelog.txt.zip](https://github.com/libimobiledevice/idevicerestore/files/9552842/restorelog.txt.zip)

Patched "restore.c" is attached FYI.
[restore.c.zip](https://github.com/libimobiledevice/idevicerestore/files/9552778/restore.c.zip)

Thanks.

heissmarkus, 2022-09-13:
@rekyagis 
Thank you, re-checked the patch & build.
The patch was indeed ok but somehow the build was seemingly not fine.

Tested the patch now with
iPhone SE 3rd gen
iPhone X
iPhone 6s 

Looking fine on all of them. Thank your for your effort!

pekn, 2022-09-14:
With 'Update restore.c' patch above, I get the following error from idevicerestore (tested with iPhone XS Max):
""
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168
""
So something is still wrong with TSS request..

nikias, 2022-09-14:
Yes I know, that patch is incomplete, please use this one for now:
https://github.com/libimobiledevice/idevicerestore/files/9544291/idevicerestore_Cryptex1LocalPolicy_TSS.diff.txt

pekn, 2022-09-14:
Still same problem with that patch:

$ ./idevicerestore -e -l -y
idevicerestore 1.0.0-111-gd0921e4
Found device in Normal mode
ECID: 8572763880161326
Identified device as d331pap, iPhone11,6
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 16.0 (build 20A362)
WARNING: Could not lock file 'iPhone11,2,iPhone11,4,iPhone11,6_16.0_20A362_Restore.ipsw.lock'
Verifying 'iPhone11,2,iPhone11,4,iPhone11,6_16.0_20A362_Restore.ipsw'...
Checksum matches.
WARNING: Could not unlock file 'iPhone11,2,iPhone11,4,iPhone11,6_16.0_20A362_Restore.ipsw.lock'
Extracting BuildManifest from IPSW
Product Version: 16.0
Product Build: 20A362 Major: 20
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone11,2,iPhone11,4,iPhone11,6_16.0_20A362_Restore/098-08938-001.dmg'
Getting ApNonce in normal mode... b1 6a 2d 8b 29 bc 97 d9 dd dc 11 9a 4d bd 82 db b5 0d 56 a6 d2 af 16 6b c2 61 65 43 00 c8 9c cb 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 6b eb b8 7c 45 27 1a 66 54 a1 04 ee fe a5 7e 9e be 02 cb 24 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168

pekn, 2022-09-14:
With the following hack, I was able to restore iPhone XS Max with idevicerestore:

plist_t tss_request_send(plist_t tss_request, const char* server_url_string)
{
#if 1 // HACKHACK
        static bool first = false;
        if (!first) {
            first = true;
            //plist_dict_set_item(tss_request, "NeRDEpoch", plist_new_uint(0));
            plist_dict_remove_item(tss_request, "@eUICC,Ticket");
            plist_dict_remove_item(tss_request, "eUICC,ChipID");
        }
#endif

I compared iTunes tss-request.plist with idevicerestore's tss request. It looks like NeRDEpoch is not mandatory but Windows version sends it (apparently it can be found from build manifest).

pekn, 2022-09-14:
Damn, github changed code.. instead of @euicc,Ticket it should be @ followed by eUICC,Ticket

pekn, 2022-09-15:
So is "if (_plist_dict_get_uint(parameters, "eUICC,ChipID") >= 5) {" condition in idevicerestore.c incorrect?
Those eUICC items are added inside that if block.

nikias, 2022-09-15:
@rekyagis thanks for the feedback. So I checked with the "main" TSS request, and apparently it doesn't need the `UniqueBuildID` entry either (also the Cryptex1 works without); however I am quite sure that Apple still adds it to some of the other requests. @pekn I checked the code and it adds a bunch of items to the request that are not even relevant, and manual testing shows they are not required (like eUICC* or BMU* entries).
There is a lot of code regarding TSS that is somehow messed up, I guess it needs some major cleanup.

nikias, 2022-09-16:
Ok I pushed a commit with a more elegant fix. I tested it with the data from your log files, and I am able to get both the Cryptex1, and the Cryptex1LocalPolicy tickets.
See: d97f560eb1dad839f68a2b8c970ce62432893954.
Also I realized that `@BBTicket` is not required but always added to ALL TSS requests that idevicerestore creates; but in the end the relevant code paths always set this property anyway (at least from what saw). Pushed commit 403d2956bee60c321669421edbde29c97c238928 to address that.

nikias, 2022-10-09:
This should be fixed now that I have confirmation of successful iPhone 14 family restores.

closed 2022-10-09 by nikias