libimobiledevice/idevicerestore #520: How can you skip verification of ipsw

o-kotb, 2022-08-17:
I'm trying to erase and flash an iphone 6s+ in DFU mode with the latest firmware. But my laptop is very slow and by the time it's done verifying the ipsw, the phone gets out of DFU mode. I already know that the ipsw is good because it verified it many times successfully. So I'd like to skip this process when flashing.

When I put the phone back in DFU mode and enter "YES," it fails flashing it. One time I was fast enough and entered YES while the phone was still in DFU, but then it got out of the mode in the middle of one of the progress bars and the flashing failed. After my 4th attempt, I now always get a segmentation fault every time I try to flash, but it's hard to try again because it takes too long.

xanadu-lee, 2022-08-18:
I think you can unpack ahead of time and pass the parameter -C

o-kotb, 2022-08-18:
It's already unpacked, but I found that it only does that when you pass -l to it. If you point to the ipsw directly, it skips verification. I'm still not able to restore, but I'll close this as it is not related.

closed 2022-08-18 by o-kotb