libimobiledevice/idevicerestore #528: unable to restore latest iOS version 16.0

akkkityagi, 2022-09-14:
Unable to restore latest version. got a error i.e Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... ERROR: Unable to send TSS request. any solution

N4zah, 2022-09-14:
I have the same problem, tried with an iphone 12 and an iPhone 13 mini. The result is the same each time. It crash after the verifyin restore with : 
First : "Sending TSS request attempt 1... TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build."
Then : 
"ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to fetch Cryptex1LocalPolicy ticket
ERROR: restore_send_firmware_updater_data: Couldn't get Cryptex1LocalPolicy firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device"
[Bug iOS 16.txt](https://github.com/libimobiledevice/idevicerestore/files/9565127/Bug.iOS.16.txt)

NanisBZH, 2022-09-16:
Same problem here, any news ?

nikias, 2022-09-16:
It's being worked on, there's a preliminary patch to work around this issue here: https://github.com/libimobiledevice/idevicerestore/issues/518#issuecomment-1246443076

nikias, 2022-09-18:
Please check with latest commit. I pushed some changes.

100rb, 2022-09-19:
Getting below error :

idevicerestore 1.0.0-113-g403d295
Found device in Recovery mode
ECID: 7886664266612766
Identified device as d54pap, iPhone13,4
Extracting BuildManifest from IPSW
Product Version: 16.0
Product Build: 20A362 Major: 20
INFO: device serial number is F2LG10PJ0D50
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'D:\Client\Warehouse-X\Firmware\Firmware\iPhone13,4_20A362/098-08979-001.dmg'
Getting ApNonce in recovery mode... a9 22 26 61 0e 09 46 b7 0d 08 21 e4 af 70 bd 6a 05 6d 78 2a eb 79 84 dc 88 bf 7e 9b 4c 62 45 6b
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 94 97 dd 80 2d e9 ee 55 a6 56 54 7f 46 9c 97 41 9d c9 53 a8
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d54p.RELEASE.im4p (Firmware/dfu/iBEC.d54p.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (872331 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-8419.2.4
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2778~iphone.im4p (Firmware/all_flash/applelogo@2778~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29081 bytes)...
Extracting h13_ane_fw_styx_d5x.im4p (Firmware/ane/h13_ane_fw_styx_d5x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (5094230 bytes)...
Extracting aopfw-iphone13aop.im4p (Firmware/AOP/aopfw-iphone13aop.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (2426558 bytes)...
Extracting AppleAVE2FW_H13.im4p (Firmware/ave/AppleAVE2FW_H13.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1336022 bytes)...
Extracting armfw_g13p.im4p (Firmware/agx/armfw_g13p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (2538030 bytes)...
Extracting adc-pallas-d5x.im4p (Firmware/isp_bni/adc-pallas-d5x.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (22198889 bytes)...
Extracting t8101pmp.im4p (Firmware/pmp/t8101pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (427710 bytes)...
Extracting rans.t8101.release.im4p (Firmware/rans.t8101.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (912943 bytes)...
Extracting iphone13dcp.im4p (Firmware/dcp/iphone13dcp.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (4220606 bytes)...
Extracting 098-08840-001.dmg.trustcache (Firmware/098-08840-001.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (15593 bytes)...
Extracting SmartIOFirmware_ASCv4.im4p (Firmware/SmartIOFirmware_ASCv4.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1163186 bytes)...
[... 2652 lines cut ...]
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>DataRequestMsg</string>
        <key>DataType</key>
        <string>SourceBootObjectV4</string>
        <key>Arguments</key>
        <dict>
                <key>Variant</key>
                <string>Erase</string>
                <key>ImageName</key>
                <string>Cryptex1,SystemOS</string>
        </dict>
</dict>
</plist>
About to send Cryptex1,SystemOS...
common.c:printing 0 bytes plist:
(null)Extracting 098-09281-001.dmg (098-09281-001.dmg)...
ERROR: Out of memory
ERROR: Unable to extract 098-09281-001.dmg from D:\Client\Warehouse-X\Firmware\Firmware\iPhone13,4_20A362.ipsw
ERROR: Unable to extract component Cryptex1,SystemOS
ERROR: Unable to send SourceBootObjectV4
ERROR: Unable to successfully restore device
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

NanisBZH, 2022-09-22:
> Please check with latest commit. I pushed some changes.

Seem's to be ok for iPhone 13 with IOS 16. I can't check with iPhone 14 yet.

nikias, 2022-10-10:
Both the memory issue and the TSS issue should be fixed with the most recent commits.

closed 2022-10-10 by nikias