libimobiledevice/idevicerestore #529: Restore with iOS 16 is making my computer run out of memory with multiple devices

c0ry, 2022-09-16:
In the new update for iOS 16 at the point where the output says "About to send Cryptex1,SystemOS" it seems to be taking up a lot of computer resources.  My computer is now freezing whenever I have more than 5 devices running at once when it gets to this point in the process.  I used to be able to restore 30 devices at once with no issues before this update and now my computer cannot handle more than 5 at once.

Is there anything that can be done about this?  Is it possible to skip this part of the process or are we stuck with this moving forward?

kamatam9, 2022-09-18:
If you are running idevicerestore on Linux, you may want to try https://github.com/libimobiledevice/idevicerestore/pull/521.

100rb, 2022-09-20:
> If you are running idevicerestore on Linux, you may want to try #521.

Tried #521 on windows but getting error during build.
Can you please provide code for windows also.

Getting below error :
make  all-recursive
make[1]: Entering directory '/home/saurabh/build/idevicerestore'
Making all in src
make[2]: Entering directory '/home/saurabh/build/idevicerestore/src'
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
  CC       idevicerestore-ftab.o
  CC       idevicerestore-ipsw.o
ipsw.c:33:10: fatal error: sys/mman.h: No such file or directory
   33 | #include <sys/mman.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:655: idevicerestore-ipsw.o] Error 1
make[2]: Leaving directory '/home/saurabh/build/idevicerestore/src'
make[1]: *** [Makefile:430: all-recursive] Error 1
make[1]: Leaving directory '/home/saurabh/build/idevicerestore'
make: *** [Makefile:362: all] Error 2

Orbif, 2022-09-22:
I have the same problem, the component Cryptex1,SystemOS takes more than 2.5Gb of RAM for one single idevicerestore process, which is a lot. 
mman seems a promising solution for Linux, but is not available on Windows. 

Can't we just transfer  the component from disk to device without copying all content on RAM at once, like we do for the OS filesystem?

nicolassan, 2022-09-22:
Same issue, a lot of RAM memory used and crash with a lot of devices (IOS 16)

nikias, 2022-09-22:
The mmap solution doesn't work like this on Windows; there is an API for something similar on Windows but I don't know how it works. I think we need to work out - like @Orbif says - a way to directly stream the component to the device. However some or most components needs stitching so we need to store them on disk and then send them over. Not sure about the SystemOS image if it requires stitching/personalization.

nikias, 2022-09-22:
According to the build manifest, images like `Cryptex1,SystemOS` don't require personalization, so it should be possible to change the code to directly stream the images. I will look into it.

kamatam9, 2022-09-22:
Yeah, streaming may be better approach here. It could also avoid 32-bit limitation with a certain IPSW; macOS 13 IPSW has a component larger than 4GB and therefore malloc/mmap runs out of the virtual address space. While mmap (and possibly file mapping object in WIN32) could be still helpful to make memory usage as little as possible, streaming is probably simpler and more platform-agnostic.

kamatam9, 2022-09-23:
@100rb, just in case, I pushed (only compile tested) code for Windows. You may want to try it.

https://github.com/kamatam9/idevicerestore/tree/mmap-win32-untested

100rb, 2022-09-26:
> @100rb, just in case, I pushed (only compile tested) code for Windows. You may want to try it.
> 
> https://github.com/kamatam9/idevicerestore/tree/mmap-win32-untested

Getting same error "Out of memory". Please check attached log.

[Log.txt](https://github.com/libimobiledevice/idevicerestore/files/9644539/Log.txt)

Orbif, 2022-09-26:
> Getting same error "Out of memory". Please check attached log.
>
> [Log.txt](https://github.com/libimobiledevice/idevicerestore/files/9644539/Log.txt)

@100rb This patch is useful only if you use it with a fully extracted IPSW.

I tried @kamatam9 code on Windows but I had an "Failed to map a view of a file mapping" directly at the first component.
I managed to go further by removing FILE_MAP_WRITE flag because files are opened with read only parameter. 
Nevertheless I still have a failure at the end when trying to send Savage component, but I'm not sure why.

weikaizu, 2022-09-27:
I have the same problem too, is this problem solved?

nikias, 2022-09-27:
It's not solved yet. I am looking into how to improve this; can anyone in this thread run idevicerestore with added `-d` option (without any of the aforementioned patches) and capture the console output to a file, and upload it here? I need to understand a few things so I can improve it. Thanks.

Orbif, 2022-09-27:
@nikias here the console output on Windows for iphone 11:

[idevicerestore executed with extracted ipsw folder](https://github.com/libimobiledevice/idevicerestore/files/9657725/idevicerestore_iphone_11_with_path_extracted_ipsw.txt)

[idevicerestore executed with compressed .ipsw file](https://github.com/libimobiledevice/idevicerestore/files/9657872/idevicerestore_iphone_11_with_compressed_ipsw.txt)

nikias, 2022-09-27:
Thanks @Orbif, that should help

kamatam9, 2022-09-28:
@Orbif, according to https://stackoverflow.com/a/55023760, removing FILE_MAP_WRITE seems correct. I wasn't able to test the patch myself because I don't have a physical Windows machine. I may not update the branch anymore in favor of nikias's more appropriate solution, but I appreciate for your testing.

nikias, 2022-10-01:
Ok I came up with the following patch:
https://gist.github.com/nikias/cc5c8eead95629e1ced53e57951af97f
This will directly send over the file data of the newer images (like Cryptex1,SystemOS) instead of extracting them entirely into memory. I would be glad if someone would test this and report back.
For other (especially larger) images this might be useful too, but if they require personalization we need to store them to file first.

kamatam9, 2022-10-02:
@nikias, I tested the patch with macOS 13 beta7 IPSW. Restoring successfully completed with both zipped and fully extracted IPSW while limiting the memory usage to 512MB. The IPSW required more than 4GB without the patch.

nikias, 2022-10-02:
Great, this sounds like a good improvement as a start. Thanks @kamatam9 for testing.

nikias, 2022-10-02:
Committed https://github.com/libimobiledevice/idevicerestore/commit/aa98e764bff59b45173a86d2587b59f5692982eb