libimobiledevice/idevicerestore #530: j314cap, MacBookPro18,4

abuyaseen, 2022-09-16:
mac@cs-MacBook-Pro idevicerestore % idevicerestore -l
idevicerestore 1.0.0-111-gd0921e4
Found device in DFU mode
ECID: 349xxxxxxxxxxxxx
Identified device as j314cap, MacBookPro18,4
The following firmwares are currently being signed for MacBookPro18,4:
  [1] 12.6 (build 21G115)
  [2] 12.5.1 (build 21G83)
  [3] 12.5 (build 21G72)
  [4] 12.4 (build 21F79)
  [5] 12.3.1 (build 21E258)
  [6] 12.3 (build 21E230)
  [7] 12.2.1 (build 21D62)
  [8] 12.2 (build 21D49)
  [9] 12.1 (build 21C52)
  [10] 12.0.1 (build 21A559)
Select the firmware you want to restore: 5
Selected firmware 12.3.1 (build 21E258)
Verifying 'UniversalMac_12.3.1_21E258_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.3.1
Product Build: 21E258 Major: 21
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Performing macOS restore
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_12.3.1_21E258_Restore/018-25400-296.dmg'
Getting ApNonce in dfu mode... dc 65 9c 7a b7 73 ab f7 22 07 6b 33 3d 98 78 52 6e fc 55 0f 32 f6 dd 5c 7e 3d 2d 10 23 80 0d 80 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.j314c.RELEASE.im4p (Firmware/dfu/iBSS.j314c.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (963562 bytes)...
[==================================================] 100.0%
ERROR: Device did not reconnect in DFU or recovery mode. Possibly invalid iBSS. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode
mac@cs-MacBook-Pro idevicerestore % 
i try all ipsw signed

abuyaseen, 2022-09-20:
[sudo] password for macos: 
[10:07:43.249][3] usbmuxd v1.1.1 starting up
[10:07:43.249][4] Creating socket
[10:07:43.249][4] Not dropping privileges to root
[10:07:43.249][4] Initializing USB
[10:07:43.249][3] Using libusb 1.0.26
[10:07:43.254][4] Registering for libusb hotplug events
[10:07:43.254][4] 0 devices detected
[10:07:43.254][3] Initialization complete
[10:07:43.254][4] Client 9 accepted: /usr/libexec/gvfs-afc-volume-monitor[1532]
[10:08:04.167][4] Client 10 accepted: idevicerestore[25013]
[10:09:28.466][4] Client 10 connection closed
[10:09:28.467][4] Client 10 is going to be disconnected: idevicerestore[25013]
--------------------------------------------------------------------------------------------------------------------------------------
Received SHSH blobs
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.j317.RELEASE.im4p (Firmware/dfu/iBEC.j317.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (1464624 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 001919e12280012e (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 001919e12280012e (udid: N/A) connected in Recovery mode
opening device 05ac:1281...
found device with ECID 001919e12280012e
Setting to configuration 1
Setting to interface 0:0
Recovery Mode Environment:
iBoot build-version=iBoot-7429.0.97.0.4
iBoot build-style=DEVELOPMENT
Sending RestoreLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreLogo' in TSS response
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2388~ipad-USBc.im4p (Firmware/all_flash/applelogo@2388~ipad-USBc.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (34383 bytes)...
DEBUG: ANE is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'ANE' in TSS response
NOTE: No path for component ANE in TSS, will fetch from build_identity
Extracting h11_ane_fw_pontus.im4p (Firmware/ane/h11_ane_fw_pontus.im4p)...
Personalizing IMG4 component ANE...
Tag found
Sending ANE (4686912 bytes)...
DEBUG: AOP is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'AOP' in TSS response
NOTE: No path for component AOP in TSS, will fetch from build_identity
Extracting aopfw-ipad8aop.im4p (Firmware/AOP/aopfw-ipad8aop.im4p)...
Personalizing IMG4 component AOP...
Tag found
Sending AOP (1600912 bytes)...
DEBUG: AVE is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'AVE' in TSS response
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H11.im4p (Firmware/ave/AppleAVE2FW_H11.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1317328 bytes)...
DEBUG: GFX is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'GFX' in TSS response
NOTE: No path for component GFX in TSS, will fetch from build_identity
Extracting armfw_g11g.im4p (Firmware/agx/armfw_g11g.im4p)...
Personalizing IMG4 component GFX...
Tag found
Sending GFX (897720 bytes)...
DEBUG: ISP is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'ISP' in TSS response
NOTE: No path for component ISP in TSS, will fetch from build_identity
Extracting adc-petra-j3x.im4p (Firmware/isp_bni/adc-petra-j3x.im4p)...
Personalizing IMG4 component ISP...
Tag found
Sending ISP (13951016 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreTrustCache' in TSS response
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 078-69254-013.dmg.trustcache (Firmware/078-69254-013.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (12847 bytes)...
DEBUG: SIO is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'SIO' in TSS response
NOTE: No path for component SIO in TSS, will fetch from build_identity
Extracting SmartIOFirmware_ASCv2.im4p (Firmware/SmartIOFirmware_ASCv2.im4p)...
Personalizing IMG4 component SIO...
Tag found
Sending SIO (1176924 bytes)...
ramdisk-size=0x20000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 078-69254-013.dmg (078-69254-013.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (124973971 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j318ap.im4p (Firmware/all_flash/DeviceTree.j318ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (43317 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j318.RELEASE.im4p (Firmware/all_flash/sep-firmware.j318.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (7822971 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad8 (kernelcache.release.ipad8)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (16389352 bytes)...
DEBUG: Waiting for device to disconnect...
ERROR: Failed to place device in restore mode
ERROR: Unable to place device into restore mode

abuyaseen, 2022-09-20:
linux and mac same result

nikias, 2022-10-05:
I wonder if this is fixed with latest commit c45f74cc3e075d3b04dd920141e82a51dc20bd9b?