libimobiledevice/idevicerestore #532: idevicerestore of iPhone 14 Pro Max fails

pekn, 2022-09-20:
idevicerestore of iPhone 14 Pro Max seems to fail:
```
$ sudo ./idevicerestore -e -l -y
idevicerestore 1.0.0-113-g403d295
Found device in Normal mode
ECID: 8506157881311262
Identified device as d74ap, iPhone15,3
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 16.0.1 (build 20A371)
Verifying 'iPhone15,3_16.0.1_20A371_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 16.0.1
Product Build: 20A371 Major: 20
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 098-09093-003.dmg
[==================================================] 100.0%
Getting ApNonce in normal mode... d1 41 15 90 7b 18 f4 3d 78 06 6f 98 a2 f6 b8 4a 9b a0 24 c0 d0 bc 80 41 0d 19 b8 aa 54 29 77 71 
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 56 8a a0 91 ba 93 fc d0 1c 11 c8 d6 87 8e 75 c8 cd 93 1c 8d 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=69&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=69, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device
```

I checked differences between iTunes TSS request and idevicerestore TSS request.
The following are missing from idevicerestore TSS request:
 - UID_MODE (boolean, false in iTunes TSS request - it's not clear where this is taken from)
 - eUICC,ApProductionMode (boolean, true)
 - eUICC,Gold (dict, with data Digest.. taken from BuildManifest)
 - eUICC,Main (dict, with data Digest.. taken from BuidManifest)

By adding above to TSS request, idevicerestore goes fine.. However, after idevicerestore, device does not boot up - instead it goes to DFU mode :(

nikias, 2022-09-20:
Hi, feel free to drop me a full debug log output via PM on Telegram (you can find me in the https://t.me/libimobiledevice channel)

truonggiang0710, 2022-09-22:
Hi @pekn , can you share the iTunes's request ?

pekn, 2022-09-22:
[tss-request.txt](https://github.com/libimobiledevice/idevicerestore/files/9623117/tss-request.txt)
tss request attached

DanTheMann15, 2022-09-22:
This also affects tsschecker as well, any tss request for the iPhone 14 Pro models always fails.

pekn, 2022-09-23:
I noticed that idevicerestore's baseband firmware is different than iTunes version.
 - vinyl_07.vnlfw from ipsw is extracted on top of bbfw in iTunes version
 - 509659a8066d2415f2ff956c417f7f8d6e663f4560170cf1744e50eeaa628e0d/update/gold/ and 509659a8066d2415f2ff956c417f7f8d6e663f4560170cf1744e50eeaa628e0d/update/main/ contain additional file - ticket.der (not sure how this file is generated)
 - bbticket.der file size is different
 - also there is Info.plist in iTunes version

There are same differences with iPhone 14 - that works OK with idevicerestore.. so above changes might not explain why iPhone 14 Pro Max does not boot after idevicerestore.

DanTheMann15, 2022-09-24:
did you try adding `plist_dict_set_item(request, "UID_MODE", plist_new_bool(0));` to tss.c on line 75?
i suggest you give it a try if you are still getting error 69 during the tss request.

the device won't boot if the firmwares can't get their signing tickets.

nikias, 2022-09-24:
There's a new key RequiresUIDMode (true) in the BuildManifest/BuildIdentity but regardless it is set to false, I haven't understood the code yet. There's also a key Ap,SikeFuse in the iTunes request but I have no idea where the value is taken from.
All a bit hard to experiment with without having the actual device 🙃

DanTheMann15, 2022-09-25:
if i set AP,SkiaFuse in the request to 1, it ends in error 94. it seems to be much like ApProductionMode and related integers in behavior.

nikias, 2022-09-25:
Ok so I added the missing `eUICC,Gold` and `eUICC,Main` to `tss_request_add_vinyl_tags` and also `eUICC,ApProductionMode` which - I believe - is just a copy of `ApProductionMode` with commit 88aeb4ce1313a9e89209c08efa62fb6b7eb428c4.
Regarding `UID_Mode` I looked at MobileDevice framework but didn't fully understand how this is handled exactly, and came up some preliminary handling code with commit bb7f206090649933ad616baa1b9497ee978052c8.
Still unsure how to decide when to set `Ap,SikaFuse`. Maybe just set it alongside UID_MODE if it doesn't interfere with requests for older devices.

truonggiang0710, 2022-09-28:
Hi @pekn ,
Do you find out the way ?

pekn, 2022-09-28:
Here's BasebandCertId: 3559316616

No, I haven't solved this one yet. In TSS response, there are LLB-TBM and iBSS-TBM entries. Probably those should be used for something. See here:
[tss-response.txt](https://github.com/libimobiledevice/idevicerestore/files/9662304/tss-response.txt)

pekn, 2022-09-29:
Finally, now I know what changes are needed.
In TSS response, we get LLB-TBM ucer and ucon fields (see tss-response.txt above). Those need to be added with specific syntax to the end of LLB (part of NorData) - so there is kind of two levels of personalization in that file.
[llb.txt](https://github.com/libimobiledevice/idevicerestore/files/9672275/llb.txt)

I attached example llb - see llb.txt.. now you can use xxd -d llb.txt > llb.dat.. then use https://lapo.it/asn1js/ to decode llb.dat. At end of decoded data, you can see IM4R structure - that needs to be added.

pekn, 2022-09-29:
Also forgot to mention.. similar change is probably needed for iBSS (iBSS-TBM in tss-response.txt).. However, DFU mode seems to have changed - idevicerestore does not recognize that.

truonggiang0710, 2022-09-30:
Hi @pekn, which need add into request to get LLB-TBM, iBSS-TBM in response ?

truonggiang0710, 2022-09-30:
Hi @pekn , what news in the new IM4R structure ?

nikias, 2022-10-04:
Put together this patch that should handle the TBM / ucon / ucer during stitching:
https://gist.github.com/nikias/c766020303b62484eff9c601d281c5d8

pekn, 2022-10-04:
Tested your patch -> now it works.

nikias, 2022-10-04:
Awesome! Thanks for testing. Pushed commit a4f5a0c1a65c9df239a737c350d4723c2a8cbc55.

nikias, 2022-10-04:
@pekn I know that there is no more DFU mode, but afaik there is still recovery mode. If you manually put the device in recovery mode, does it still work? While looking through the code I realized that it only does only add e.g. `eUICC*` elements when starting from normal mode. So I wonder if restoring would actually still work when starting from recovery mode?

pekn, 2022-10-04:
Yes, at least with this device idevicerestore works OK from recovery mode.
I think eUICC ticket is placed to baseband zip file (see my comment about vinyl_07 above).

And I think there is still DFU mode (or Apple Debug mode). When  device did not restart after idevicerestore, then iTunes recognized that mode as DFU. However, idevicerestore did not recognize that device anymore.

nikias, 2022-10-04:
@pekn ah right, that actually makes sense. You don't happen to have a sample of stitched basebandfirmware by any chance?

pekn, 2022-10-05:
File is too large to attach here, I will send it to you via telegram.