libimobiledevice/idevicerestore #534: idevicerestore to iPhone 8 doesn't work

SternXD, 2022-10-01:
yeah so i'm trying to restore to iOS 15.6 RC (from iOS 16.1B2) to my iPhone 8 but it just does this

sudo idevicerestore -e IPSW location -k
idevicerestore 1.0.0-118-gbb7f206
Found device in Recovery mode
ECID: 942084844879918
Identified device as d20ap, iPhone10,1
Extracting BuildManifest from IPSW
Product Version: 15.6
Product Build: 19G69 Major: 19
INFO: device serial number is F4GVP52AJC6F
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/media/live/Windows/Users/xcstern-12/Documents/iPhone_4.7_P3_15.6_19G69_Restore/078-33721-075.dmg'
Getting ApNonce in recovery mode... 14 2e ad 44 e9 f5 81 d6 9a 3c e1 69 67 c5 de f5 40 21 f6 87 bd 73 97 d5 10 ac 81 a2 d1 28 e3 ef 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 3a 7a 46 b5 32 ca 29 2b f0 34 af 4b 21 7d 78 d4 76 98 34 74 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d20.RELEASE.im4p (Firmware/dfu/iBEC.d20.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1083925 bytes)...
Recovery Mode Environment:
iBoot build-version=
iBoot build-style=
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (13821 bytes)...
Extracting AppleAVE2FW_H10.im4p (Firmware/ave/AppleAVE2FW_H10.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1065585 bytes)...
Extracting adc-nike-d20.im4p (Firmware/isp_bni/adc-nike-d20.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (9824325 bytes)...
Extracting 078-34285-076.dmg.trustcache (Firmware/078-34285-076.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (12464 bytes)...
ramdisk-size=
Extracting 078-34285-076.dmg (078-34285-076.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (113867416 bytes)...
Extracting DeviceTree.d20ap.im4p (Firmware/all_flash/DeviceTree.d20ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (41190 bytes)...
Extracting sep-firmware.d20.RELEASE.im4p (Firmware/all_flash/sep-firmware.d20.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (1575984 bytes)...
Extracting kernelcache.release.iphone10 (kernelcache.release.iphone10)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (15489219 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.

vungocmobile, 2022-10-03:
idevicerestore now not work with 16.1 beta

nikias, 2022-10-08:
Ok try this patch (on top of latest code from git):
```diff
diff --git a/src/recovery.c b/src/recovery.c
index 1a9c9709..24d0c3d0 100644
--- a/src/recovery.c
+++ b/src/recovery.c
@@ -498,7 +498,7 @@ int recovery_send_kernelcache(struct idevicerestore_client_t* client, plist_t bu
 		recovery_error = irecv_send_command(client->recovery->client, setba);
 	}
 
-	recovery_error = irecv_send_command_breq(client->recovery->client, "bootx", client->macos_variant ? 1 : 0);
+	recovery_error = irecv_send_command_breq(client->recovery->client, "bootx", 1);
 	if (recovery_error != IRECV_E_SUCCESS) {
 		error("ERROR: Unable to execute %s\n", component);
 		return -1;
 ```

vungocmobile, 2022-10-08:
it's Working...

Extracting sep-firmware.d20.RELEASE.im4p (Firmware/all_flash/sep-firmware.d20.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (1516208 bytes)...
Extracting kernelcache.release.iphone10 (kernelcache.release.iphone10)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (18050021 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000001 has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 32789

vungocmobile, 2022-10-08:
16.1b working with this patch 

recovery_error = irecv_send_command_breq(client->recovery->client, "bootx", 1);

nikias, 2022-10-08:
Committed that change with 961077074cf4f113d685ce19f3e0b18d343a74d6.

nikias, 2022-10-17:
Reported as working, so closing.

closed 2022-10-17 by nikias