libimobiledevice/idevicerestore #539: UNABLE TO RESTORE DEVICE

jenn-arai, 2022-10-11:
i have an iphone 7 plus model A1784
when running `idevicerestore --erase --latest`
i get 
```Identified device as d111ap, iPhone9,4
The following firmwares are currently being signed for iPhone9,4:
  [1] 15.7 (build 19H12)
```

i pick 1 and it keeps running and after a while i get some errors
then it ends with

```ERROR: Unable to successfully restore device
Checkpoint 1608 complete with code 0
ERROR: Unable to restore device```

any idea what to do?

alfiecg24, 2022-10-12:
Try running with `—debug`. What do you see there?

klei22, 2022-10-20:
I also had the same error message when trying to restore an iPad 2 Air.

Sharing `--debug` output, appears that the ASR checksum fails:

```
ASR VERIFY PROGRESS: 98%
ramrod_display_set_granular_progress_forced: 49.000000
ASR VERIFY PROGRESS: 100%
ramrod_display_set_granular_progress_forced: 50.000000
ASR: Calculated SHA-1 checksum 6852087808 bytes in 21.75 seconds, 307591.02 KiB/s
ASR: Checksum failed.
ASR: Expected 9D3B2B3ABFB6E3F23E173D19230753BA90A6EDB9
ASR: but got 44271610AB2A82B073CE3466B8263EFD324C51B6
ASR STATUS: fail: checksum failure
ASR: Could not restore - Inappropriate ioctl for device
ASR STATUS: fail
ASR attempt failed. Last error from asr is (No error string returned from asr)
Invalid/empty asr error string. Returning standard asr error
restore_apfs_image : failed to restore an APFS image, error = 1100
Unknown error(1100). Nothing to do
Current system volume stats: 61015 MiB blocks, 54365 MiB available, 54365 MiB free
Failed to asr image and not reattempting for erase install
ASR and invert Failed
[04:20:47.0521-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:1100) (null):[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:1100}}
restore-step-codes = {0x1107065B:{0:1100}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0523-GMT]{3>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=50 [sync=true] (first failure)
[04:20:47.0523-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:1100) RESTORED:[0x0677] perform_main_os_prepare [0]D(failed to restore APFS image)
restore-step-results = {0x11070677:{0:1100};0x1107065B:{0:1100}}
restore-step-codes = {0x11070677:{0:1100};0x1107065B:{0:1100}}
restore-step-domains = {0x11070677:{0:"AMRestoreErrorDomain"};0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070677:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0524-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x067C] cleanup_boot_command
restore-step-ids = {0x11030677:49;0x1103065B:57;0x1103067C:58}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x1103067C:cleanup_boot_command}
restore-step-uptime = 278
restore-step-user-progress = 50
entering reset_boot_command_if_in_values
executing /usr/sbin/nvram -d recovery-boot-mode
Successfully deleted recovery-boot-moderecovery-boot-mode
executing /usr/sbin/nvram -d iboot-failure-reason
Successfully deleted iboot-failure-reasoniboot-failure-reason
[04:20:47.0583-GMT]{3>7} CHECKPOINT END: RESTORED:[0x067C] cleanup_boot_command
restore-step-ids = {0x11030677:49;0x1103065B:57}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0584-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:49;0x1103065B:57;0x11031613:59}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11031613:cleanup_recovery_os_volume}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0585-GMT]{3>7} CHECKPOINT END: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:49;0x1103065B:57}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0586-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:49;0x1103065B:57;0x11030647:60}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 278
restore-step-user-progress = 50
[04:20:47.0587-GMT]{3>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:49;0x1103065B:57}
[... 47 lines cut ...]
        <string>idevicerestore</string>
        <key>Request</key>
        <string>Goodbye</string>
</dict>
</plist>
04:19:34 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
04:19:34 property_list_service.c:211 internal_plist_receive_timeout(): 142 bytes following
04:19:34 property_list_service.c:224 internal_plist_receive_timeout(): received 142 bytes
04:19:34 property_list_service.c:253 internal_plist_receive_timeout(): printing 385 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>CHECKPOINT_COMPLETE</key>
        <false/>
        <key>CHECKPOINT_RESULT</key>
        <integer>0</integer>
        <key>CHECKPOINT_ID</key>
        <integer>1609</integer>
        <key>MsgType</key>
        <string>CheckpointMsg</string>
</dict>
</plist>
ERROR: Unable to restore device
```

alfiecg24, 2022-10-20:
Near the top it says `ASR: Could not restore - Inappropriate ioctl for device`

Take a look online - this seems to indicate that there is an issue with the USB connection

alfiecg24, 2022-10-20:
But yes you’re right, the checksum did also fail by the looks of it. Try re-downloaded the IPSW file?

klei22, 2022-10-22:
Thank you for your advice. I tried redownloading the IPSW file and ran into the same errors. I also used different USB cables, and had the same issues. When I originally tried to update via itunes, I kepthearing a connect/disconnect sound for the USB, I wonder if this may be cable or possible device related? After using three different cables, the issues still persist.  I did some reserch online about a device storage problems for the IPad Air 2, and saw comment regarding not enough space in 16Gb / 32Gb models. The device I have includes storage for 16GB with a SIM slot. Perhaps there's just not enough space on the IPad to accommodate the restore. The reason for why the device stopped working was because of an update from firmware version 14 to 15.

jenn-arai, 2022-11-20:
thank u for helping the issue was my cable

closed 2022-11-20 by jenn-arai