libimobiledevice/idevicerestore #545: Unable to compile r663.0b4810f on Arch Linux from AUR package

kode54, 2022-10-19:
It gets to restore.c and fails with a specific error:

```
restore.c: In function ‘restore_send_personalized_boot_object_v3’:
restore.c:3409:29: error: ‘message’ undeclared (first use in this function)
 3409 |                 debug_plist(message);
      |                             ^~~~~~~
restore.c:3409:29: note: each undeclared identifier is reported only once for each function it appears in
restore.c: In function ‘restore_send_source_boot_object_v4’:
restore.c:3515:29: error: ‘message’ undeclared (first use in this function)
 3515 |                 debug_plist(message);
      |                             ^~~~~~~
make[2]: *** [Makefile:712: idevicerestore-restore.o] Error 1
```

nikias, 2022-10-19:
My bad, copy&paste error. Fixed with 7b89019.

kode54, 2022-10-19:
Yup, thanks.