libimobiledevice/idevicerestore #546: Watch series 2, ERROR: Unable to restore device

hiepnh0, 2022-10-20:
I am restoring my Watch series 2 but error: ERROR: Unable to restore device
Here is the terminal output:
```
$ sudo ./idevicerestore -e -y /tmp/Watch2,7_n28dap_42mm_Watch2,4_n75ap_42mm_6.3_17U208_restore.ipsw -i 2863729584115494
idevicerestore 1.0.0-122-ge74ae4d
Found device in Recovery mode
ECID: 2863729584115494
Identified device as n75ap, Watch2,4
Extracting BuildManifest from IPSW
Product Version: 6.3
Product Build: 17U208 Major: 17
INFO: device serial number is FH7TT19PHDY9
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/tmp/Watch2,7_n28dap_42mm_Watch2,4_n75ap_42mm_6.3_17U208_restore/038-87845-009.dmg'
Getting ApNonce in recovery mode... c6 38 ab 9a b3 6d 1c 5c 1a 60 6d ff 2a 1f b5 9b ee 57 76 1a a2 db 85 ab 4d 0e eb fa c6 ed c1 4e 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 40 e5 1e 35 4e 6b 88 18 e0 03 af 07 27 76 c6 f5 d2 84 ba e4 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1...
response successfully received
Received SHSH blobs
Extracting iBEC.n65.RELEASE.im4p (Firmware/dfu/iBEC.n65.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (747975 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-5540.144.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@132~watch.im4p (Firmware/all_flash/applelogo@132~watch.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (8706 bytes)...
Extracting 038-87586-010.dmg.trustcache (Firmware/038-87586-010.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (13743 bytes)...
ramdisk-size=0x8000000
Extracting 038-87586-010.dmg (038-87586-010.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (67956485 bytes)...
Extracting DeviceTree.n75ap.im4p (Firmware/all_flash/DeviceTree.n75ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (110731 bytes)...
Extracting sep-firmware.n75.RELEASE.im4p (Firmware/all_flash/sep-firmware.n75.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (2103233 bytes)...
Extracting kernelcache.release.watch2 (kernelcache.release.watch2)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (11149214 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 3e5b827893c13d722ec4a017564f709845f2910e has successfully entered restore mode
Hardware Information:
BoardID: 14
ChipID: 32770
UniqueChipID: 2863729584115494
ProductionMode: true
Previous restore exit status: 0x100
Starting Reverse Proxy
ReverseProxy[Ctrl]: (status=1) Ready
Checkpoint 1621 complete with code 0
Checkpoint 1540 complete with code 0
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Checkpoint 1547 complete with code 0
Waiting for NAND (28)
Checkpoint 1549 complete with code 0
Checkpoint 1550 complete with code 0
Checkpoint 1551 complete with code 0
Checkpoint 1628 complete with code 0
Checkpoint 1552 complete with code 0
Checkpoint 1555 complete with code 0
Checkpoint 1662 complete with code 0
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n65.RELEASE.im4p (Firmware/all_flash/LLB.n65.RELEASE.im4p)...
Personalizing IMG4 component LLB...
Extracting applelogo@132~watch.im4p (Firmware/all_flash/applelogo@132~watch.im4p)...
Personalizing IMG4 component AppleLogo...
Extracting dali-ui.n75.release.im4p (Firmware/all_flash/dali-ui.n75.release.im4p)...
Personalizing IMG4 component Dali...
Extracting DeviceTree.n75ap.im4p (Firmware/all_flash/DeviceTree.n75ap.im4p)...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@132~watch.im4p (Firmware/all_flash/recoverymode@132~watch.im4p)...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n65.RELEASE.im4p (Firmware/all_flash/iBoot.n65.RELEASE.im4p)...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n75.RELEASE.im4p (Firmware/all_flash/sep-firmware.n75.RELEASE.i" 
Reflash Error "ERROR: Unable to receive data from ASR
ERROR: Unable to receive validation packet
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device" 
Reflash Error "" 
Reflash Error "ERROR: Unable to restore device" 
"m4p)...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n75.RELEASE.im4p (Firmware/all_flash/sep-firmware.n75.RELEASE.im4p)...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Checkpoint 1545 complete with code 0
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checkpoint 1637 complete with code 0
Checking filesystems (15)
Checkpoint 1556 complete with code 0
Checkpoint 1620 complete with code 0
Checkpoint 1557 complete with code 0
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Checkpoint 1558 complete with code 0
Checkpoint 1559 complete with code 0
Checkpoint 1560 complete with code 0
Checkpoint 1561 complete with code 0
Checkpoint 1562 complete with code 0
Checkpoint 1563 complete with code 0
Checkpoint 1633 complete with code 0
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checkpoint 1565 complete with code 0
Checkpoint 1614 complete with code 0
Checkpoint 1619 complete with code 0
Checkpoint 1567 complete with code 0
Creating partition map (11)
Checkpoint 1569 complete with code 0
Checkpoint 1632 complete with code 0
Checkpoint 1570 complete with code 0
Checkpoint 1629 complete with code 0
Checkpoint 5645 complete with code 0
Creating filesystem (12)
Checkpoint 1624 complete with code 0
Checkpoint 1625 complete with code 0
Checkpoint 1626 complete with code 0
About to send filesystem...
Connected to ASR
Validating the filesystem
Checkpoint 1627 complete with code 14
ReverseProxy[Ctrl]: (status=2) Terminated
```

hiepnh0, 2022-11-04:
Retrying to receive validation packet fix this issue

closed 2022-11-04 by hiepnh0

nikias, 2022-11-04:
@hiepnh0 question, did you debug this issue? I wonder if there's a timeout situation in asr.c in function `asr_receive`:
```c
	device_error = idevice_connection_receive(asr->connection, buffer, ASR_BUFFER_SIZE, &size);
	if (device_error != IDEVICE_E_SUCCESS) {
		error("ERROR: Unable to receive data from ASR\n");
		free(buffer);
		return -1;
	}
```
Because the error is badly handled, any situation will just be interpreted as failure, while maybe just the device was too slow to send a reply within the default libusbmuxd 5 second timeout window. Then again, libimobiledevice isn't properly handling the errors from libusbmuxd...

hiepnh0, 2022-11-09:
@nikias
device_error = idevice_connection_receive(asr->connection, buffer, ASR_BUFFER_SIZE, &size);
device_error = -2 (IDEVICE_E_UNKNOWN_ERROR)