libimobiledevice/idevicerestore #549: Error when executing

Rix565, 2022-10-30:
After building this project (and necessary libraries, there was a ton), when I want to execute it, I got this error:
```idevicerestore: symbol lookup error: idevicerestore: undefined symbol: plist_from_json```
I do not understand, I have all of the necessary libraries, all of the autogen.sh scripts worked fine without errors, building too, and I have this "undefined symbol error."

I tried reinstalling libplist, rebuilding it, same. Nothing seems to change at all.

Rix565, 2022-10-30:
EDIT: Same error for all libplist programs.

kuper0201, 2022-10-30:
From which os did you compile?

Rix565, 2022-10-31:
Ubuntu 22.04, x64, with an AMD A6-9225 processor.

Rix565, 2022-10-31:
BTW, if anyone needs to know for some reason, the device I'm trying to restore is an iPhone 3G S with a corruputed firmware (stuck at apple logo).

Rix565, 2022-10-31:
It might be a problem with the codebase itself, official artifacts from the repo are bugged with the same exact problem too.