libimobiledevice/idevicerestore #551: Restoring iPad Pro 11 (2nd gen) (iOS 16.1) fails

skortela, 2022-11-04:
Seems that restoring fails for iPad Pro 11.
Something about:
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.

Full debug log:
[erasure_failed.log](https://github.com/libimobiledevice/idevicerestore/files/9937217/erasure_failed.log)

a-gunter, 2022-11-09:
I had this issue sporadically on iOS 16.0 (on different models, not any model in particular) until the bRequest changes were made (961077074cf4f113d685ce19f3e0b18d343a74d6 and f6950c2461b28333bcef84a98f2132e93b216faf). Do you have those changes included in your idevicerestore build?

a-gunter, 2022-11-16:
I had a chance to try out 1 of these devices and saw the same thing. I don't have the device with me anymore though... will try to get a hold of another one. I guess it's good that this error happens pretty early on in the restore process so it may be easier to track down.

skortela, 2023-02-01:
Cannot reproduce anymore with iOS 16.3 image.

I wonder could the original root cause be related to this:
https://github.com/libimobiledevice/idevicerestore/issues/555

As 'iBEC' component size for iOS 16.1 was 1574400 bytes, which is dividable by 512. And device stops responding after that.
Maybe NULL packet needs to sent if transferred component is dividable by 512.

With iOS 16.3 image iBEC size was 1574590 bytes, so not dividable by 512. And process was able to continue after that.

closed 2023-02-01 by skortela