libimobiledevice/idevicerestore #555: iPhone 8 can't Upgrade/Restore iOS 16.1.2 : ERROR: Failed to place device in restore mode

vungocmobile, 2022-12-12:
irecv_event_cb: device 000e34c920d0e02e (udid: N/A) connected in Recovery mode
opening device 05ac:1281 @ 0xfa130000...
Setting to configuration 1
Setting to interface 0:0
Recovery Mode Environment:
iBoot build-version=iBoot-8419.40.112
iBoot build-style=RELEASE
Sending RestoreLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreLogo' in TSS response
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (13868 bytes)...
DEBUG: AVE is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'AVE' in TSS response
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H10.im4p (Firmware/ave/AppleAVE2FW_H10.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1082486 bytes)...
DEBUG: ISP is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'ISP' in TSS response
NOTE: No path for component ISP in TSS, will fetch from build_identity
Extracting adc-nike-d20.im4p (Firmware/isp_bni/adc-nike-d20.im4p)...
Personalizing IMG4 component ISP...
Tag found
Sending ISP (9842086 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreTrustCache' in TSS response
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 078-64668-116.dmg.trustcache (Firmware/078-64668-116.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (12865 bytes)...
ramdisk-size=0x20000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 078-64668-116.dmg (078-64668-116.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (102767817 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.d20ap.im4p (Firmware/all_flash/DeviceTree.d20ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (41472 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.d20.RELEASE.im4p (Firmware/all_flash/sep-firmware.d20.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (1516289 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone10 (kernelcache.release.iphone10)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (17959662 bytes)...
DEBUG: Waiting for device to disconnect...
ERROR: Failed to place device in restore mode
ERROR: Unable to place device into restore mode

Phone in recovery mode with logo Apple...

vungocmobile, 2022-12-13:
iOS 16.2 RC same error..

=======================================
::
:: Supervisor iBoot for d20, Copyright 2007-2022, Apple Inc.
::
::	Remote boot, Board 0x2 (d20ap)/Rev 0xf
::
::	BUILD_TAG: iBoot-8419.40.112
::
::	BUILD_STYLE: RELEASE
::
::	USB_SERIAL_NUMBER: SDOM:01 CPID:8015 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000E34C920xxxxxxx IBFL:3C SRNM:[xxxxxxxxxxx]
::
=======================================

3974bfd3d441da3:1343
Entering recovery mode, starting command prompt
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
b5255f381ca81f9:223
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
3d8efc6452c6cc3:417
ea0f64a4253252:901
3d8efc6452c6cc3:417
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
ea0f64a4253252:901
loaded ramdisk at <ptr> of size 0x6400000, from image at <ptr>
ea0f64a4253252:901
filesize variable invalid or not set, aborting
ea0f64a4253252:901
62039c63193986a:1102
ea0f64a4253252:901
ea0f64a4253252:901
7ab90c923dae682:1165
7ab90c923dae682:1255

rekyagis, 2022-12-14:
I have the same problem on iPhone 8 (Phone10,1) with iOS 16.1.2, but don't have with iOS 16.2 that is released today.
I don't have any problem on iPhone 14 Pro Max with both of iOS 16.1.2 and iOS 16.2.

I may have found why we have the error only with iPhone 8 + iOS 16.1.2.
iTunes on Windows does zero-byte usb bulk transfer only after sending RestoreDeviceTree(rdar) with iPhone 8 + iOS 16.1.2.
For iPhone 8 + iOS 16.1.2, the actual size to transfer of RestoreDeviceTree is 41472 bytes. Since It will be divided into 32768 bytes, it will be 32768+8704 bytes transfer normally but iTunes does 32768+8704+0 bytes transfer.
This happens only for "RestoreDeviceTree with iPhone 8 + iOS 16.1.2". It does not happens for any other data with iPhone 8 + iOS 16.1.2 nor for RestoreDeviceTree (the size is 41473, different from 16.1.2's) with iPhone 8 + iOS 16.2.
I'm not sure but the transfer size "41472" may require 0 byte transfer after sending.

The attached patch (for libirecovery) will fix the problem for now. You can restore iPhone 8 with iOS 16.1.2 with this patch.
It always add zero byte bulk transfer after sending restore data (not only for RestoreDeviceTree nor 41472 bytes). I don't have any problem with adding zero byte for all data.

Thanks.

[libirecovery_zero_bytes_transfer.patch.txt](https://github.com/libimobiledevice/idevicerestore/files/10225416/libirecovery_zero_bytes_transfer.patch.txt)

vungocmobile, 2022-12-14:
Yes 16.2 Final Working....

closed 2022-12-14 by vungocmobile

nikias, 2022-12-14:
I think this might have been "fixed" accidentally, because the size of the image changed so it's not a multiple of 512 anymore. If the zero length transfer fixes that, we need to integrate this into the code, but we also need to make sure it doesn't break on orher platforms.