libimobiledevice/idevicerestore #556: Error while recovering device

Akami-Slayer, 2022-12-18:
Hi
I've been trying to recover a device, iphone 6s (already recover another one just fine) but keep getting this error, does anyone knows what it means? 
`Found device in Recovery mode
Identified device as n61ap, iPhone7,2
Extracting BuildManifest from IPSW
Product Version: 12.5.6
Product Build: 16H71 Major: 16
INFO: device serial number is F17P2US0G5MC
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/home/akami/Downloads/iPhone_4.7_12.5.6_16H71_Restore/038-87263-049.dmg'
Found ECID 656219868735526
Getting ApNonce in recovery mode... e2 6b 3e 36 0a 83 f1 57 11 ca 0b 3e 1d 43 6f b7 66 f7 ef 2e 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 4c c0 ee 2b 7d 03 4c ee aa 2a 75 6c b5 a1 90 c3 e6 8e 0e 82 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n61.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731534 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12334 bytes)...
Extracting 038-87170-059.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9681 bytes)...
ramdisk-size=0x10000000
Extracting 038-87170-059.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (91715275 bytes)...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (125714 bytes)...
Extracting kernelcache.release.iphone7...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14069209 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device f5d74fa2c1fb70260c51bd1c37520fc0f722e1a6 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28672
UniqueChipID: 656219868735526
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n61.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n61.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
ERROR: Unable to receive message from FDR 0x559c29cc16c0 (-2). 0/2 bytes
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device`

Thanks in advance

nikias, 2022-12-18:
Can you make sure to use the latest code from git?

Akami-Slayer, 2022-12-18:
Yes, I'm using the latest code, even tried deleting and reinstalling everything from scratch, still getting the same error

`Found device in Recovery mode
Identified device as n61ap, iPhone7,2
The following firmwares are currently being signed for iPhone7,2:
  [1] 12.5.6 (build 16H71)
Select the firmware you want to restore: 1
Selected firmware 12.5.6 (build 16H71)
Verifying 'iPhone_4.7_12.5.6_16H71_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.5.6
Product Build: 16H71 Major: 16
INFO: device serial number is F17P2US0G5MC
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone_4.7_12.5.6_16H71_Restore/038-87263-049.dmg'
Found ECID 656219868735526
Getting ApNonce in recovery mode... db 34 b5 47 7a 6d d3 63 f4 ff cc b9 af e7 f6 0c df 03 d5 f4 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 86 3b 0b 0f 69 d0 7e 7a 17 e5 01 83 49 e7 31 54 2a e4 22 c7 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.n61.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731534 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12334 bytes)...
Extracting 038-87170-059.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9681 bytes)...
ramdisk-size=0x10000000
Extracting 038-87170-059.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (91715275 bytes)...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (125714 bytes)...
Extracting kernelcache.release.iphone7...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14069209 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device f5d74fa2c1fb70260c51bd1c37520fc0f722e1a6 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28672
UniqueChipID: 656219868735526
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.n61.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n61ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n61.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n61.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
ERROR: Unable to receive message from FDR 0x55d94a431760 (-2). 0/2 bytes
ERROR: Unable to send NORImageData data
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device`

nikias, 2022-12-18:
Well the thing is, the latest code will print the idevicerestore version as the first line of output. What does idevicerestore --version say?

Akami-Slayer, 2022-12-19:
I was on 1.0.0-128g7b89019

went ahead and deleted everything again installing from scratch thinking i was missing something but got the same error just different codes before the errors

`┌──(root㉿kali)-[/]
└─# idevicerestore --erase --latest
idevicerestore 1.0.0-128-g7b89019
Found device in Recovery mode
ECID: 656219868735526
Identified device as n61ap, iPhone7,2
The following firmwares are currently being signed for iPhone7,2:
  [1] 12.5.6 (build 16H71)
Select the firmware you want to restore: 1
Selected firmware 12.5.6 (build 16H71)
Downloading firmware (https://updates.cdn-apple.com/2021WinterFCS/fullrestores/012-51870/2A494DC9-8DE4-43F8-B084-87E91D5AF9BA/iPhone_4.7_12.5.6_16H71_Restore.ipsw)
downloading: 1%
downloading: 2%
downloading: 3%
downloading: 4%
downloading: 5%
downloading: 6%
downloading: 7%
downloading: 8%
downloading: 9%
downloading: 10%
downloading: 11%
downloading: 12%
downloading: 13%
downloading: 14%
downloading: 15%
downloading: 16%
downloading: 17%
downloading: 18%
downloading: 19%
downloading: 20%
downloading: 21%
downloading: 22%
downloading: 23%
downloading: 24%
downloading: 25%
downloading: 26%
downloading: 27%
downloading: 28%
downloading: 29%
downloading: 30%
downloading: 31%
downloading: 32%
downloading: 33%
downloading: 34%
downloading: 35%
downloading: 36%
downloading: 37%
downloading: 38%
downloading: 39%
downloading: 40%
downloading: 41%
downloading: 42%
downloading: 43%
downloading: 44%
downloading: 45%
downloading: 46%
downloading: 47%
downloading: 48%
downloading: 49%
downloading: 50%
downloading: 51%
downloading: 52%
downloading: 53%
downloading: 54%
downloading: 55%
[... 138 lines cut ...]
Extracting batterylow0@2x~iphone.im4p (Firmware/all_flash/batterylow0@2x~iphone.im4p)...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p (Firmware/all_flash/batterylow1@2x~iphone.im4p)...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p (Firmware/all_flash/glyphplugin@1334~iphone-lightning.im4p)...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n61ap.im4p (Firmware/all_flash/DeviceTree.n61ap.im4p)...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1334~iphone-lightning.im4p (Firmware/all_flash/recoverymode@1334~iphone-lightning.im4p)...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n61.RELEASE.im4p (Firmware/all_flash/iBoot.n61.RELEASE.im4p)...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n61.RELEASE.im4p (Firmware/all_flash/sep-firmware.n61.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n61.RELEASE.im4p (Firmware/all_flash/sep-firmware.n61.RELEASE.im4p)...
Personalizing IMG4 component SEP...
Sending NORData now...
ERROR: Unable to send NORData
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device

`