libimobiledevice/idevicerestore #560: ERROR: Unable to receive message from FDR 10E97BF8 (-4). 0/2 bytes iPhone XR

Acceptedapp, 2023-01-06:
15:09:20.671:  LibVersion: 20220922
15:09:20.671:  Start flash
15:09:20.671:  Checking PC DiskSpace
15:09:20.671:  Detecting Connection
15:09:20.674:  Found a device with mode: Recovery 
15:09:20.674:  Detecting ProductType
15:09:20.674:  matching device type through original information
15:09:20.675:  Checking iDevice firmware C:\3uTools\Firmware\iPhone11,8_16.2_20C65_Restore.ipsw
15:09:20.675:  Extract BuildManifest.plist
15:09:20.675:  Extracting Restore.plist from IPSW
15:09:20.686:  Firmware version: 16.2
15:09:20.686:  Product Build: 20C65 Major: 20
15:09:20.687:  Is Image4 supported: TRUE
15:09:20.687:  Extracting Firmware
15:09:20.691:  Reading ECID
15:09:20.691:  Found ECID 7886874997293102
15:09:20.691:  Reading NONCE
15:09:20.691:  Requesting SHSH
15:09:21.999:  Enter Recovery Mode
15:09:25.570:  Sending AppleLogo
15:09:26.796:  Sending Ramdisk
15:09:34.990:  Sending DeviceTree
15:09:35.009:  Sending RestoreSep
15:09:35.214:  Sending KernelCache
15:09:43.884:  Restoring
15:09:43.884:  About to restore device... 
15:09:46.424:  ERROR: build identity info does not contain a MacOSVariant
15:09:46.464:  Sending RootTicket
15:09:46.468:  Waiting for NAND
15:09:46.479:  Updating NAND Firmware
15:09:54.143:  Sending NORData
15:10:06.005:  ERROR: Unable to receive message from FDR 10E97BF8 (-4). 0/2 bytes

nikias, 2023-01-06:
This is not the output of idevicerestore. I cannot help you if you are not using the official tool, also make sure if you do that you build the latest code from source.

Chipatenii, 2023-08-18:
this likely has to do with some hardware issue. did you fix it?