libimobiledevice/idevicerestore #563: ERROR: Device did not reconnect in DFU or recovery mode. Possibly invalid iBSS. Reset device and try again.

Gahrlaag, 2023-02-03:
I've spent the better part of the evening trying to sort this out without any luck. I've put the device in DFU, it gets accepted, download finishes, it reboots but never enters recovery mode and I don't have enough time to put it back into DFU since the application exists after about 8 seconds. Any idea as to how to solve this? Tried with 4 different cables as well so I doubt it's a hardware issue

nikias, 2023-02-03:
You didn't provide the most important details. What platform are you running on, and what device is it?

Gahrlaag, 2023-02-03:
My apologies! I'm running on latest Linux mint and the device is a Apple iPad 9.7 (2018)

nikias, 2023-02-03:
Just to make sure, `idevicerestore --version` shows what?

Gahrlaag, 2023-02-04:
> Just to make sure, `idevicerestore --version` shows what?

1.0.0

nikias, 2023-02-04:
Then you have to compile an updated version of libirecovery and idevicerestore so we can make sure it's not because of old code...

Gahrlaag, 2023-02-04:
I assume 1.0.0-128-g7b89019 is the latest? no?

nikias, 2023-02-04:
Yes that is latest. You wrote 1.0.0 though which is not the same as 1.0.0-128-g7b89019. Anyway, can you run it with -d -k , then try again from DFU mode and upload the iBSS file?

Gahrlaag, 2023-02-04:
Yes, I had to update it to the latest, just wanted to make sure that the newest version is 1.0.0-128-g7b89019 :) Yessir! I'll get right on it

Gahrlaag, 2023-02-05:
Here's the response.

Setting to configuration 1
Setting to interface 0:0
DEBUG: tss_response_get_path_by_entry: No entry 'iBSS' in TSS response
NOTE: No path for component iBSS in TSS, will fetch from build_identity
Extracting iBSS.ipad7b.RELEASE.im4p (Firmware/dfu/iBSS.ipad7b.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Tag found
Writing data to iBSS
Sending iBSS (553781 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
irecv_event_cb: device 000329cc28d1803a (udid: N/A) disconnected
Waiting for device to reconnect...
ERROR: Device did not reconnect in DFU or recovery mode. Possibly invalid iBSS. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode

nikias, 2023-02-05:
This reminds me of why I have to improve logging with timestamps... how long is the delay between
```
Waiting for device to reconnect...
```
and
```
ERROR: Device did not reconnect in DFU or recovery mode. 
```
Also it wrote a file `iBSS`, can you attach it?

Gahrlaag, 2023-02-05:
[iBSS.txt](https://github.com/libimobiledevice/idevicerestore/files/10611781/iBSS.txt)

From the reconnect promt until it exits it takes about 8 seconds, ironically exactly as long as it takes me to set the unit in DFU mode :D Thing is, the application works. I tried it on another tablet I had laying around and it restored the ipad just fine so this has to do with the unit itself... I just can't for the life of me figure out how to get the damn thing to go into recovery mode. This happened after a failed update and it just...died.

nikias, 2023-02-05:
Hmm wait, your output suggests that the device is already in DFU mode. In that case ot sends iBSS and then you just have to wait to see what happens

Gahrlaag, 2023-02-05:
It is in DFU upon starting the application, that's correct. And after the iBSS is sent, it takes about 8 seconds before the application ends since the ipad doesn't go into recovery mode but probably just shuts down.  If I had time, I could probably try and manually put it back into DFU, but the timer is too short

nikias, 2023-02-05:
Well there is certainly a disconnect, that's what the log shows, but the question is what happens after that... can you run
```
watch -n 1 'lsusb| grep 05ac'
```
In another terminal and watch what happens? In the beginning it should show the 05ac:1227 entry.

Gahrlaag, 2023-02-06:
This is what I'm getting.
Bus 003 Device 006: ID 05ac:1227 Apple, Inc. Mobile Device (DFU Mode)

After that, the message simply disappears and nothing else comes up

nikias, 2023-02-07:
Hmm so it doesn't like the image at all, but then again usually it would re-appear in DFU mode and not just go offline. Not really sure what is going on, but the (stitched) iBSS image looks fine too. If you just reconnect the USB cable, would the device re-appear?

Gahrlaag, 2023-02-07:
It does not. It simply (I guess?) powers down and I do not have enough time between the iBSS send and application exit to re-initiate DFU mode manually.

b4bekkie, 2023-02-16:
i had fix this issue by simply first put device in recovery mode and then DFU check it with this method it will comletely fix

Orbif, 2023-03-03:
I have a similar issue with an iPad pro 5 WIFI on Windows, but only in recovery mode.

In normal mode, if I do an "idevicerestore -e -l", the restore works as expected.
[in_normal_mode.txt](https://github.com/libimobiledevice/idevicerestore/files/10885110/in_normal_mode.txt)

On the same device, if I do an "ideviceenterrecovery" and then an "idevicerestore -e -l", I have a failure with "ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again."
[in_recovery_mode.txt](https://github.com/libimobiledevice/idevicerestore/files/10885111/in_recovery_mode.txt)

RossComputerGuy, 2023-08-08:
Same issue, I have the device in recovery mode (`MacBookPro18,3`) but it keeps on saying the same error.

EvanCarroll, 2026-01-03:
Is this a dupe of #356