libimobiledevice/idevicerestore #569: macOS 13 Ventura not able to restore due to digest mismatch error

twocanoes, 2023-03-20:
Restoring firmware to a 2020 M1 MacBook Air fails with an error about a digest mismatch:

_get_and_save_large_file_internal: digest mismatch (expected = 9BC571DE280262FD15B073077735CD7792802827C47CE6C7DC7CA04320FB2A6D9101A6DD289715A56CA37C094D8CF940, actual = CCBB076FEC9D1D962C85E6F3CF4BAB55F665062A6F764F447C1E0BB267B0D872024138B1F77CB2FCEBED4CD0B5F04B18)

I have attached the full log.

To replicate the issue:
1. Install idevicerestore from source to a RPi 4 (4 GB) using this script: https://github.com/tperfitt/vdmtool/blob/master/idevicerestoresetup.sh
2. Put MacBook Air (2020 MacBook Air) into DFU mode and attach to RPi.
3. Unzip IPSW for 13.2 (or 13.2.1) to an accessible location (UniversalMac_13.2_22D49_Restore or UniversalMac_13.2.1_22D68_Restore).
4. run `idevicerestore -e UniversalMac_13.2_22D49_Restore` from that location.

After about 4 minutes, the process will fail.

However, following the same process with a macOS 12.6 (UniversalMac_12.6_21G115_Restore) will succeed.

[macos13.log](https://github.com/libimobiledevice/idevicerestore/files/11022972/macos13.log)

PicoMitchell, 2023-03-21:
I also got this exact same error when restoring macOS 13.2.1 while restoring 12.6 also worked for me.

During that testing with the 13.2.1 restore failure, I was using the 32-bit rPiOS.

I then tried on 64-bit rPiOS and restoring macOS 13.2.1 worked!

If my success can be reproduced by others, then using a 64-bit OS could be enough of a "solution" to this issue, but it's still odd that restoring one version works and another version doesn't.

Maybe someone which more knowledge of the inner workings of `idevicerestore` has some insight about why this may be failing on a 32-bit OS but working on a 64-bit OS? Could there be some build flag that can be applied to any of the tools to make it work on 32-bit OS (like is needed for `libzip` to be able to unzip large IPSWs on a 32-bit OS)?

potinlai, 2023-10-05:
I also face same problem. I tried adding `-D_FILE_OFFSET_BITS=64` when compiling libzip & idevicerestore, but result still failed.
Does anyone find the root cause? or a hint where the error `_get_and_save_large_file_internal: digest mismatch ...` comes from?