libimobiledevice/idevicerestore #575: iPhone SE 2020 (aka iPhone 12,8) Unable to place device into restore mode with iOS 16.4.1

N4zah, 2023-04-13:
Only on iOS 16.4.1 and only with iPhone SE 2020. I always get this error. I have the last build of libimobile (rebuilded today)

[outputSE2020.txt](https://github.com/libimobiledevice/idevicerestore/files/11220676/outputSE2020.txt)

00corgs, 2023-04-16:
I too faced the same problem.
I rebuilt idevicerestore and idevicerecovery but it did not solve the problem.

Device:iPhone SE 2020

xanadu-lee, 2023-04-17:
I had the same problem, tried sending 0 bytes from [https://github.com/libimobiledevice/idevicerestore/issues/555#issuecomment-1350554316]
but it didn't do anything.

xanadu-lee, 2023-05-03:
Please tell me when this issue will be fixed. . .

AKT04, 2023-05-04:
I had the same problem. i have update the idevicerestore.

Orbif, 2023-05-05:
I managed to have it working by adapting the patch proposed in issue #555 like this:

```
if (recovery_mode && length % 512 == 0) {
    bytes = 0;
    irecv_usb_bulk_transfer(client, 0x04, &buffer[length], 0, &bytes, USB_TIMEOUT);
}
```

Tested on Windows with iOS 16.4.1 ipsw for iPhone SE 2020.

closed 2023-05-05 by N4zah

nikias, 2023-05-08:
Thanks for confirming. I pushed a commit 12ed446fb17b8ea34728641cf4ca836eedab7aeb; even though the transfer has a size of 0 I didn't feel comfortable to use `&buffer[length]` since that points right after the actual buffer, so I just used `buffer` instead. It shouldn't make a difference - as I said - since the size is 0.

deyvs02, 2023-06-28:
> Only on iOS 16.4.1 and only with iPhone SE 2020. I always get this error. I have the last build of libimobile (rebuilded today)
> 
> [outputSE2020.txt](https://github.com/libimobiledevice/idevicerestore/files/11220676/outputSE2020.txt)

can you provide me with the files already compiled for windows?