libimobiledevice/idevicerestore #580: Failed to enter recovery mode.

Crepuscul0, 2023-06-26:
Hello, i'm trying to put my device in recovery mode with the following commands:
_$ ideviceenterrecovery -d *******************_

ideviceenterrecovery **********************
Telling device with udid ******************** to enter recovery mode.
Failed to enter recovery mode.
Device is successfully switching to recovery mode.

As you can see, it first says "Failed to enter recovery mode".  And then "Device is successfully switching to recovery mode.". 

But the device isn't rebooting or doing anything.

I'm currently running Fedora 38 and  ideviceenterrecovery 1.3.0.

I need help with this issue,  thank you so much.

nikias, 2023-06-26:
This is an old build. The newer code doesn't have that issue...

Crepuscul0, 2023-07-17:
I hope this message finds you well. I encountered during the compilation process of libirecovery. The ideviceenterrecovery binary is not being generated in the tools directory, despite successfully completing the installation.

Here are the details of the problem:

I have downloaded the latest version of libirecovery (version 1.3.0) from the official repository.

Following the provided instructions, I ran the ./autogen.sh and make commands to configure and compile the library.
The compilation process completed without any errors or warnings.

However, upon inspecting the tools directory (/home/victor/Downloads/libirecovery/tools), I discovered that the ideviceenterrecovery binary is not present.

I have attempted to uninstall and reinstall libirecovery, ensuring that all dependencies are met and following the recommended installation instructions. However, the issue persists.

Crepuscul0, 2023-07-17:
I hope this message finds you well. I encountered during the compilation process of libirecovery. The ideviceenterrecovery binary is not being generated in the tools directory, despite successfully completing the installation.

Here are the details of the problem:

I have downloaded the latest version of libirecovery from the official repository.

Following the provided instructions, I ran the ./autogen.sh and make commands to configure and compile the library.
The compilation process completed without any errors or warnings.

However, upon inspecting the tools directory (/libirecovery/tools), I discovered that the ideviceenterrecovery binary is not present.

I have attempted to uninstall and reinstall libirecovery, ensuring that all dependencies are met and following the recommended installation instructions. However, the issue persists.