libimobiledevice/idevicerestore #582: idevicerestore fails with devices that have SecurePageTableMonitor with 17.0 IPSWs

alfiecg24, 2023-07-09:
Trying to update with a pre-downloaded IPSW from 17.0b1 to 17.0b3 results in the following error:
```
idevicerestore 1.0.0-135-g609f7f0
Found device in Normal mode
ECID: XXXXXXXXXXXXXXXX
Identified device as d73ap, iPhone15,2
Extracting BuildManifest from IPSW
Product Version: 17.0
Product Build: 21A5277h Major: 21
Device supports Image4: true
Variant: Developer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 096-01403-042.dmg
[==================================================] 100.0%
Checking if device requires stashbag...
ERROR: img4_create_local_manifest: Unhandled component 'Ap,SecurePageTableMonitor' - can't create manifest
ERROR: Unable to create preboard manifest.
```
This only applies to device with SecurePageTableMonitor, which is apparently only A15 and A16 devices. Feel free to ask for any help with testing and I will happily respond to you. Thank you.

nirhen, 2023-09-21:
say - i have the same issue
did you manage to go around it 
BTW - flash with erase works fine

alfiecg24, 2023-09-21:
No, I haven’t managed to fix it as of now. I may look into it if I get any time but I ended up updating OTA instead,

alfiecg24, 2023-09-24:
Oh great, do you want to make a pull request to fix this?