libimobiledevice/idevicerestore #588: Refactor suggestion: making the TSS code from idevicerestore out into it's own library (libTSS essentially)

amarioguy, 2023-07-28:
Hello,

Given the scattered (and personally speaking not very clean) state of TSS tooling publicly available (like the high number of futurerestore forks), and how the iDeviceRestore TSS codebase seems to be the cleanest and most documented code available publicly for making TSS requests and getting responses back, I'd like to propose that the TSS portion of the code be made into it's own library (making a libTSS essentially, kind of like how we have a libirecovery right now)

I believe having this code be a separate library will allow TSS tooling in the iOS research community to be much more standardized and not reliant on fork after fork after fork of a utility. It would also help drastically with making any application that can make TSS requests in the future which can seriously aid doing research on the TSS server. Refactoring the code into its own library can allow changes to TSS requests required for newer devices to be made a lot easier without affecting the mainline iDeviceRestore project significantly. Given that currently, only idevicerestore's TSS code seems to request Cryptex tickets correctly (as in the manner that Apple itself requests Cryptex tickets), and how possibly more firmware will need a TSS response for a proper restore, I feel it's critical that libimobiledevice have a dedicated TSS library so these changes can be made independent of idevicerestore.

**I'd be more than happy and willing to do all of the refactoring work myself**, along with unit testing, proof of concept, CLI utility development on this, and anything else to make sure this refactor is up to the standards of the project.

I just wanted to bring this up here to make sure that it's known that _I want to do the work_ to make this happen, and if the refactored TSS code library can potentially be something that can be added to mainline libimobiledevice.

Any feedback on this refactor suggestion would be appreciated.

nikias, 2023-07-28:
Yes, I was planning to split out the TSS part for quite some time tbh. So feel free to make a pull request for it :)

Cryptiiiic, 2023-08-03:
I mean for refactor there's already 3 python versions and a c++ version I think. But if wanting to keep it c I understand.

amarioguy, 2023-08-03:
The main point of this is explicitly to have the separated TSS code be part of **upstream** libimobiledevice so working on making sure that idevicerestore upstream works with the split version of the TSS codebase.