libimobiledevice/idevicerestore #593: hang at Checkpoint 3844 complete with code 0

iOSPrincekin, 2023-09-02:
iphone 5s, iOS 12.5.7
macos 13.0
iPhone_4.0_64bit_12.5.7_16H81_Restore.ipsw
command:idevicerestore -i 6283712406792 -l
always hang at `Checkpoint 3844 complete with code 0`,any help will be a pleasure, Thanks a lot!

`
libimobiledevice 1.3.0-190-g6fc41f5
Found device in Recovery mode
ECID: 6283712406792
Identified device as n53ap, iPhone6,2
The following firmwares are currently being signed for iPhone6,2:
  [1] 12.5.7 (build 16H81)
Select the firmware you want to restore: 1
1
Selected firmware 12.5.7 (build 16H81)
Verifying 'iPhone_4.0_64bit_12.5.7_16H81_Restore.ipsw'...

Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.5.7
Product Build: 16H81 Major: 16
INFO: device serial number is DQGPWCEMFRC6
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone_4.0_64bit_12.5.7_16H81_Restore/038-87199-051.dmg'
Getting ApNonce in recovery mode... c3 23 ed 4f 7c 9b 37 b4 53 86 9c b6 7f 75 f5 6d 40 39 85 22 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 54 85 be 20 f3 e1 02 90 ce 9e 50 82 10 92 9d 93 22 c2 72 88 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.iphone6.RELEASE.im4p (Firmware/dfu/iBEC.iphone6.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (731057 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (11857 bytes)...
Extracting 038-87260-065.dmg.trustcache (Firmware/038-87260-065.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9336 bytes)...
ramdisk-size=0x10000000
Extracting 038-87260-065.dmg (038-87260-065.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (93958382 bytes)...
Extracting DeviceTree.n53ap.im4p (Firmware/all_flash/DeviceTree.n53ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (109009 bytes)...
Extracting sep-firmware.n53.RELEASE.im4p (Firmware/all_flash/sep-firmware.n53.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (2606433 bytes)...
Extracting kernelcache.release.iphone6 (kernelcache.release.iphone6)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13337438 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device dc4f477bd5602d9401b6c5ad6069c2c720ddcfd5 has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 35168
UniqueChipID: 6283712406792
[... 334 lines cut ...]
About to send BasebandData...
Sending BasebandData now...
Done sending BasebandData
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating baseband (19)
Updating Baseband completed.
Checkpoint 4294972163 complete with code 0
Checkpoint 4294972169 complete with code 0
Checkpoint 18446744069414589197 complete with code 0
Checkpoint 1589 complete with code 0
Checkpoint 1635 complete with code 0
Checkpoint 1596 complete with code 0
Checkpoint 1647 complete with code 0
Fixing up /var (17)
Checkpoint 3840 complete with code 0
Checkpoint 3841 complete with code 0
Checkpoint 3842 complete with code 0
Checkpoint 3846 complete with code 0
Checkpoint 3843 complete with code 0
Checkpoint 3844 complete with code 0

`

iOSPrincekin, 2023-09-02:
I find it hang at `sret = select(fd + 1, &fds, NULL, NULL, pto);` , but when I using i4Tools to flash the iPhone_4.0_64bit_12.5.7_16H81_Restore.ipsw to iPhone5s, has no this problem,I don't know why

`		
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00007ff80c119292 libsystem_kernel.dylib`__select + 10
  * frame #1: 0x000000010005f451 idevicerestore`socket_check_fd(fd=9, fdm=FDM_READ, timeout=90000) at socket.c:1129:11
    frame #2: 0x000000010005f727 idevicerestore`socket_receive_timeout(fd=9, data=0x00007ff7bfefd21c, length=4, flags=0, timeout=90000) at socket.c:1212:8
    frame #3: 0x0000000100066811 idevicerestore`usbmuxd_recv_timeout(sfd=9, data="", len=4, recv_bytes=0x00007ff7bfefd1bc, timeout=90000) at libusbmuxd.c:1609:18
    frame #4: 0x000000010005188e idevicerestore`internal_connection_receive_timeout(connection=0x0000600000c77540, data="", len=4, recv_bytes=0x00007ff7bfefd1bc, timeout=90000) at idevice.c:759:20
    frame #5: 0x0000000100051817 idevicerestore`idevice_connection_receive_timeout(connection=0x0000600000c77540, data="", len=4, recv_bytes=0x00007ff7bfefd1bc, timeout=90000) at idevice.c:846:9
    frame #6: 0x0000000100068b59 idevicerestore`service_receive_with_timeout(client=0x0000600000028090, data="", size=4, received=0x00007ff7bfefd218, timeout=90000) at service.c:166:33
    frame #7: 0x000000010006df4b idevicerestore`internal_plist_receive_timeout(client=0x0000600000028010, plist=0x00007ff7bfefd4e0, timeout=90000) at property_list_service.c:195:25
    frame #8: 0x000000010006e352 idevicerestore`property_list_service_receive_plist(client=0x0000600000028010, plist=0x00007ff7bfefd4e0) at property_list_service.c:273:9
    frame #9: 0x000000010007f49b idevicerestore`restored_receive(client=0x000060000028c6c0, plist=0x00007ff7bfefd4e0) at restore.c:157:24
    frame #10: 0x0000000100043251 idevicerestore`restore_device(client=0x0000000101304860, build_identity=0x0000600000c5ff00, filesystem="iPhone_4.0_64bit_12.5.7_16H81_Restore/038-87199-051.dmg") at restore.c:4261:19
    frame #11: 0x0000000100027baf idevicerestore`idevicerestore_start(client=0x0000000101304860) at idevicerestore.c:1396:12
    frame #12: 0x000000010002b978 idevicerestore`main(argc=0, argv=0x00007ff7bfeff5a0) at idevicerestore.c:1803:11
    frame #13: 0x00007ff80bdf541f dyld`start + 1903
`

nikias, 2023-10-07:
In this stage, idevicerestore is waiting for something coming from the device. If it doesn't then the device seems to get stuck, but I don't see any reason why idevicerestore would be the reason. Usually if "we" do something wrong, the process is aborted with an error.