libimobiledevice/idevicerestore #594: iPod touch 2nd generation not restoring

yesitsmaxwell, 2023-09-10:
Hi there,
I recently received an iPod touch 2nd generation which has been locked out. I can't use iTunes to restore it so I'm trying this tool.

The command I'm running is `sudo idevicerestore -e -l`, and I'm downloading iOS 4.2 (I also tried 4.1.2 but that put out the same error)

It errors out like this:
```
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 018-7058-113.dmg
[==================================================] 100.0%
checking for local shsh
no local file shsh/4158536652175-iPod2,1-4.1.shsh
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img3 tags to TSS request
ERROR: Unable to get SHSH blobs for this device
```

I don't know much about SHSH blobs or anything else it talks about so help would be appreciated!

Thanks!

DanTheMann15, 2023-09-12:
it would help a bit more if you wrote down what platform you are using e.g: macOS, Linux, Windows

The latest available firmware for iPod touch 2nd generation is: [iPod2,1_4.2.1_8C148_Restore.ipsw](https://secure-appldnld.apple.com/iPhone4/061-9855.20101122.Lrft6/iPod2,1_4.2.1_8C148_Restore.ipsw)

try doing it again with the iPod in DFU mode, here's a video tutorial on how to get into DFU mode:
https://youtu.be/rMujSKuKmnY

If you want to try iTunes, you can easily get it on Windows, i linked to the latest versions depending on what version of windows,
all of these are compatible with the iPod touch 2nd generation.

Latest version (Windows 10 and later): 
64bit: [iTunes64Setup.exe](https://secure-appldnld.apple.com/itunes12/032-75918-20230522-C36EFAE8-4B60-4DFC-AA7F-221E3BC91803/iTunes64Setup.exe)
32bit: [iTunesSetup.exe](https://secure-appldnld.apple.com/itunes12/032-75919-20230522-6B7A256F-6541-48EC-AA40-60D621885783/iTunesSetup.exe)

Last version for Windows 7 through 8.1: 
64bit: [iTunes64Setup.exe](https://secure-appldnld.apple.com/itunes12/001-80053-20210422-E8A3B28C-A3B2-11EB-BE07-CE1B67FC6302/iTunes64Setup.exe)
32bit: [iTunesSetup.exe](https://secure-appldnld.apple.com/itunes12/001-80042-20210422-E8A351F2-A3B2-11EB-9A8F-CF1B67FC6302/iTunesSetup.exe)

Last version for Windows XP and Vista:
32bit: [iTunesSetup.exe](https://secure-appldnld.apple.com/itunes12/031-34002-20150916-98D32A92-5C11-11E5-80AC-C25A6DA99CB1/iTunesSetup.exe)

Now, if you connect your device to your computer in DFU mode, iTunes will recognize it and will ask you to restore it, to that you can click yes, but if you already downloaded the IPSW you can say no, then hold down "SHIFT" and click "Restore", it'll open a window where you browse for your IPSW. select your IPSW and let iTunes do it's thing.

nikias, 2023-09-13:
Are you trying to restore from DFU mode? I just pushed a small fix for older devices, since in DFU mode there will not be an ApNonce available.