libimobiledevice/idevicerestore #597: Soft flash

nirhen, 2023-09-21:
Hi
	Soft flash (without erasing) failed due to 
	 ERROR: img4_create_local_manifest: Unhandled component 'Ap,SecurePageTableMonitor' - can't create manifest
	 
	 
	 also reported in https://github.com/libimobiledevice/idevicerestore/issues/582
	 
	 
	i added to  idevicerestore\src\img4.c
	 comp_tags component_tags[] = {
		{ "Ap,SecurePageTableMonitor", "sptm" },
		{ "Ap,TrustedExecutionMonitor", "txm" },
		
in order to fix issue

nikias, 2023-09-26:
I understood now why this happens. The component selection for the preboard manifest should not include those, but I haven't figured out yet what criteria are used exactly to determine which component to include.
I am seeing a logical or of the following values:
`IsFirmwarePayload | IsSecondaryFirmwarePayload | IsFUDFirmware | IsLoadedByiBoot | IsEarlyAccessFirmware | IsiBootEANFirmware | IsiBootNonEssentialFirmware`
So basically any one of these is true means it will be included, otherwise not. However I also see at least one being true for `Ap,SecurePageTableMonitor`, so something is missing in the logic...

nikias, 2023-12-11:
This should be resolved with most recent commits.

closed 2023-12-11 by nikias