libimobiledevice/idevicerestore #598: iPhone 15 model Restore issue with iOS 17.0.2 ERROR: Unable to send TSS request

NazKmet, 2023-09-28:
Hope app is well, and thank you owners and everyone who is supporting and maintaining this libraries.
We having issue with a device restore, here is helpful logs:

Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 087-09137-002.dmg
[==================================================] 100.0%
Getting ApNonce in recovery mode... 50 39 0d 56 87 ef ce c1 14 fe ba cd 47 c9 64 ac 43 3d 60 de 30 3d 2d dd 63 06 18 88 d7 97 38 34 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 66 21 44 a6 c0 c3 9c 64 5e 99 23 da 0f 0f 10 20 ed 09 19 c3 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=69&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=69, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

nikias, 2023-09-28:
Thank you for reporting. Would you mind reaching out to me on telegram so we can debug this issue? Thanks. You can find me in the https://t.me/libimobiledevice channel, please message me directly (Nikias / @pimskeks).

nikias, 2023-09-29:
I pushed commit cf22a1c0fa5994be2f02ed431461524499b06267, that should get us past the TSS error.

nikias, 2023-10-07:
This and some more issues have been fixed with the latest commits.

closed 2023-10-07 by nikias