libimobiledevice/idevicerestore #601: Unhandled component 'Ap,SecurePageTableMonitor' on iPad mini 6th gen

skortela, 2023-10-17:
Software update failed for iPad mini 6th gen.

```
[16:49:58.415][10968] : Found device in Normal mode
[16:49:58.469][10968] : ECID: 7956538753056798
[16:49:58.469][10968] : Identified device as j311ap, iPad14,2
[16:49:58.469][10968] : Extracting BuildManifest from IPSW
[16:49:58.489][10968] : Product Version: 17.0.3
[16:49:58.489][10968] : Product Build: 21A360 Major: 21
[16:49:58.494][10968] : Device supports Image4: true
[16:49:58.494][10968] : Variant: Customer Upgrade Install (IPSW)
[16:49:58.494][10968] : This restore will update the device without erasing user data.
[16:49:58.497][10968] : Checking IPSW for required components...
[16:49:58.497][10968] : All required components found in IPSW
[16:49:58.512][10968] : Checking if device requires stashbag...
[16:49:58.515][10968] : ERROR: img4_create_local_manifest: Unhandled component 'Ap,SecurePageTableMonitor' - can't create manifest
[16:49:58.516][10968] : ERROR: Unable to create preboard manifest.
```

skortela, 2023-10-19:
Cannot reproduce this anymore.

idevicerestore was run using erase flag, and after that cannot reproduce.

closed 2023-10-19 by skortela

nikias, 2023-10-19:
This only happens on update restore...