libimobiledevice/idevicerestore #602: ERROR: Unable to read OOB data from filesystem offset: No such file or directory

exploit3dguy, 2023-10-22:
Seems like on older versions (iOS 10 in this case) there is issue with reading the OOB data from filesystem offset, modern versions like iOS 15 work just fine. Old idevicerestore versions seem to work fine, something to do with the new version. Hope it gets fixed in the future. Thanks a lot 

Log:

Checkpoint started   id: 0x619 (unknown)
Checkpoint started   id: 0x61A (unknown)
Checkpoint started   id: 0x61B (unknown)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checkpoint started   id: 0x61D (unknown)
Checkpoint started   id: 0x64E (unknown)
Checkpoint started   id: 0x653 (unknown)
Checkpoint started   id: 0x61F (unknown)
Checkpoint started   id: 0x643 (unknown)
Checkpoint started   id: 0x620 (unknown)
Creating partition map (11)
Creating filesystem (12)
Creating filesystem (12)
Checkpoint started   id: 0x621 (unknown)
Checkpoint started   id: 0x65E (unknown)
Checkpoint started   id: 0x622 (unknown)
Checkpoint started   id: 0x65D (unknown)
Checkpoint started   id: 0x900 (unknown)
Checkpoint started   id: 0x901 (unknown)
Checkpoint started   id: 0x645 (unknown)
About to send filesystem...
Connected to ASR
Validating the filesystem
ERROR: Unable to read OOB data from filesystem offset: No such file or directory
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device

ASR STATUS: setup
ASR: Validating target...done
ASR STATUS: metadata
ASR: Validating source...
ASR: Can't gather image metadata
ASR: Could not validate source - error 110
ASR STATUS: fail

Merculous, 2023-10-22:
Not sure if this is directly related, but I'm also getting issues regarding "Authentication error" on my iPhone 3GS on 6.1.6. Also, something is wrong with doing custom restores too, all versions after iOS 5 give me the same error. This is my log.

```
executing /usr/sbin/asr -source asr://localhost:12345 -target /dev/disk0s1s1 -erase -noprompt --chunkchecksum --puppetstrings
ASR: asr: Unable to disable idle sleep - -536870199
ASR STATUS: start    241.1    multicast-client
ASR: Waiting for connection attempt from server
void AppleUSBDeviceMux::handleConnectResult(struct BulkUSBMuxSession *, errno_t) new session to port 12345 failed: 61
ASR STATUS: setup
ASR: Validating target...done
ASR STATUS: metadata
ASR: Validating source...
AppleUSBDeviceMux::handleConnectResult new session 0xa120ee18 established 12345<-lo0->49162 12345<-usb->11
ASR: Can't gather image metadata
ASR: Could not validate source - error 110
ASR STATUS: fail

ERROR: Unable to restore device
```

nikias, 2023-11-22:
I updated the code with commit c871c591e36d2a4083e3dda4c70144a0321ce70f about two weeks ago. Problem is that it can't seek in compressed files in the zip archive, and older versions used compressed (vs. stored) root file system images. 
Feel free to re-open if it should be a different issue.

closed 2023-11-22 by nikias