libimobiledevice/idevicerestore #603: Failing to restore iPhone 12 with `iPhone13,2,iPhone13,3_17.0.3_21A360_Restore.ipsw`

doronz88, 2023-10-24:
Using the latest version of `idevicerestore`:

```shell
idevicerestore -e -d '~/Downloads/iPhone13,2,iPhone13,3_17.0.3_21A360_Restore.ipsw'
```

I'm getting the following output: [idevicerestore-log.txt](https://github.com/libimobiledevice/idevicerestore/files/13118856/idevicerestore-log.txt).

I'm not sure at what phase started the error, but I tried to debug it. I believe the first line to give a hint about the error is:

```
[07:55:46.0779-GMT]{3>5} CHECKPOINT WARNING: FIRMWARE_SEALING:[0x1500] decompress_pearl_data missing prf1
```

I guess it's because of a change in the `T200` updater (since its the last component to be sent before the error).
Any ideas?

> NOTE: Updating iPhone 14 Pro Max to the same version succeeds. It occurs on my iPhone 12 specifically

doronz88, 2023-10-29:
**TL;DR: iPhone 12 doesn't support `.ipsw` updates as of iOS `17.0.3`**

Okay so I'm just updating Apple Configurator also fails on the same exact error - but somehow ignores it and advancing the state to send the filesystem (ASR), though it doesn't allow the device the boot afterwards.

Now I'm sure this is definetley an Apple's bug for the specific device. I'll keep this issue open for in case someone else gets this error.

nikias, 2023-10-29:
Maybe there is a mixup with the research devices?

doronz88, 2023-10-29:
I'm trying to update a production device. Anyhow this is a pretty serious bug since this means Apple leaves this device in brick state if the user tried to update via iTunes (instead of OTA)

nikias, 2023-10-29:
Yes I saw it's a production device, but maybe Apple messed something up with it, I believe it's the same ipsw.

hiepnh0, 2023-11-17:
@doronz88 @nikias 
The same error on iOS 17.1.1 (iPhone 12)