libimobiledevice/idevicerestore #605: ERR0R : Device failed to enter restore mode.

Jiro0262, 2023-11-02:
Hello,
I try to use idevicerestore with a Macbook Pro 2022 M2 13" and i'm blocked at : 
`ERR0R : Device failed to enter restore mode.`
`Please make sure that usbmuxd is running.`

I already tryied everything from others posts (`sudo usbmuxd`, etc.).
Btw, time between : ``Waiting for device to enter restore mode...`` and  ``irecv_event_cb: device 001071e93421401e (udid: N/A) connected in Recovery mode`` is almost instant.

Feel free to ask me everything like logs, etc.
Thanks !

kode54, 2023-11-08:
Try running:

```
sudo usbmuxd -f
```

in another terminal, then:

```
sudo idevicerestore <your parameters>
```

Yes, sudo idevicerestore too.

Jiro0262, 2023-11-08:
Thank you for your reply. I couldn't try since i've restore my device with another mac since.

closed 2023-11-08 by Jiro0262