libimobiledevice/idevicerestore #613: Restore iPhone blocked in Demo Mode

Auties00, 2023-11-26:
Hello,
I recently bought a pile of used broken iphones. After fixing one of them, I discovered that this iPhone was once a Demo device as it displays the message "This iPhone is property of an Apple Authorized Reseller and is not for sale". I checked the IMEI of the iphone and it isn't flagged as stolen, blocked or locked and so it can be activated . So I decided to reset the iPhone through iTunes, but on the second loading screen(https://support.apple.com/en-us/102317), the phone stops the reset process and returns to the screen I mentioned at the beginning of this post. I'm pretty sure this is a local issue as the iPhone isn't connected to the internet and even if it were, the imei is clean so it should be able to reset. After trying itunes, I tried to use idevicerestore, but it ended just as it did with iTunes. I was wondering then if there's any way to forcefully reset the phone using this library. My only other option is to call Apple and convince them to unenroll the phone from Demo mode, but I'm pretty sure that would be hard to do, so I figured I might as well ask.

Good-ToSeeYou, 2025-01-15:
Did you have a solution of this? I bought 5 phones on FB Marketplace from a reseller and one 15 + is saying Property of apple. I tried to contact the seller but I am having no response. I contacted apple and there is nothing they can do. They checked the phone and there is no block no ban just the message. Can you Help?

Auties00, 2025-01-15:
> Did you have a solution of this? I bought 5 phones on FB Marketplace from a reseller and one 15 + is saying Property of apple. I tried to contact the seller but I am having no response. I contacted [...]

I also contacted Apple, but they couldn’t do anything and wouldn’t even take it back(lol). I managed to fix it by swapping out the memory of the device, there seems to be no other way because the phone will just refuse to update in any other way(I’ve truly tried everything) or to be even wiped, even though there is no lock on the device itself or anything linked to its imei. There could be a way to do it if you waste enough time reverse engineering the protocol, but it seemed like way too much work. I might try in the future though just because I’m still curious

Good-ToSeeYou, 2025-01-15:
Thank you for the quick response.
There are people that do it in India but it is too expensive. $200 USD

DaddyKrauze, 2025-01-24:
sry, you told that you swap memory and that solve the problem?

closed 2025-01-24 by Auties00