libimobiledevice/idevicerestore #615: STATUS=94&MESSAGE=This device isn't eligible for the requested build.

sybetjeh, 2023-12-01:
Hi

When Trying to restore my iPhone 11 i'm constantly greeted with

Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)

I've downloaded the latest version of the IPSW file from https://ipsw.me/

Any ideas why it keeps saying it?

sybetjeh, 2023-12-01:
even when letting idevicerestore download it themselves makes it go the same route

Merculous, 2024-06-17:
Just recompiled with latest dependencies and using commit `10cd5f7f0fe14fbf51f2142ea2df153da33d1a21`, I'm getting the same error for MSYS2 Clang64. This is for iPhone3,3 for 7.1.2.

This is the command I ran: `idevicerestore -e -l -y -d`

This also includes the new dependency libtatsu with commit `816fb135aca1aab300f6fb98c8a5c7296da160e0`

nikias, 2024-06-17:
Ok can you run with `-dd` and upload the complete console output somewhere? I mean technically I only need the TSS request XML plist, so feel free to post only that.

nikias, 2024-06-17:
I released a new version of libtatsu, use latest commit and it should work again.

Merculous, 2024-06-18:
Issue is fixed with libtatsu 1.0.3.

nikias, 2024-06-18:
Great!

closed 2024-06-18 by nikias