libimobiledevice/idevicerestore #619: Unable to send data to ASR. Sent 0

Vollthrottle, 2023-12-22:
Unable to send data to ASR. Sent 0
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device
[LogPM14Black_error12.23root@heinz-Latitude-E5520homeheinzD.txt](https://github.com/libimobiledevice/idevicerestore/files/13753969/LogPM14Black_error12.23root%40heinz-Latitude-E5520homeheinzD.txt)
[LogPM14Bsilver_error12.23root@heinz-Latitude-E5520homeheinzD.txt](https://github.com/libimobiledevice/idevicerestore/files/13753970/LogPM14Bsilver_error12.23root%40heinz-Latitude-E5520homeheinzD.txt)

By:
idevicerestore --latest
devicees.   
2x iphone14pm256
linux on dell laptop usbA 22.04 /(same on debian)/UBUNTU

log.
The following firmwares are currently being signed for iPhone15,3:
  [1] 17.2.1 (build 21C66)
Verifying 'iPhone15,3_17.2_21C62_Restore.ipsw'...
Checksum matches.

Sending iBEC (1355134 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-10151.62.1
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2796~iphone.im4p (Firmware/all_flash/applelogo@2796~iphone.im4p)...

ReverseProxy[Conn]: (status=5) Connected to gg.apple.com:443
Checkpoint completed id: 0x67D (fdr_auto_challenge_claim) result=0
Checkpoint started   id: 0x635 (update_device_firmware)
Checkpoint started   id: 0x1300 (update_iBoot)
Unknown operation (80)
Found IsiBootEANFirmware component DCP
Sending IsiBootEANFirmware image list
Extracting iphone15dcp.im4p (Firmware/dcp/iphone15dcp.im4p)...
Personalizing IMG4 component DCP...

Sending filesystem now...
[==================                                ]  35.0%ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send data to ASR. Sent 0 of 131092 bytes.
ERROR: Unable to send filesystem payload
Done sending filesystem
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to restore device

Vollthrottle, 2023-12-22:
[LogPM14Black_error12.23root@heinz-Latitude-E5520homeheinzD.txt](https://github.com/libimobiledevice/idevicerestore/files/13753991/LogPM14Black_error12.23root%40heinz-Latitude-E5520homeheinzD.txt)
[LogPM14Bsilver_error12.23root@heinz-Latitude-E5520homeheinzD.txt](https://github.com/libimobiledevice/idevicerestore/files/13753992/LogPM14Bsilver_error12.23root%40heinz-Latitude-E5520homeheinzD.txt)

yannb2159, 2024-06-12:
I have the same problem as you, have you found a solution ?

navarrojhanrafael-beep, 2026-04-26:
> I have the same problem as you, have you found a solution ?

any updates on this kind of problem?