libimobiledevice/idevicerestore #621: Unable to build

ianmmyers, 2024-02-06:
I'm attempting to build on MacOS Ventura 13.6.4, while using as many pre-built packages from Homebrew as possible. All appears to go fine until I attempt to make and I get the following

```
6802-IMY-1:idevicerestore imy$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
  CC       idevicerestore-ftab.o
  CC       idevicerestore-ipsw.o
ipsw.c:1180:2: warning: 'SHA1_Init' is deprecated [-Wdeprecated-declarations]
        SHA1_Init(&sha1ctx);
        ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/sha.h:49:1: note: 'SHA1_Init' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Init(SHA_CTX *c);
^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:194:49: note: expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:62:52: note: expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1184:3: warning: 'SHA1_Update' is deprecated [-Wdeprecated-declarations]
                SHA1_Update(&sha1ctx, (const void*)buf, sz);
                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/sha.h:50:1: note: 'SHA1_Update' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Update(SHA_CTX *c, const void *data, size_t len);
^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:194:49: note: expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:62:52: note: expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
ipsw.c:1186:2: warning: 'SHA1_Final' is deprecated [-Wdeprecated-declarations]
        SHA1_Final(tsha1, &sha1ctx);
        ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/sha.h:51:1: note: 'SHA1_Final' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Final(unsigned char *md, SHA_CTX *c);
^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:194:49: note: expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:62:52: note: expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
3 warnings generated.
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
  CC       idevicerestore-recovery.o
  CC       idevicerestore-restore.o
  CC       idevicerestore-asr.o
asr.c:349:3: warning: 'SHA1_Init' is deprecated [-Wdeprecated-declarations]
                SHA1_Init(&sha1);
                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/sha.h:49:1: note: 'SHA1_Init' has been explicitly marked deprecated here
OSSL_DEPRECATEDIN_3_0 int SHA1_Init(SHA_CTX *c);
^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:194:49: note: expanded from macro 'OSSL_DEPRECATEDIN_3_0'
#   define OSSL_DEPRECATEDIN_3_0                OSSL_DEPRECATED(3.0)
                                                ^
/opt/homebrew/Cellar/openssl@3/3.2.1/include/openssl/macros.h:62:52: note: expanded from macro 'OSSL_DEPRECATED'
#     define OSSL_DEPRECATED(since) __attribute__((deprecated))
                                                   ^
1 warning generated.
  CC       idevicerestore-fdr.o
  CC       idevicerestore-limera1n.o
  CC       idevicerestore-download.o
  CC       idevicerestore-locking.o
  CCLD     idevicerestore
Making all in docs
make[2]: Nothing to be done for `all'.
make[2]: Nothing to be done for `all-am'.
6802-IMY-1:idevicerestore imy$ 
```

With this only being warnings I would expect that the build would have completed, but there is no idevicerestore binary, so I'm guessing that flags are set such that this won't build if warnings are encountered.

Can anyone point me in the right direction on how to get this to compile?

ianmmyers, 2024-02-07:
upon further digging I found that I am getting the object files created, it's just the linking that seems to be failing. I did this
```

6802-IMY-1:src imy$ make V=2
/bin/sh ../libtool  --tag=CC   --mode=link gcc -Wno-multichar -O2  -I/opt/homebrew/Cellar/libirecovery/1.1.0/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libimobiledevice/HEAD-73b6fd1_2/include -I/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/openssl@3/3.2.1/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libzip/1.10.1/include  -I/opt/homebrew/Cellar/openssl@3/3.2.1/include  -g -O2 -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include  -L/opt/homebrew/Cellar/libirecovery/1.1.0/lib -lirecovery-1.0 -L/opt/homebrew/Cellar/libimobiledevice/HEAD-73b6fd1_2/lib -L/opt/homebrew/Cellar/libplist/HEAD-082b69d/lib -limobiledevice-1.0 -lplist-2.0 -L/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/lib -lusbmuxd-2.0 -L/opt/homebrew/Cellar/libplist/HEAD-082b69d/lib -lplist-2.0 -L/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/lib -L/opt/homebrew/Cellar/libplist/HEAD-082b69d/lib -limobiledevice-glue-1.0 -lplist-2.0 -L/opt/homebrew/Cellar/libzip/1.10.1/lib -lzip -lz -L/opt/homebrew/Cellar/openssl@3/3.2.1/lib -lssl -lcrypto -lcurl  -o idevicerestore idevicerestore-idevicerestore.o idevicerestore-common.o idevicerestore-tss.o idevicerestore-fls.o idevicerestore-mbn.o idevicerestore-img3.o idevicerestore-img4.o idevicerestore-ftab.o idevicerestore-ipsw.o idevicerestore-normal.o idevicerestore-dfu.o idevicerestore-recovery.o idevicerestore-restore.o idevicerestore-asr.o idevicerestore-fdr.o idevicerestore-limera1n.o idevicerestore-download.o idevicerestore-locking.o   
libtool: link: gcc -Wno-multichar -O2 -I/opt/homebrew/Cellar/libirecovery/1.1.0/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libimobiledevice/HEAD-73b6fd1_2/include -I/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/openssl@3/3.2.1/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/include -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -I/opt/homebrew/Cellar/libzip/1.10.1/include -I/opt/homebrew/Cellar/openssl@3/3.2.1/include -g -O2 -I/opt/homebrew/Cellar/libplist/HEAD-082b69d/include -o idevicerestore idevicerestore-idevicerestore.o idevicerestore-common.o idevicerestore-tss.o idevicerestore-fls.o idevicerestore-mbn.o idevicerestore-img3.o idevicerestore-img4.o idevicerestore-ftab.o idevicerestore-ipsw.o idevicerestore-normal.o idevicerestore-dfu.o idevicerestore-recovery.o idevicerestore-restore.o idevicerestore-asr.o idevicerestore-fdr.o idevicerestore-limera1n.o idevicerestore-download.o idevicerestore-locking.o  -L/opt/homebrew/Cellar/libirecovery/1.1.0/lib -lirecovery-1.0 -L/opt/homebrew/Cellar/libimobiledevice/HEAD-73b6fd1_2/lib -L/opt/homebrew/Cellar/libplist/HEAD-082b69d/lib -limobiledevice-1.0 -L/opt/homebrew/Cellar/libusbmuxd/2.0.2_1/lib -lusbmuxd-2.0 -L/opt/homebrew/Cellar/libimobiledevice-glue/1.0.0/lib -limobiledevice-glue-1.0 -lplist-2.0 -L/opt/homebrew/Cellar/libzip/1.10.1/lib -lzip -lz -L/opt/homebrew/Cellar/openssl@3/3.2.1/lib -lssl -lcrypto -lcurl
```

which did not link, but I got the command that make is running. I then tried to run that command to libtool manually (ie outside of make) and still got no binary and no error.

nikias, 2024-02-07:
Your original post does not show any linkkng error. Have you checked if src/.libs/idevicerestore is produced?

ianmmyers, 2024-02-07:
> Your original post does not show any linkkng error. Have you checked if src/.libs/idevicerestore is produced?

@nikias Thanks for the reply. Yeah, I checked and it's not present. .libs exists but is empty.

ianmmyers, 2024-02-07:
I built this on a second system, noticed I had the binary, looked again, and saw it was gone. So I did it again, and got the same - the binary is there briefly then getting deleted. Once I saw this I checked on my original setup and saw the same; something is deleting it.

I am doing this on corporate-managed laptops so it's possible that something is quarantining it. I can try on an unmanaged Mac and see if I get the same results or not.

ianmmyers, 2024-02-07:
I was able to confirm that the binary is getting quarantined, so the issue is on my end. Thanks for the help.

closed 2024-02-07 by ianmmyers