libimobiledevice/idevicerestore #625: Is this supposed to work on a Mac?  MacOS 11.5.7.

azav, 2024-03-15:
In the terminal, I get this error on the first step. trying `sudo apt-get install`.

sudo: apt-get: command not found

Can we get some better instructions on what is required to install and execute this project?  I'm at a loss in understanding what is expected to already be in place and where we go to set up and install this package.  

Thank you.
Alex Zavatone

circularsprojects, 2024-03-18:
you don't need to run that command if you're on a mac, it literally says "ubuntu/debian" above the command

azav, 2024-03-19:
So, what do you do then?

nikias, 2024-03-19:
Best way to make sure you have the latest code is by compiling all the dependencies yourself. As a matter of fact I created a build script for macOS:
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L https://is.gd/limdmacos
bash ./limd-build-macos.sh
```
That will build all the libimobiledevice libs and tools and install them in `/usr/local` prefix by default.

azav, 2024-03-19:
Thank you.  That appeared to build.  Now to go through the docs for how to use this.  I hope it works on the iPad 4 with iOS 10.x.