libimobiledevice/idevicerestore #628: Fail to restore. ./configure: line 8888: syntax error near unexpected token `limd_glue,'

dsdanielpark, 2024-03-24:
```
./configure: line 8888: syntax error near unexpected token `limd_glue,'
./configure: line 8888: `PKG_CHECK_MODULES(limd_glue, libimobiledevice-glue-1.0 >= $LIMD_GLUE_VERSION)'
```

I encounter this error during restore iphoin-13-mini.

nikias, 2024-03-24:
This error is not during restore. It happens when you try to build the tool. What platform are you on?

dsdanielpark, 2024-03-24:
OS: Ubuntu 21.02
I follow readme. 

What possible error in this procedure? I try to install

nikias, 2024-03-25:
You don't have libimobiledevice-glue-dev installed. I would suggest to build it from source. See instructions at https://github.com/libimobiledevice/libimobiledevice-glue.
Apart from that I would strongly suggest to build everything from source; I have this script that you can try to use: 

```shell
mkdir limd-build
cd limd-build
wget -O limd-build-linux.sh https://is.gd/limdlinux
bash ./limd-build-linux.sh
```

dsdanielpark, 2024-03-26:
[nikias](https://github.com/nikias)

First, I sincerely thank you for this wonderful project and your passionate assistance. I greatly respect the hard and demanding work involved in contributing to the open-source community.

I manually built all the required packages as per the manual, but I faced the following error message: (After connect Iphone using USB cable)
```
(base) user@DESKTOP-A:~/idevicerestore$ idevicerestore --latest
idevicerestore 1.0.0-178-ge4a5ac4
Segmentation fault (core dumped)
```

dsdanielpark, 2024-03-26:
[image]

Thanks for your support!

nikias, 2024-03-26:
Oh, my bad. The command line is a bit different:
```
gdb --args idevicerestore --latest
```
Then once you get the gdb prompt, type `r` and press ENTER. This will start the process, then when it stops (it will print something about segmentation fault), type `bt` and press ENTER.

dsdanielpark, 2024-03-26:
[image]

Here it is!

nikias, 2024-03-26:
Hmm it looks like this is inside libusb-1.0... never seen that before. I saw now that you are using this inside WSL, did you know you can compile idevicerestore directly for Windows too?

dsdanielpark, 2024-03-26:
Is it possible to compile in windows directly? Maybe WSL2 have trouble to access device port.

nikias, 2024-03-26:
Not entirely sure why it is crashing (crashes when trying to access a lock, so it's not even trying to talk to any usb device at all). In regards to building this on Windows, you need to compile all the components. Compilation is done with MSYS2 environment: https://www.msys2.org/
Download the MSYS2 installer and follow the installation steps.

Run the MSYS2 MinGW 64-bit shell and make sure the required dependencies are installed:
```
pacman -S base-devel curl git mingw-w64-x86_64-gcc make libtool autoconf automake-wrapper liblzma openssl-devel mingw-w64-x86_64-readline
```
Now, build all the libs and tools of the libimobiledevice project:
```
export PKG_CONFIG_PATH=/local/lib/pkgconfig

git clone https://github.com/libimobiledevice/libplist
cd libplist
./autogen.sh --enable-debug --without-cython
make
make install
cd ..

git clone https://github.com/libimobiledevice/libimobiledevice-glue
cd libimobiledevice-glue
./autogen.sh
make
make install
cd ..

git clone https://github.com/libimobiledevice/libusbmuxd
cd libusbmuxd
./autogen.sh
make
make install
cd ..

git clone https://github.com/libimobiledevice/libimobiledevice
cd libimobiledevice
./autogen.sh --enable-debug
make
make install
cd ..

git clone https://github.com/libimobiledevice/libirecovery
cd libirecovery
./autogen.sh
make
make install
cd ..

git clone https://github.com/libimobiledevice/idevicerestore
cd idevicerestore
curl -o libzip-static.b64 -Ls "https://gist.github.com/nikias/3da15d03120382f87b44029cd8495a02/raw/99cd8138fed99e8f6530b6f179f787342c698e1f/libzip-1.7.1_static_x86_64-mingw64.tar.bz2"
base64 -d < libzip-static.b64 > libzip-static.tar.bz2
mkdir deps
tar -C deps -xjf libzip-static.tar.bz2
curl -o libcurl-static.b64 -Ls "https://gist.github.com/nikias/6c397a0a2f4f4eafd91b81cccd22b761/raw/85216e60af6787f3b351291165eb91bd585ff09a/libcurl-8.1.0-static-x86_64-mingw64.tar.bz2"
base64 -d < libcurl-static.b64 > libcurl-static.tar.bz2
tar -C deps -xjf libcurl-static.tar.bz2
./autogen.sh libzip_VERSION=1.7.1 libzip_CFLAGS="-I`pwd`/deps/include" libzip_LIBS="`pwd`/deps/lib/libzip.a /mingw64/lib/libbz2.a /mingw64/lib/liblzma.a" zlib_LIBS="/mingw64/lib/libz.a" libcurl_CFLAGS="-I`pwd`/deps/include" libcurl_LIBS="`pwd`/deps/lib/libcurl.a /mingw64/lib/libzstd.a -lws2_32 -lcrypt32 -lwldap32 -lbcrypt -lssl -lcrypto"
make
make install
cd ..
```
Now if everything worked you should be able to run directly from the MSYS2 MinGW 64-bit shell:
```
idevicerestore --version
```

dsdanielpark, 2024-03-27:
[image]

When I make `libirecovery`, I see this error messages

dsdanielpark, 2024-03-27:
Thank you again!! last 3 row, I have some error.

This is full history log!
[history.txt](https://github.com/libimobiledevice/idevicerestore/files/14770765/history.txt)

```
./autogen.sh libzip_VERSION=1.7.1 libzip_CFLAGS="-I`pwd`/deps/include" libzip_LIBS="`pwd`/deps/lib/libzip.a /mingw64/lib/libbz2.a /mingw64/lib/liblzma.a" zlib_LIBS="/mingw64/lib/libz.a" libcurl_CFLAGS="-I`pwd`/deps/include" libcurl_LIBS="`pwd`/deps/lib/libcurl.a /mingw64/lib/libzstd.a -lws2_32 -lcrypt32 -lwldap32 -lbcrypt -lssl -lcrypto"
make
make install
cd ..
```

```
user@DESKTOP-xxx MINGW64 ~/idevicerestore
$ make
make  all-recursive
make[1]: Entering directory '/home/parkm/idevicerestore'
Making all in src
make[2]: Entering directory '/home/parkm/idevicerestore/src'
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
In file included from tss.c:28:
tss.c: In function 'tss_request_send':
C:/msys64/home/parkm/idevicerestore/deps/include/curl/typecheck-gcc.h:50:15: warning: call to '_curl_easy_setopt_err_long' declared with attribute warning: curl_easy_setopt expects a long argument for this option [-Wattribute-warning]
   50 |               _curl_easy_setopt_err_long();                             \
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/home/parkm/idevicerestore/deps/include/curl/curl.h:46:7: note: in definition of macro 'CURL_IGNORE_DEPRECATION'
   46 |       statements \
      |       ^~~~~~~~~~
tss.c:1558:17: note: in expansion of macro 'curl_easy_setopt'
 1558 |                 curl_easy_setopt(handle, CURLOPT_POSTFIELDSIZE, strlen(request));
      |                 ^~~~~~~~~~~~~~~~
  CC       idevicerestore-fls.o
  CC       idevicerestore-mbn.o
  CC       idevicerestore-img3.o
  CC       idevicerestore-img4.o
  CC       idevicerestore-ftab.o
  CC       idevicerestore-ipsw.o
ipsw.c: In function 'sha1_verify_fp':
ipsw.c:1180:9: warning: 'SHA1_Init' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 1180 |         SHA1_Init(&sha1ctx);
      |         ^~~~~~~~~
In file included from ipsw.c:38:
C:/msys64/usr/include/openssl/sha.h:49:27: note: declared here
   49 | OSSL_DEPRECATEDIN_3_0 int SHA1_Init(SHA_CTX *c);
      |                           ^~~~~~~~~
ipsw.c:1184:17: warning: 'SHA1_Update' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 1184 |                 SHA1_Update(&sha1ctx, (const void*)buf, sz);
      |                 ^~~~~~~~~~~
C:/msys64/usr/include/openssl/sha.h:50:27: note: declared here
   50 | OSSL_DEPRECATEDIN_3_0 int SHA1_Update(SHA_CTX *c, const void *data, size_t len);
      |                           ^~~~~~~~~~~
ipsw.c:1186:9: warning: 'SHA1_Final' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
 1186 |         SHA1_Final(tsha1, &sha1ctx);
      |         ^~~~~~~~~~
C:/msys64/usr/include/openssl/sha.h:51:27: note: declared here
   51 | OSSL_DEPRECATEDIN_3_0 int SHA1_Final(unsigned char *md, SHA_CTX *c);
      |                           ^~~~~~~~~~
  CC       idevicerestore-normal.o
  CC       idevicerestore-dfu.o
  CC       idevicerestore-recovery.o
  CC       idevicerestore-restore.o
  CC       idevicerestore-asr.o
asr.c: In function 'asr_send_payload':
asr.c:349:17: warning: 'SHA1_Init' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  349 |                 SHA1_Init(&sha1);
      |                 ^~~~~~~~~
In file included from asr.c:34:
C:/msys64/usr/include/openssl/sha.h:49:27: note: declared here
[... 41 lines cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-tss.o: in function `tss_request_send':
C:/msys64/home/parkm/idevicerestore/src/tss.c:1535:(.text+0x4120): undefined reference to `__imp_curl_slist_append'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/tss.c:1550:(.text+0x412a): undefined reference to `__imp_curl_easy_setopt'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/tss.c:1530:(.text+0x4130): undefined reference to `__imp_curl_easy_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/tss.c:1570:(.text+0x4250): undefined reference to `__imp_curl_easy_perform'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/tss.c:1571:(.text+0x4259): undefined reference to `__imp_curl_slist_free_all'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/tss.c:1572:(.text+0x4262): undefined reference to `__imp_curl_easy_cleanup'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-download.o: in function `download_to_buffer':
C:/msys64/home/parkm/idevicerestore/src/download.c:49:(.text+0xd5): undefined reference to `__imp_curl_easy_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/download.c:57:(.text+0xfa): undefined reference to `__imp_curl_easy_setopt'
[... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/parkm/idevicerestore/src/download.c:140:(.text+0x2f1): undefined reference to `__imp_curl_easy_cleanup'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:494: idevicerestore.exe] Error 1
make[2]: Leaving directory '/home/parkm/idevicerestore/src'
make[1]: *** [Makefile:435: all-recursive] Error 1
make[1]: Leaving directory '/home/parkm/idevicerestore'
make: *** [Makefile:367: all] Error 2

```

nikias, 2024-03-27:
Ok looks pretty good already, also this error:
```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find C:/msys64/mingw64/lib/libbz2.a: No such file or directory
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find C:/msys64/mingw64/lib/liblzma.a: No such file or directory
collect2.exe: error: ld returned 1 exit status
```
seems to be resolved. I noticed that I made a small mistake. Run these commands:
```
cd idevicerestore
./autogen.sh libzip_VERSION=1.7.1 libzip_CFLAGS="-I`pwd`/deps/include" libzip_LIBS="`pwd`/deps/lib/libzip.a /mingw64/lib/libbz2.a /mingw64/lib/liblzma.a" zlib_LIBS="/mingw64/lib/libz.a" libcurl_CFLAGS="-I`pwd`/deps/include -DCURL_STATICLIB" libcurl_LIBS="`pwd`/deps/lib/libcurl.a /mingw64/lib/libzstd.a -lws2_32 -lcrypt32 -lwldap32 -lbcrypt -lssl -lcrypto"
make
make install
cd ..
```

dsdanielpark, 2024-03-27:
I god this!
```bash
user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ make
make  all-recursive
make[1]: Entering directory '/home/user/idevicerestore'
Making all in src
make[2]: Entering directory '/home/user/idevicerestore/src'
  CCLD     idevicerestore.exe
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-idevicerestore.o: in function `main':
C:/msys64/home/user/idevicerestore/src/idevicerestore.c:1904:(.text.startup+0x539): undefined reference to `__imp_curl_global_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/idevicerestore.c:1910:(.text.startup+0x551): undefined reference to `__imp_curl_global_cleanup'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-tss.o: in function `tss_request_send':
C:/msys64/home/user/idevicerestore/src/tss.c:1535:(.text+0x4120): undefined reference to `__imp_curl_slist_append'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/tss.c:1550:(.text+0x412a): undefined reference to `__imp_curl_easy_setopt'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/tss.c:1530:(.text+0x4130): undefined reference to `__imp_curl_easy_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/tss.c:1570:(.text+0x4250): undefined reference to `__imp_curl_easy_perform'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/tss.c:1571:(.text+0x4259): undefined reference to `__imp_curl_slist_free_all'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/tss.c:1572:(.text+0x4262): undefined reference to `__imp_curl_easy_cleanup'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-download.o: in function `download_to_buffer':
C:/msys64/home/user/idevicerestore/src/download.c:49:(.text+0xd5): undefined reference to `__imp_curl_easy_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:57:(.text+0xfa): undefined reference to `__imp_curl_easy_setopt'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:76:(.text+0x190): undefined reference to `__imp_curl_easy_perform'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:77:(.text+0x199): undefined reference to `__imp_curl_easy_cleanup'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-download.o: in function `download_to_file':
C:/msys64/home/user/idevicerestore/src/download.c:108:(.text+0x217): undefined reference to `__imp_curl_easy_init'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:122:(.text+0x24c): undefined reference to `__imp_curl_easy_setopt'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:139:(.text+0x2e8): undefined reference to `__imp_curl_easy_perform'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/msys64/home/user/idevicerestore/src/download.c:140:(.text+0x2f1): undefined reference to `__imp_curl_easy_cleanup'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:494: idevicerestore.exe] Error 1
make[2]: Leaving directory '/home/user/idevicerestore/src'
make[1]: *** [Makefile:435: all-recursive] Error 1
make[1]: Leaving directory '/home/user/idevicerestore'
make: *** [Makefile:367: all] Error 2

user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ make clean
Making clean in src
make[1]: Entering directory '/home/user/idevicerestore/src'
 rm -f idevicerestore.exe
 rm -f idevicerestore
rm -rf .libs _libs
rm -f *.o
rm -f *.lo
make[1]: Leaving directory '/home/user/idevicerestore/src'
Making clean in docs
make[1]: Entering directory '/home/user/idevicerestore/docs'
rm -rf .libs _libs
rm -f *.lo
make[1]: Leaving directory '/home/user/idevicerestore/docs'
make[1]: Entering directory '/home/user/idevicerestore'
rm -rf .libs _libs
rm -f *.lo
make[1]: Leaving directory '/home/user/idevicerestore'

user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ make
make  all-recursive
make[1]: Entering directory '/home/user/idevicerestore'
Making all in src
make[2]: Entering directory '/home/user/idevicerestore/src'
  CC       idevicerestore-idevicerestore.o
  CC       idevicerestore-common.o
  CC       idevicerestore-tss.o
In file included from tss.c:28:
tss.c: In function 'tss_request_send':
C:/msys64/home/user/idevicerestore/deps/include/curl/typecheck-gcc.h:50:15: warning: call to '_curl_easy_setopt_err_long' declared with attribute warning: curl_easy_setopt expects a long argument for this option [ [... cut ...]
make[2]: Leaving directory '/home/user/idevicerestore'
make[1]: Leaving directory '/home/user/idevicerestore'

user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ cd ..

user@DESKTOP-xxxx MINGW64 ~
$ idevicerestore --version
      0 [main] idevicerestore (10732) child_copy: cygheap read copy failed, 0x800000000..0x800029A08, done 0, windows pid 10732, Win32 error 6
    468 [main] idevicerestore 1569 C:\msys64\mingw64\bin\idevicerestore.exe: *** fatal error - couldn't create signal pipe, Win32 error 5

```

dsdanielpark, 2024-03-27:
And I closed mysys2 and re-run, I god this.

```bash
parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ curl -o libzip-static.b64 -Ls "https://gist.github.com/nikias/3da15d03120382f87b44029cd8495a02/raw/99cd8138fed99e8f6530b6f179f787342c698e1f/libzip-1.7.1_static_x86_64-mingw64.tar.bz2"

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ base64 -d < libzip-static.b64 > libzip-static.tar.bz2

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ mkdir deps
mkdir: cannot create directory ‘deps’: File exists

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ tar -C deps -xjf libzip-static.tar.bz2

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$
curl -o libcurl-static.b64 -Ls "https://gist.github.com/nikias/6c397a0a2f4f4eafd91b81cccd22b761/raw/85216e60af6787f3b351291165eb91bd585ff09a/libcurl-8.1.0-static-x86_64-mingw64.tar.bz2"

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ curl -o libcurl-static.b64 -Ls "https://gist.github.com/nikias/6c397a0a2f4f4eafd91b81cccd22b761/raw/85216e60af6787f3b351291165eb91bd585ff09a/libcurl-8.1.0-static-x86_64-mingw64.tar.bz2"

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ base64 -d < libcurl-static.b64 > libcurl-static.tar.bz2

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ tar -C deps -xjf libcurl-static.tar.bz2

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ ./autogen.sh libzip_VERSION=1.7.1 libzip_CFLAGS="-I`pwd`/deps/include" libzip_LIBS="`pwd`/deps/lib/libzip.a /mingw64/lib/libbz2.a /mingw64/lib/liblzma.a" zlib_LIBS="/mingw64/lib/libz.a" libcurl_CFLAGS="-I`pwd`/deps/include -DCURL_STATICLIB" libcurl_LIBS="`pwd`/deps/lib/libcurl.a /mingw64/lib/libzstd.a -lws2_32 -lcrypt32 -lwldap32 -lbcrypt -lssl -lcrypto"
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:37: installing './compile'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... no
checking for cc... no
checking for cl.exe... no
checking for clang... no
configure: error: in '/home/parkm/idevicerestore':
configure: error: no acceptable C compiler found in $PATH
See 'config.log' for more details

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ make
 cd . && /bin/sh ./config.status Makefile
config.status: creating Makefile
make  all-recursive
make[1]: Entering directory '/home/parkm/idevicerestore'
Making all in src
make[2]: Entering directory '/home/parkm/idevicerestore/src'
 cd .. && /bin/sh ./config.status src/Makefile depfiles
config.status: creating src/Makefile
config.status: executing depfiles commands
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/parkm/idevicerestore/src'
Making all in docs
make[2]: Entering directory '/home/parkm/idevicerestore/docs'
 cd .. && /bin/sh ./config.status docs/Makefile
config.status: creating docs/Makefile
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/parkm/idevicerestore/docs'
make[2]: Entering directory '/home/parkm/idevicerestore'
make[2]: Leaving directory '/home/parkm/idevicerestore'
make[1]: Leaving directory '/home/parkm/idevicerestore'

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ make install
Making install in src
make[1]: Entering directory '/home/parkm/idevicerestore/src'
make[2]: Entering directory '/home/parkm/idevicerestore/src'
 /usr/bin/mkdir -p '/mingw64/bin'
  /bin/sh ../libtool   --mode=install /usr/bin/install -c idevicerestore.exe '/mingw64/bin'
libtool: install: /usr/bin/install -c .libs/idevicerestore.exe /mingw64/bin/idevicerestore.exe
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/parkm/idevicerestore/src'
make[1]: Leaving directory '/home/parkm/idevicerestore/src'
Making install in docs
make[1]: Entering directory '/home/parkm/idevicerestore/docs'
make[2]: Entering directory '/home/parkm/idevicerestore/docs'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/mingw64/share/man/man1'
 /usr/bin/install -c -m 644 idevicerestore.1 '/mingw64/share/man/man1'
make[2]: Leaving directory '/home/parkm/idevicerestore/docs'
make[1]: Leaving directory '/home/parkm/idevicerestore/docs'
make[1]: Entering directory '/home/parkm/idevicerestore'
make[2]: Entering directory '/home/parkm/idevicerestore'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/parkm/idevicerestore'
make[1]: Leaving directory '/home/parkm/idevicerestore'

parkm@DESKTOP-A2NI0VH CLANG64 ~/idevicerestore
$ cd ..

parkm@DESKTOP-A2NI0VH CLANG64 ~
$ idevicerestore --version
-bash: idevicerestore: command not found

parkm@DESKTOP-A2NI0VH CLANG64 ~
$ idevicerestore --version
-bash: idevicerestore: command not found
```

dsdanielpark, 2024-03-27:
I'm feeling hopeful as it seems like we're nearing a solution. I'm keeping my fingers crossed for my phone to be back in action!! Thank you so much for your help.
```bash
user@DESKTOP-xxxx MINGW64 ~
$ cd idevicerestore

user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ ./autogen.sh libzip_VERSION=1.7.1 libzip_CFLAGS="-I`pwd`/deps/include" libzip_LIBS="`pwd`/deps/lib/libzip.a /mingw64/lib/libbz2.a /mingw64/lib/liblzma.a" zlib_LIBS="/mingw64/lib/libz.a" libcurl_CFLAGS="-I`pwd`/deps/include -DCURL_STATICLIB" libcurl_LIBS="`pwd`/deps/lib/libcurl.a /mingw64/lib/libzstd.a -lws2_32 -lcrypt32 -lwldap32 -lbcrypt -lssl -lcrypto"
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:37: installing './compile'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /mingw64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
[... 95 lines cut ...]
Making install in docs
make[1]: Entering directory '/home/user/idevicerestore/docs'
make[2]: Entering directory '/home/user/idevicerestore/docs'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/mingw64/share/man/man1'
 /usr/bin/install -c -m 644 idevicerestore.1 '/mingw64/share/man/man1'
make[2]: Leaving directory '/home/user/idevicerestore/docs'
make[1]: Leaving directory '/home/user/idevicerestore/docs'
make[1]: Entering directory '/home/user/idevicerestore'
make[2]: Entering directory '/home/user/idevicerestore'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/user/idevicerestore'
make[1]: Leaving directory '/home/user/idevicerestore'

user@DESKTOP-xxxx MINGW64 ~/idevicerestore
$ cd ..

user@DESKTOP-xxxx MINGW64 ~
$ idevicerestore --latest
      0 [main] idevicerestore (49484) child_copy: cygheap read copy failed, 0x800000000..0x8000116C8, done 0, windows pid 49484, Win32 error 6
    415 [main] idevicerestore 1871 C:\msys64\mingw64\bin\idevicerestore.exe: *** fatal error - couldn't create signal pipe, Win32 error 5

```
```

nikias, 2024-03-27:
Hmm ok it built fine, now it seems it is running with some cygwin library, weird... Let me try to build this on a Windows VM and see what happens.

nikias, 2024-03-30:
Ok I took a look and it turns out that I had to install two packages:
```
pacman -S mingw-w64-x86_64-bzip2 mingw-w64-x86_64-xz
```
After that idevicerestore just compiled normally. I saw that you somehow used `x86_64-w64-mingw32`, have you been using the `MSYS2 MinGW64` shell and installed the packages exactly as I stated above? Maybe you want to start from scratch, uninstall msys2 and reinstall it again, and then run all the commands from the start.

h1laryz, 2024-05-21:
> Ok I took a look and it turns out that I had to install two packages:
>
> ```
> pacman -S mingw-w64-x86_64-bzip2 mingw-w64-x86_64-xz
> ```
>
> After that idevicerestore just compiled normally. I saw that you [...]

strange.. i have installed MSYS2 Mingw64. Openned the mingw64 shell, put all commands into terminal as above and got same error as this one:
```
user@DESKTOP-xxxx MINGW64 ~
$ idevicerestore --latest
      0 [main] idevicerestore (49484) child_copy: cygheap read copy failed, 0x800000000..0x8000116C8, done 0, windows pid 49484, Win32 error 6
    415 [main] idevicerestore 1871 C:\msys64\mingw64\bin\idevicerestore.exe: *** fatal error - couldn't create signal pipe, Win32 error 5
```

compiled successfully from scratch without any errors