libimobiledevice/idevicerestore #633: iPhone Still Locked after --erase --latest

neestzjov, 2024-05-08:
Hello!
After executing the following command : sudo idevicerestore --erase --latest. Everything looks like was installed good from my linux terminal but at the end the iphone opens up in locked state. This is the last log: 

Any help please? iphone 8 plus 16.6 firmware

[==================================================] 100.0%
Done sending SE,UpdatePayload
Extracting Stockholm4.RELEASE.sefw (Firmware/SE/Stockholm4.RELEASE.sefw)...
Sending SE TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SE,Ticket
Sending FirmwareResponse data now...
Done sending FirmwareUpdater data
Checkpoint completed id: 0xA8FBF3200001309 (update_se) result=0
Checkpoint started   id: 0x50000130D (update_savage)
Checkpoint completed id: 0x50000130D (update_savage) result=0
Checkpoint started   id: 0x500001311 (update_veridian)
Updating Veridian (66)
Checkpoint completed id: 0x500001311 (update_veridian) result=0
Checkpoint started   id: 0x1316 (update_appletcon)
Unknown operation (79)
Checkpoint completed id: 0x1316 (update_appletcon) result=0
Checkpoint started   id: 0x1318 (update_usbcretimer)
Unknown operation (82)
Checkpoint completed id: 0x1318 (update_usbcretimer) result=0
Checkpoint started   id: 0x1320 (update_manta_mcu)
Unknown operation (85)
Checkpoint completed id: 0x1320 (update_manta_mcu) result=0
Checkpoint completed id: 0x635 (update_device_firmware) result=0
Checkpoint started   id: 0x10000063C (verify_fdr_data)
Checkpoint completed id: 0x10000063C (verify_fdr_data) result=0
Checkpoint started   id: 0x500000668 (update_firmware_post_sealing)
Checkpoint started   id: 0x1501 (update_savage_postseal)
Checkpoint completed id: 0x1501 (update_savage_postseal) result=0
Checkpoint started   id: 0x1500 (decompress_pearl_data)
Checkpoint completed id: 0x1500 (decompress_pearl_data) result=0
Checkpoint started   id: 0x1503 (update_rose_postseal)
Checkpoint completed id: 0x1503 (update_rose_postseal) result=0
Checkpoint started   id: 0x131A (stage_fdr_ean)
Checkpoint completed id: 0x131A (stage_fdr_ean) result=0
Checkpoint started   id: 0x1319 (ensure_fdr_ean)
Checkpoint completed id: 0x1319 (ensure_fdr_ean) result=0
Checkpoint started   id: 0x1504 (update_fdr_ean)
Requesting EAN Data (74)
Checkpoint completed id: 0x1504 (update_fdr_ean) result=0
Checkpoint started   id: 0x1314 (update_fillmore)
Checkpoint completed id: 0x1314 (update_fillmore) result=0
Checkpoint completed id: 0x500000668 (update_firmware_post_sealing) result=0
Checkpoint started   id: 0x63D (fdr_verify_sealed_manifest)
Checkpoint completed id: 0x63D (fdr_verify_sealed_manifest) result=0
Checkpoint started   id: 0x50000066F (commit_sep_os)
Checkpoint completed id: 0x50000066F (commit_sep_os) result=0
Checkpoint started   id: 0xAA91DFF600000674 (create_protected_filesystems)
Creating Protected Volume (67)
Checkpoint completed id: 0xAA91DFF600000674 (create_protected_filesystems) result=0
Checkpoint started   id: 0x30000065F (reserve_overprov_space)
Checkpoint completed id: 0x30000065F (reserve_overprov_space) result=0
Checkpoint started   id: 0x300000628 (install_kernel_cache)
Checkpoint started   id: 0xE00 (install_kernel_cache)
About to send KernelCache...
Extracting kernelcache.release.iphone10 (kernelcache.release.iphone10)...
Personalizing IMG4 component KernelCache...
Sending KernelCache now...
Done sending KernelCache
Installing kernelcache (27)
Checkpoint completed id: 0xE00 (install_kernel_cache) result=0
Checkpoint started   id: 0xE01 (install_device_tree)
About to send DeviceTree...
Extracting DeviceTree.d211ap.im4p (Firmware/all_flash/DeviceTree.d211ap.im4p)...
[... 103 lines cut ...]
Checkpoint completed id: 0x669 (create_system_snapshot) result=0
Checkpoint started   id: 0x6AA (split_data_volume)
Checkpoint completed id: 0x6AA (split_data_volume) result=0
Checkpoint started   id: 0x6AB (commit_temporary_checkpoint)
Checkpoint completed id: 0x6AB (commit_temporary_checkpoint) result=0
Checkpoint started   id: 0x642 (unmount_filesystems)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checkpoint completed id: 0x642 (unmount_filesystems) result=0
Checkpoint started   id: 0x67C (cleanup_boot_command)
Checkpoint completed id: 0x67C (cleanup_boot_command) result=0
Checkpoint started   id: 0x1613 (cleanup_recovery_os_volume)
Checkpoint completed id: 0x1613 (cleanup_recovery_os_volume) result=0
Checkpoint started   id: 0x647 (cleanup_check_result)
Checkpoint completed id: 0x647 (cleanup_check_result) result=0
Checkpoint started   id: 0x648 (cleanup_send_final_status)
Got status message
Status: Restore Finished
ReverseProxy[Ctrl]: (status=2) Terminated
DONE
[==================================================] 100.0%

nikias, 2024-05-08:
What does "locked state" mean? If you can't activate the device then it's activation locked. A firmware restore won't allow you to bypass that.

neestzjov, 2024-05-08:
> What does "locked state" mean? If you can't activate the device then it's activation locked. A firmware restore won't allow you to bypass that.

It requires icloud email and pass to continue. The story is ... I forgot my icloud password aswell as the email and lost the phone number too. Thought I can factory reset this old iPhone so I can still use it. Also thought about erasing everything but it seems its not working this way?

neestzjov, 2024-05-08:
What could be done to bypass the activation lock?

neestzjov, 2024-05-09:
I know it sounds strange but its my personal iPhone and not stolen or found lol. I really want to still use it somehow rather than throwing it to trash. Any help suggested?

nikias, 2024-05-09:
Reset your icloud password. If it's your account but you forgot the password you should be able to recover it. You have to apply for a callback and they will call you in 14 days. https://support.apple.com/en-us/102656 Probably you need the last option.