libimobiledevice/idevicerestore #635: Iphone 13 stuck in recovery mode out of nowhere.

Bammanx, 2024-05-23:
Logs:
sudo usbmuxd -f -U root
[sudo] password for piet: 
[00:29:54.693][3] usbmuxd v1.1.1-56-g360619c starting up
[00:29:54.693][3] Using libusb 1.0.27
[00:29:54.696][3] Initialization complete

sudo idevicerestore --latest 
idevicerestore 1.0.0-188-g56d2c01
Found device in Recovery mode
ECID: 4842824977104926
Identified device as d17ap, iPhone14,5
Device Product Version: N/A
Device Product Build: N/A
The following firmwares are currently being signed for iPhone14,5:
  [1] 17.5.1 (build 21F90)
  [2] 17.5 (build 21F79)
Select the firmware you want to restore: 2
Selected firmware 17.5 (build 21F79)
Verifying 'iPhone14,5_17.5_21F79_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
IPSW Product Version: 17.5
IPSW Product Build: 21F79 Major: 21
INFO: device serial number is C2D32JXRHY
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... 29 b7 b0 c9 31 1f 11 d6 67 b0 19 15 8d 34 92 92 cb 3a 79 73 07 ba 31 0c 64 78 08 80 3b c5 9c 16 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 6c fa 22 e8 72 e0 f2 98 ad bb 85 1e 39 a9 0e 4b 5e dc 3d a1 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d17.RELEASE.im4p (Firmware/dfu/iBEC.d17.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (1031596 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-10151.122.3
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2532~iphone.im4p (Firmware/all_flash/applelogo@2532~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29649 bytes)...
Extracting h14_ane_fw_bia_d6x.im4p (Firmware/ane/h14_ane_fw_bia_d6x.im4p)...
Personalizing IMG4 component ANE...
Sending ANE (4909292 bytes)...
Extracting aopfw-iphone14aop.RELEASE.im4p (Firmware/AOP/aopfw-iphone14aop.RELEASE.im4p)...
Personalizing IMG4 component AOP...
Sending AOP (2697452 bytes)...
Extracting AppleAVE2FW_H14.im4p (Firmware/ave/AppleAVE2FW_H14.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1409524 bytes)...
Extracting sptm.t8110.release.im4p (Firmware/sptm.t8110.release.im4p)...
Personalizing IMG4 component Ap,RestoreSecurePageTableMonitor...
Sending Ap,RestoreSecurePageTableMonitor (115069 bytes)...
Extracting txm.iphoneos.release.im4p (Firmware/txm.iphoneos.release.im4p)...
Personalizing IMG4 component Ap,RestoreTrustedExecutionMonitor...
Sending Ap,RestoreTrustedExecutionMonitor (144675 bytes)...
Extracting armfw_g14p.im4p (Firmware/agx/armfw_g14p.im4p)...
Personalizing IMG4 component GFX...
Sending GFX (2057228 bytes)...
Extracting adc-kratos-d1y.im4p (Firmware/isp_bni/adc-kratos-d1y.im4p)...
Personalizing IMG4 component ISP...
Sending ISP (21113071 bytes)...
Extracting D17_MtpFirmware.im4p (Firmware/D17_MtpFirmware.im4p)...
Personalizing IMG4 component MtpFirmware...
Sending MtpFirmware (6409975 bytes)...
Extracting t8110pmp.im4p (Firmware/pmp/t8110pmp.im4p)...
Personalizing IMG4 component PMP...
Sending PMP (534764 bytes)...
Extracting rans.t8110.release.im4p (Firmware/rans.t8110.release.im4p)...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (1031708 bytes)...
Extracting iphone14d1ydcp_restore.im4p (Firmware/dcp/iphone14d1ydcp_restore.im4p)...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (5138668 bytes)...
Extracting 090-24511-112.dmg.trustcache (Firmware/090-24511-112.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (16745 bytes)...
Extracting SmartIOFirmware_ASCv5.im4p (Firmware/SmartIOFirmware_ASCv5.im4p)...
Personalizing IMG4 component SIO...
Sending SIO (1164024 bytes)...
Extracting WirelessPower.iphone14.im4p (Firmware/WirelessPower/WirelessPower.iphone14.im4p)...
Personalizing IMG4 component WCHFirmwareUpdater...
Sending WCHFirmwareUpdater (206894 bytes)...
Extracting iBootData.d17.RELEASE.im4p (Firmware/all_flash/iBootData.d17.RELEASE.im4p)...
Personalizing IMG4 component iBootData...
Sending iBootData (10710 bytes)...
ramdisk-size=0x20000000
Extracting 090-24511-112.dmg (090-24511-112.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (203434223 bytes)...
Extracting DeviceTree.d17ap.im4p (Firmware/all_flash/DeviceTree.d17ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (60132 bytes)...
Extracting sep-firmware.d17.RELEASE.im4p (Firmware/all_flash/sep-firmware.d17.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (6762987 bytes)...
Extracting kernelcache.release.iphone14 (kernelcache.release.iphone14)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (17260619 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
my iphone appears to just go back to the "insert recovery cable" screen here
Doesn't work on a mac finder, windows itunes or native linux idevicerestore. HELP

nikias, 2024-05-23:
If it doesn't work with Finder or iTunes then idevicerestore most likely won't be able to make it work. You can try to restore from DFU mode, but I fear it won't make a difference...

Bammanx, 2024-05-24:
It was the front assembly. Trying with the front cable unhooked completely resolved the issue.

Bammanx, 2024-05-24:
(the upper cable coming from the display)