libimobiledevice/idevicerestore #640: Cannot restore iPhone when iPhone is in DFU mode

hoangnam714, 2024-05-31:
When the iphone is in DFU mode, I can't restore the iphone and I get this error, if I use itune I can still restore, is there a way to overcome this error, thank you
```
Verifying 'iPhone_5.5_P3_16.7.8_20H343_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
IPSW Product Version: 16.7.8
IPSW Product Build: 20H343 Major: 20
ERROR: Device is in an invalid state
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> yes
Invalid input. Please type YES or hit CTRL+C to abort.
> YES
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... INFO: device serial number is C39VD9V2JCM2
8a 1c ca c1 0f 30 14 cb e8 c7 49 52 b7 9f 91 0a 64 cf 99 c9 f8 ba 7f c6 07 c3 b9 28 96 e4 f2 7b 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 48 f3 12 ae 8f 5e f0 cc 86 d2 c2 61 ee ce 1d 9f bf 3d 9a ac 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device

```

nikias, 2024-05-31:
Can you run it with `-d` so we can see the TSS request?

hoangnam714, 2024-05-31:
it was a suspicious error, after i reboot and re-enter DFU my iphone the command worked perfectly

closed 2024-05-31 by nikias