libimobiledevice/idevicerestore #642: No package 'libtatsu-1.0' found

TPJoseph, 2024-06-07:
How to install this package in mac

nikias, 2024-06-07:
You have to compile it yourself. For macOS, my build script - that builds everything- is taking care of it:
```bash
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L is.gd/limdmacos
bash ./limd-build-macos.sh
```

TPJoseph, 2024-06-07:
Hey Nikias,

I got that.

But it is not restoring any devices

ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_aht_block_invoke: multi-touch update failed with code 0x00000001.
Error: ((os/kern) invalid address)
)
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device

nikias, 2024-06-07:
That is not the error you mentioned above though. Feel free to paste a log output, preferably run with -d.

kotleni, 2024-06-07:
@nikias I get same error now. Check the mentioned issue please, maybe you know somethings about this.
https://github.com/libimobiledevice/idevicerestore/issues/643

kotleni, 2024-06-08:
Hahah, good morning. After night a just tried it one more time, and blah - i just have a different error.ERROR: Unable to successfully restore device

```
Checkpoint completed id: 0x648 (cleanup\_send\_final\_status) result=0
Checkpoint FAILURE id: 0x648 result=0: \[0\]D(failed to update device firmware)\[1\]D(ramrod firmware update failed)\[2\]D(update\_aht\_block\_invoke: ipd update failed with code -3
)
FDR 0x63e7aac74880 terminating...
ERROR: Unable to restore device
```

kotleni, 2024-06-08:
Oh, and one more thing:
in usbmuxd i have this log error:
```
[11:02:41.914][1] ERROR: Failed to read '/var/lib/lockdown/00008103-000240E91A3B001E.plist': No such file or directory
```

coolboy55, 2024-06-12:
Is there a way to fix this on arch Linux or use the same command the first comment provided?

igexogen, 2024-09-13:
> You have to compile it yourself. For macOS, my build script - that builds everything- is taking care of it:
> 
> ```shell
> mkdir -p limd-build
> cd limd-build
> curl -o ./limd-build-macos.sh -L is.gd/limdmacos
> bash ./limd-build-macos.sh
> ```

You need to change build order in script to build libtatsu before libimobiledevice otherwise it fails.

nikias, 2024-09-13:
@igexogen oh good catch!

nikias, 2024-09-13:
@igexogen I updated my build script

iNeedHelpX, 2024-09-18:
> @igexogen I updated my build script

thank you! just one question where is the libtatsu-1.0 installed?I ran the script but I don't understand where it may be installed