libimobiledevice/idevicerestore #643: Restore gets error: (os/kern) invalid address

kotleni, 2024-06-07:
I have just unique error when trying to restore my mac (air m1 8gb).

```
ERROR: Unable to successfully restore device
Checkpoint completed id: 0x648 (cleanup_send_final_status) result=0
Checkpoint FAILURE id: 0x648 result=0: [0]D(failed to update device firmware)[1]D(ramrod firmware update failed)[2]D(update_aht_block_invoke: ipd update failed with code 0x00000001.
Error: ((os/kern) invalid address)
)
FDR 0x5c134798fdd0 terminating...
ERROR: Unable to restore device
```

kotleni, 2024-06-07:
Github search is really strange, but i found this manually: https://github.com/libimobiledevice/idevicerestore/issues/642

kotleni, 2024-06-07:
[idevicerestore_log1.log](https://github.com/user-attachments/files/15745623/idevicerestore_log1.log)
(a first part of logs is trimmed by console limits)

kotleni, 2024-06-08:
Hahah, good morning. After night a just tried it one more time, and blah - i just have a different error.ERROR: Unable to successfully restore device

```
Checkpoint completed id: 0x648 (cleanup\_send\_final\_status) result=0
Checkpoint FAILURE id: 0x648 result=0: \[0\]D(failed to update device firmware)\[1\]D(ramrod firmware update failed)\[2\]D(update\_aht\_block\_invoke: ipd update failed with code -3
)
FDR 0x63e7aac74880 terminating...
ERROR: Unable to restore device
```

nikias, 2024-06-08:
I suspect that there is some new component that has to be added to the restore logic... A full output would be helpful, the partial output doesn't show the cause of the problem (we'd want to see the checkpoint 0x648 messages from the device logs). Did you have this working before or is it your first attempt to restore a macbook?

kotleni, 2024-06-08:
@nikias Thanks for you answer.
Yes, it's my first attempt to restore macbook.

I have some problems to get all logs from idevicerestore... just my console is cut old logs and after i can't copy it.
And... i can't do '>>' for output, because not all logs is getting with it to file. And here is no any command line argument to set log output file.

kotleni, 2024-06-08:
A part of logs you can see in message up, i just pinned it.

kotleni, 2024-06-08:
Idk, maybe it's can help. My irecovery logs:

```
=======================================
::
:: Microkernel iBootStage1 for j313, Copyright 2007-2022, Apple Inc.
::
::      Local boot, Board 0x26 (j313ap)/Rev 0x6
::
::      BUILD_TAG: iBoot-7459.141.1
::
::      BUILD_STYLE: RELEASE
::
::      USB_SERIAL_NUMBER: SDOM:01 CPID:8103 CPRV:11 CPFM:03 SCEP:01 BDID:26 ECID:00024XXXX01E IBFL:FD
::
=======================================
```

kotleni, 2024-06-08:
@nikias Here is full logs.
[full_idevicerestore_latest.log](https://github.com/user-attachments/files/15749285/full_idevicerestore_latest.log)

kotleni, 2024-06-08:
So...
[image]

Dazag, 2024-06-09:
I am experiencing the same error, is there any way I can help? (air m1 16GB) is it maybe the version we are trying to install?

kotleni, 2024-06-09:
@Dazag I tried a lot of different versions, no any changes here. We just should wait help.

kotleni, 2024-06-10:
@nikias Done.

Tecnio, 2024-06-11:
I am also having this issue, however I cannot join the Telegram as I was restoring my main device.

yannb2159, 2024-06-11:
Got the same type of error, apparently the dev is working on it, we better let him fix this issue.

kotleni, 2024-06-11:
@Tecnio Just wait, developer is already trying to fix this problem. :)

kotleni, 2024-06-12:
Related commit https://github.com/libimobiledevice/idevicerestore/issues/643

kotleni, 2024-06-12:
**Latest commit in libtatsu repo is fixing this issue:**
https://github.com/libimobiledevice/libtatsu/commit/9e52f1c75666036e4d524d7e7096309d06d1cab9

Just rebuild and reinstall lib from source code.

kotleni, 2024-06-12:
Fixed

closed 2024-06-12 by kotleni